olikraus/u8g2 #506: Can't use HW_SPI on BluePill board

baoyingcheng, 2018-02-11:
I've tested U8G2_SSD1306_128X64_NONAME_F_4W_SW_SPI, and it works fine.
I also tested Adafruit_SSD1306 display ,it works fine with both hw&sw mode.
but when I tried to use U8G2_SSD1306_128X64_NONAME_F_4W_HW_SPI, the OLED remained dark.
The hardware is OK because it can work with Adafruit driver in hw spi mode.
I don't know what's going on here, ucglib works fine in hw spi mode, the same bluepill board with a st7735 1.8 inch display.

olikraus, 2018-02-11:
Strange. There is some code which deactivates HW SPI, but this should not be true for a STM32 controller:  https://github.com/olikraus/u8g2/blob/master/cppsrc/U8x8lib.h#L49

Which STM32 do you use? How did you setup the STM32 environment for the Arduino IDE?

olikraus, 2018-02-11:
I have ordered a STM32F103C board for further testing. But it will take > 8 weeks to reach me. Is there some festival in China?

baoyingcheng, 2018-02-11:
I'm using bluepill,like this one http://wiki.stm32duino.com/index.php?title=Blue_Pill
I followed the instructions in the link above to flash the firmwire and install develop environment.
I'm sure there is nothing wrong with the hardware or environment because ucglib on st7735 1.8 inch works fine, both hw and sw spi. and also Adafruit_SSD1306 works fine with the same hardware untouched.
Maybe the default hw spi pins are wrong or something? The compile process finished without any error,but no display.
The next week is Chinese new year,aka spring festival. Everyone in China have a vacation for 7-20days.UPS, DHL and FedEX still working on there days,but if you choose free-shiping, the seller on aliexpress will use China Post international mail, and that's cheap and slow.
A piece of blue pill only cost about 1.2 usd in China,and slow mail to most countries cost only 0.5 dollar per 100g.

olikraus, 2018-02-11:
ok, thanks, I understand. Nevertheless, I guess I can not do much at the moment...

baoyingcheng, 2018-02-12:
I tried to comment out those lines you provide above, still the same.
I don't have a signal analyser,so I use some led instead, the led connect to PA7 blinks same as in sw mode,so the pin set is right.

baoyingcheng, 2018-02-22:
Really,which board did you use?
I upload a video to https://youtu.be/22BX6e9g-qE
It shows sw works fine but hw not working.
I checked for many times there's no problem with the wiring.

olikraus, 2018-03-22:
I have received two STM32 boards. I tried hard, but I was unable to upload anything. I fixed the toolchain (upload-reset did not work), I debugged usb, i tried to change boot pins...  nothing. One of the boards reports usb errors, the other simple does not react. I am sorry, I can not support here....

closed 2018-03-22 by olikraus