olikraus/u8g2 #508: Support for atmega32m1/64m1

geraldjust, 2018-02-12:
Thank you for the library! Its been very helpful with other uC. But i was wondering if there could be proper support the atmega32m1/64m1. There are only a few differences between that and a regular 328p. The only major difference is that it does not have a i2c hardware, nor timer2 i believe. Ive tried using the SPI mode, and i keep running into a error that it wants to compile the twoWire libary. Is there a way i could disable tha from happening to be able to use SPI only? or would it be easier to modify the cores "wire" library to make it compile.
Using the following with arduino IDE: https://github.com/thomasonw/ATmegaxxM1-C1
The SPI library works 

`Compiling library "SPI"
"/Users/geraldrubalcava/Library/Arduino15/packages/arduino/tools/avr-gcc/4.8.1-arduino5/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega64m1 -DF_CPU=16000000L -DARDUINO=10802 -DARDUINO_ATmega64M1_16xtl -DARDUINO_ARCH_AVR   "-I/Users/geraldrubalcava/Library/Arduino15/packages/ATmegaM1-C1/hardware/avr/1.0.5/cores/arduino" "-I/Users/geraldrubalcava/Library/Arduino15/packages/ATmegaM1-C1/hardware/avr/1.0.5/variants/ATmegaxxM1" "-I/Users/geraldrubalcava/Documents/Arduino/libraries/U8g2/src" "-I/Users/geraldrubalcava/Library/Arduino15/packages/ATmegaM1-C1/hardware/avr/1.0.5/libraries/SPI" "-I/Users/geraldrubalcava/Library/Arduino15/packages/ATmegaM1-C1/hardware/avr/1.0.5/libraries/Wire" "/Users/geraldrubalcava/Library/Arduino15/packages/ATmegaM1-C1/hardware/avr/1.0.5/libraries/SPI/SPI.cpp" -o "/var/folders/dv/2kvc2yqj1n76pfp9lbmpt_540000gn/T/arduino_build_496096/libraries/SPI/SPI.cpp.o"
Compiling library "Wire"
"/Users/geraldrubalcava/Library/Arduino15/packages/arduino/tools/avr-gcc/4.8.1-arduino5/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega64m1 -DF_CPU=16000000L -DARDUINO=10802 -DARDUINO_ATmega64M1_16xtl -DARDUINO_ARCH_AVR   "-I/Users/geraldrubalcava/Library/Arduino15/packages/ATmegaM1-C1/hardware/avr/1.0.5/cores/arduino" "-I/Users/geraldrubalcava/Library/Arduino15/packages/ATmegaM1-C1/hardware/avr/1.0.5/variants/ATmegaxxM1" "-I/Users/geraldrubalcava/Documents/Arduino/libraries/U8g2/src" "-I/Users/geraldrubalcava/Library/Arduino15/packages/ATmegaM1-C1/hardware/avr/1.0.5/libraries/SPI" "-I/Users/geraldrubalcava/Library/Arduino15/packages/ATmegaM1-C1/hardware/avr/1.0.5/libraries/Wire" "-I/Users/geraldrubalcava/Library/Arduino15/packages/ATmegaM1-C1/hardware/avr/1.0.5/libraries/Wire/utility" "/Users/geraldrubalcava/Library/Arduino15/packages/ATmegaM1-C1/hardware/avr/1.0.5/libraries/Wire/Wire.cpp" -o "/var/folders/dv/2kvc2yqj1n76pfp9lbmpt_540000gn/T/arduino_build_496096/libraries/Wire/Wire.cpp.o"
/Users/geraldrubalcava/Library/Arduino15/packages/ATmegaM1-C1/hardware/avr/1.0.5/libraries/Wire/Wire.cpp: In member function 'void TwoWire::setClock(uint32_t)':
/Users/geraldrubalcava/Library/Arduino15/packages/ATmegaM1-C1/hardware/avr/1.0.5/libraries/Wire/Wire.cpp:80:3: error: 'TWBR' was not declared in this scope
   TWBR = ((F_CPU / frequency) - 16) / 2;

   ^
Using library U8g2 at version 2.20.13 in folder: /Users/geraldrubalcava/Documents/Arduino/libraries/U8g2 
Using library SPI at version 1.0 in folder: /Users/geraldrubalcava/Library/Arduino15/packages/ATmegaM1-C1/hardware/avr/1.0.5/libraries/SPI 
Using library Wire at version 1.0 in folder: /Users/geraldrubalcava/Library/Arduino15/packages/ATmegaM1-C1/hardware/avr/1.0.5/libraries/Wire 
exit status 1
Error compiling for board ATmega64M1 (16Mhz - Xtal - Arduino).`

olikraus, 2018-02-12:
you could place an
#undef U8X8_HAVE_HW_I2C
in this line to disable I2C:

https://github.com/olikraus/u8g2/blob/master/cppsrc/U8x8lib.h#L77

Or better user

#ifdef __AVR_ATmega32M1__
#ifdef U8X8_HAVE_HW_I2C
#undef U8X8_HAVE_HW_I2C
#endif 
#endif

geraldjust, 2018-02-12:
Thank you! it seemed to compile just fine! Hopefully there could be support for software i2c in the future, would it be hard? would you need timer2 for that?. 

I added the second suggestion!

Thanks! gerald R.

geraldjust, 2018-02-12:
forgot to mention, I added both for the 32m1 & 64m1. Perhaps it could be added t the ain branch! thanks!

olikraus, 2018-02-12:
yes, I also noticed, that we need to add 32M1.

Regarding the I2C support, it really looks like there is no I2C hardware support. So there is nothing what could be done. Software I2C is always there. You can still use this.

geraldjust, 2018-02-12:
Yea hopefully i can get more support for this chip as its very unique in some of its peripheral configurations. I Also tested the Software i2c and it worked like a charm.  The atmega16m1,32m1,64m1 and the 32c1,64c1 are all basically the same uC so perhaps would be realistic to include all those ina single line?

olikraus, 2018-02-12:
>  ina single line?

es probably this makes these... But i need to figure out the buildin #defines of the avr-gcc for these controllers.

geraldjust, 2018-02-13:
Very true! But thank for fix!

Gerald R.

olikraus, 2018-02-17:
I think this is done...

closed 2018-02-17 by olikraus