olikraus/u8g2 #509: How do I print the degree symbol with u8g2?

hauwarea, 2018-02-15:
Just that!
How do I print the degree symbol (little circle in superscript) to an OLED LCD using u8g2lib?

Thanks

olikraus, 2018-02-15:
You need a font which includes the degree symbol. Usually a degree symbol is located at position 176, so the degree symbol is part of all fonts ending in "f" or "e" (at least if the font author has included the degree symbol).

A suitable font could be u8g2_font_helvR12_tf https://github.com/olikraus/u8g2/wiki/fntgrpadobex11#helvr12

Then you can used use the degree symbol from you keyboard to print it, like
```
u8g2.setFont(u8g2_font_helvR12_tf);
// ...
u8g2.drawStr(0, 20, "°");
// or
u8g2.setCursor(0, 20);
u8g2.print("°");
```

hauwarea, 2018-02-15:
How can I write a code line like _**u8g2.drawStr(0, 20, "°");**_ when I do not have a key on my keyboard for the degree symbol?

hauwarea, 2018-02-15:
Here is code that works.  The key is creating a constant of the symbol as a string with a null `terminator.`
`#include <Arduino.h>
#include <SPI.h>
#include <U8g2lib.h>

  U8G2_SSD1325_NHD_128X64_1_4W_SW_SPI u8g2(
	  /* rotation zero */ U8G2_R0,  /* clock=*/ 13, /* data=*/ 11, 
	  /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);

  const char DEGREE_SYMBOL[] = { 0xB0, '\0' };

void setup(void) 
{
	u8g2.begin();
	u8g2.enableUTF8Print();
}

void loop(void) 
{
	u8g2.firstPage();
	do {
		u8g2.setFont(u8g2_font_ncenB14_tf);
		u8g2.drawStr(5, 15, "Hello World!");
		u8g2.drawStr(40, 35, "37");
		u8g2.drawUTF8(63, 35, DEGREE_SYMBOL);
		u8g2.drawStr(73, 35, "C");
		u8g2.setFont(u8g2_font_unifont_t_symbols);
		u8g2.drawStr(0, 64, "No snow in Perth");
	} while (u8g2.nextPage());
	delay(1000);
}`

Oh, and you have to use a font that includes the required symbol.

closed 2018-02-15 by hauwarea

olikraus, 2018-02-15:
You can embed hex in strings also:
.print("\xb0");

andresw312, 2020-10-08:
para escribir símbolos simplemente dibuja con esta función

u8g2.drawUTF8(   ,   ,   , " ° ") ; 

en los primeros 2 espacios colocas los pixeles de "x" y de "y" igual que como lo haces en el "u8g2.drawStr()" y al final colocas el simbolo

ejemplo:
u8g2.drawUTF8(70,10,"°");  70pixeles a la derecha , 10 hacia abajo, imprimo el simbolo de grados

aclaro que estoy usando una LCD ST7920 128X64 
LA FUENTE TIENE QUE TERMINAR EN " tf " para que habilite esos simbolos

ejemplo:
u8g2.setFont(u8g2_font_ncenB08_tf);

natoushikong, 2022-11-03:
you must use this function DrawGlyph  instead of DrawStr
eg:C
u8g2_SetFont(&u8g2,     u8g2_font_courB18_tf    );
u8g2_DrawGlyph(&u8g2, 100,100,176   );

joelgcr, 2025-03-22:
u8g2.print(" ");
    u8g2.write(0xB0);
    u8g2.print("C");

olikraus, 2025-03-22:
or this:
``` C++
u8g2.enableUTF8Print();
u8g2.print(" °C");
```