olikraus/u8g2 #511: SSD1306 with STM32F091 problems with hw I2C

jano2358, 2018-02-17:
hi!

I'm using GNU MCU Eclipse as IDE and CubeMX to generate code. After I managed to have it working with the software I2C you provide I'm trying to set up the hardware interface. The problem is the screen shows his own art, I added some breakpoints in the debugging interface, connected my scope, and seems like the codes sent to the screen are somewhat duplicated. 
This is the first code for the initialization of the screen: 
[image]
and in the next iteration it is repeated, but not exactly:
[image]

and it follows that way.
this is how the screen looks like:
[image]

and this is the code:
`/* Includes ------------------------------------------------------------------*/
#include "main.h"
#include "stm32f0xx_hal.h"
#include <u8g2.h>

/* USER CODE BEGIN Includes */

/* USER CODE END Includes */

/* Private variables ---------------------------------------------------------*/
I2C_HandleTypeDef hi2c1;
DMA_HandleTypeDef hdma_i2c1_tx;

RTC_HandleTypeDef hrtc;
static u8g2_t u8g2;

/* USER CODE BEGIN PV */
/* Private variables ---------------------------------------------------------*/

/* USER CODE END PV */

/* Private function prototypes -----------------------------------------------*/
void SystemClock_Config(void);
static void MX_GPIO_Init(void);
static void MX_DMA_Init(void);
static void MX_I2C1_Init(void);
static void MX_RTC_Init(void);

/* USER CODE BEGIN PFP */
/* Private function prototypes -----------------------------------------------*/

uint8_t u8x8_delay_smt32f0 (u8g2_t *u8g2, uint8_t msg, uint8_t arg_int, void *arg_ptr)
{
 switch(msg)
 {
 	case U8X8_MSG_DELAY_MILLI:
 		HAL_Delay(arg_int);
 	break;

    case U8X8_MSG_DELAY_NANO:			// delay arg_int * 1 nano second
      break;

    //Function which delays 10us
	case U8X8_MSG_DELAY_10MICRO:
	for (uint16_t n = 0; n < 320 * arg_int; n++)
	{
		__NOP();
	}

	break;
	//Function which delays 100ns
	case U8X8_MSG_DELAY_100NANO:
	__NOP();

	break;

 }
[... 292 lines cut ...]
  /* USER CODE END Error_Handler_Debug */ 
}

#ifdef USE_FULL_ASSERT

/**
   * @brief Reports the name of the source file and the source line number
   * where the assert_param error has occurred.
   * @param file: pointer to the source file name
   * @param line: assert_param error line source number
   * @retval None
   */
void assert_failed(uint8_t* file, uint32_t line)
{
  /* USER CODE BEGIN 6 */
  /* User can add his own implementation to report the file name and line number,
    ex: printf("Wrong parameters value: file %s on line %d\r\n", file, line) */
  /* USER CODE END 6 */

}

#endif
`

any ideas on what can be wrong here?

olikraus, 2018-02-17:
The implementation of U8X8_MSG_BYTE_SEND is wrong. You must not start the I2C transfer there, instead you only should transfer the data bytes. 

Let us have a look at the required sequence for the SSD1306 for display on command (0xaf). Actually 9 (!) bit have to be transfered, because we need to tell, that this is a command. This is done after the i2c transfer start by sending either 0x00 (command) or 0x40 (data, see table 8-7 of the ssd1306 datasheet), so after the i2c address, there must be either 0x00 or 0x40 depending of whether data or command has to follow. The actual sequence for the display on command would be:
0x78, 0x00, 0xaf, <stop>

Now in u8g2 if have an additional abstraction layer: The Command/Arg/Data (CAD) layer (u8g2_cad.c)  will take care of this initial setup of the data or data transfer. So it is like this:
 
U8X8_MSG_BYTE_START_TRANSFER
   start I2c transfer and send the slave address (0x078)
U8X8_MSG_BYTE_SEND
  has to blindly sent the bytes, which can be the controll byte, the arg bytes or the data bytes.
  In this case this message has to send 0x00 (the control byte with the D/C information) and the
  display on command 0xaf.
U8X8_MSG_BYTE_END_TRANSFER
  Send a stop condition

This looks complicated because the I2C is split into 3 parts, but actually the problem is, that each graphcis display behaves differently. The kind of I2C transfer discussed here is specific to the SSD1306. For other displays, the first part might be different, but handled by the u8g2 cad procedures. The advantage is, that only the byte transfer needs to be implemented.

Have a look into the implementation of Arduino I2C:
https://github.com/olikraus/u8g2/blob/master/cppsrc/U8x8lib.cpp#L558

Luckily the Arduino API also has the same separation into three steps, the to implementation for HW I2C with the Arduino API is simple:
U8X8_MSG_BYTE_START_TRANSFER
  Wire.beginTransmission
U8X8_MSG_BYTE_SEND
  Wire.write()
U8X8_MSG_BYTE_END_TRANSFER
  Wire.endTransmission();
 
The problem with STM32 and CubeMX is, that these three steps are part of one function call, which will not work as expected and will send a wrong byte sequence to the display controller.

There is a STM32 implementation of Arduino API. It looks like that this code also avoids CubeMX. All in all I think that the CubeMX calls are not helpful here and I think you need to bypass them and directly write to the registers.

jano2358, 2018-02-17:
I see...
but AFAIK the arduino's wire way packs the address with the data and sent it all together whit the order ' Wire.endTransmission();'
something similar should be possible with the HAL_I2C_Master_Transmit();
it adds the address and the termination bit by itself, all I need to do is to pack the command/data in the second argument as a 2 bytes array (correct?)
do you have any suggestion on how to do it?
because I can try the registers way, but my goal after will be to do through the DMA, and if we can make all with CubeHal this can make much more easy for others too.

olikraus, 2018-02-17:
> but AFAIK the arduino's wire way packs the address with the data and sent it all together whit the order ' Wire.endTransmission();'

True, but it does not matter, because I have three different calls and can specify the data one by one.

> something similar should be possible with the HAL_I2C_Master_Transmit();

From what I have seen in the past CubeHall does not provide the same API. The only thing what you could do is to take the source code of  HAL_I2C_Master_Transmit() and split up the three calls: Init, data transfer, stop condition.
Whether you use a buffer (like Arduino IDE) or not does not matter. 

> it adds the address and the termination bit by itself, all I need to do is to pack the command/data in the second argument as a 2 bytes array (correct?)

Yes, but you can not do this on byte level, because there is no information whether to send data or commands. You need to write your own CAD procedure:

https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_cad.c#L410

On that level you can use CubeHal and send the complete sequence including the data/command byte (as described in the SSD1306 datasheet). 

> this can make much more easy for others too

Writing your own CAD procedure will be easy and you can use CubeHal, bit this solution is very specific to the SSD13xx controller family.

olikraus, 2018-02-17:
> but you can not do this on byte level, because there is no information whether to send data or commands

Ok, this is not fully true: The byte procedure should receive U8X8_MSG_BYTE_SET_DC message which should inform the byte procedure whether the next byte is a command or data type. The problem however is, that the current SSD1306 cad i2c procedure does not send this message, because it is not required for the i2c byte procedures at the moment. So it is still true, you have to create a new cad procedure if you want to use CubeHal

olikraus, 2018-02-18:
Maybe this is not a HAL problem, but instead you have used the wrong driver. It might be a SH1106 display (which is similar to the SSD1306 but has four extra columns). So it might be a 132x32 or 132x64 OLED.

jano2358, 2018-02-18:
But if I use the software I2C it don't make this problem, I just tied again to be sure but when I choose u8x8_byte_sw_i2c as third argument of u8g2_Setup_ssd1306_i2c_128x64_noname_2 it makes the square complete.

jano2358, 2018-02-18:
Found it!
these lines from u8x8_cad.c are no necessary for my case:
`//       while( arg_int > 24 )
//      {
//	u8x8_i2c_data_transfer(u8x8, 24, p);
//	arg_int-=24;
//	p+=24;
//      }`

the relevant part of this file now looks like this:

`static void u8x8_i2c_data_transfer(u8x8_t *u8x8, uint8_t arg_int, void *arg_ptr)
{
	uint8_t buffer[arg_int];
	           uint8_t *ptr = arg_ptr;
	           buffer[0] = 0x40;
	           for (int i = 1; i <= arg_int; i++) buffer[i] = *(ptr++);
	           HAL_I2C_Master_Transmit(&hi2c1, 0x78, (uint8_t *)buffer, arg_int, HAL_MAX_DELAY);
//	           HAL_I2C_Master_Transmit_DMA(&hi2c1, 0x78, (uint8_t *)buffer, arg_int);
}

uint8_t u8x8_cad_ssd13xx_i2c(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
{
  uint8_t *p;
  uint8_t buffer[2];
  switch(msg)
  {
    case U8X8_MSG_CAD_SEND_CMD:
    case U8X8_MSG_CAD_SEND_ARG:
		buffer[0] = 0x00;
		buffer[1] = arg_int;
      HAL_I2C_Master_Transmit(&hi2c1, 0x78, (uint8_t*) buffer, 2, HAL_MAX_DELAY);
//      HAL_I2C_Master_Transmit_DMA(&hi2c1, 0x78, (uint8_t *)buffer, 2);

      break;
    case U8X8_MSG_CAD_SEND_DATA:
      p = arg_ptr;
      u8x8_i2c_data_transfer(u8x8, arg_int, p);
      break;
    default:
      return 0;
  }
  return 1;
}`

so far there is no problems with speed, the interface is set a 1MHz and working fine, next step is to make it work with DMA
Thanks for your help!

olikraus, 2018-02-18:
:-)

closed 2018-02-20 by olikraus