olikraus/u8g2 #512: framerate

blancoberg, 2018-02-17:
Not sure if this is an issue with u8g2 or just me lacking skills.
It takes about 84 milliseconds to complete the sendbuffer() function when using it on an STM32F1 board. When trying the same thing on an Arduino Uno, I get 16 milliseconds. the stm32 should be a lot faster what I'v read. Or am I misunderstanding something? 

I'm using this constructor: 

`U8G2_SSD1306_128X64_NONAME_F_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);`

it works  but is very slow.
Considering this board is clocked at 72mhz compared to the unos 16mhz, the results should be the other way around :)

blancoberg, 2018-02-17:
I should mention that i'm using the Arduino_STM32 library. The issue could also be with that library.
.

olikraus, 2018-02-18:
Software emulated SPI (SW_SPI) might be indeed slow. Are you able to use HW SPI?
```
U8G2_SSD1306_128X64_NONAME_F_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);
```
The problem might be to identify the clock and data lines for hardware SPI.

olikraus, 2018-02-18:
nice :)

closed 2018-02-18 by olikraus