olikraus/u8g2 #513: Handling multiple SPI SSD1306 issue

JohnSmith2591, 2018-02-18:
Hello,

I've seen you solved the problem with handling multiple SSD1306 OLED screens (issue #192).
I would need to drive 4 of these screens independantly. I made the test with only 2 screens.

I'm using u8g2 vers. 2.20.13 with Arduino DUE
Here is my wiring setup :
Pin 13 : clock                 Pin 9 : DC
Pin 11 : data                  Pin 8 : Reset
Pin 10 : CS Scr1             Pin 7 : CS Scr2

I merged two examples : the "Hello World" one, and the "scrolling text" one.
The problem is only the last initialized screen is displaying the correct information. The other remains black.
With the sketch below, only screen2 is displaying scrolling text.
If I switch scr1.begin(); and scr2.begin(); then screen1 will display "hello world".

Am I missing something ?

Thx for your help!

[multiple_OLEDs_test.zip](https://github.com/olikraus/u8g2/files/1734990/multiple_OLEDs_test.zip)

olikraus, 2018-02-18:
I think you must use two different reset lines, because of this init code:
```
  scr1.begin();  // this will reset both scr1 AND scr2
  scr2.begin();  // this will again reset scr1 AND scr2
```
the problem is, that a reset will turn the display off. So after this init sequence only scr2 will show something because scr1 is off.

You could just do this:
```
U8G2_SSD1306_128X64_NONAME_1_4W_SW_SPI scr1(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);
U8G2_SSD1306_128X64_NONAME_1_4W_SW_SPI scr2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* cs=*/ 7, /* dc=*/ 9, /* reset=*/ U8X8_PIN_NONE);
```
Looks odd, but should work if scr2 reset is still connected to pin 8.

JohnSmith2591, 2018-02-19:
I tested both above mentionned solutions, and both work.
I'll try to use different reset pins if I can. I'll come to the "dirty" way to save pins if more of them are needed for my project.

Thanks a lot for your support, and for your great library !

olikraus, 2018-02-19:
:-)

closed 2018-02-20 by JohnSmith2591