olikraus/u8g2 #515: Guru Meditation error "Core 0 panic" when using u8x8 NULL device

cyberman54, 2018-02-19:
I'm trying to use the new NULL display in an arduino-esp32 multiboard environment with u8g2 2.21.8, using the below sequence to initialize U8x8.

When using the NULL device the first occurance of `u8x8.drawString()` in code causes a Guru Meditation error "Core 0 panic".

It seems that this does not happen with ` u8x8.clearLine()`
   

```
// OLED driver
#include <U8x8lib.h>

#ifdef HAS_DISPLAY
    U8X8_SSD1306_128X64_NONAME_HW_I2C u8x8(OLED_RST, OLED_SCL, OLED_SDA);
#else
    U8X8_NULL u8x8;
#endif
```

cyberman54, 2018-02-19:
On the board without display (using the NULL device) i use exactly the same code with as on the board with a SSD1306 device (which runs the code). So i assume the guru mediation error is not caused by my code?

I figured out the panic happenes at this point:

```
40122864 <u8x8_get_glyph_data>:
40122864:	004136        	entry	a1, 32
40122867:	6298      	l32i.n	a9, a2, 24
**40122869:	000982        	l8ui	a8, a9, 0**
4012286c:	010992        	l8ui	a9, a9, 1
4012286f:	393937        	bltu	a9, a3, 401228ac <u8x8_get_glyph_data+0x48>
```

```
Guru Meditation Error: Core  1 panic'ed (LoadProhibited)
. Exception was unhandled.
Register dump:
**PC      : 0x40122869**  PS      : 0x00060a30  A0      : 0x800d48db  A1      : 0x3ffd1cd0
A2      : 0x3ffc49bc  A3      : 0x00000050  A4      : 0x3ffd1cf4  A5      : 0x00000001
A6      : 0x00060c20  A7      : 0x00000003  A8      : 0x80086879  A9      : 0x00000000
A10     : 0x00000001  A11     : 0x00000000  A12     : 0x00000013  A13     : 0x00000001
A14     : 0x00060a23  A15     : 0x3ffd1b54  SAR     : 0x00000004  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffd
```

ZinggJM, 2018-02-19:
"Guru Meditation Error" - I can't stop laughing!
Is this a two way translation outcome of "Bus Arbitration Error"?

ESP32 and ESP8266 do not like byte access to flash, all accesses need be 32bit. That's why these processors have pgmspace.h for Arduino IDE

olikraus, 2018-02-19:
> I figured out the panic happenes at this point:

So, did you assign a font? u8x8_get_glyph_data does not check for a proper font (if I remember correctly)
The l32i.n command probably loads the address of the font member variable (which is zero). 
Then the code tries to read the first value via 0 address (which will fail):
https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_8x8.c#L66

I am very sure you missed a call to u8x8_SetFont()

> "Guru Meditation Error" - I can't stop laughing!

Looks like they took this over from AmigaOS

> ESP32 and ESP8266 do not like byte access to flash, all accesses need be 32bit.

Partly true, but this is handled in U8g2 for ESP8266 (https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_8x8.c#L41) and should not be the root cause.
ESP32 should have no problem with byte access.

cyberman54, 2018-02-19:
It was the missing font declaration.
The relevant portion of code was disabled in my code for boards without display:
```
u8x8.begin();
u8x8.setFont(u8x8_font_chroma48medium8_r);
```
After i enabled this, everything is working fine now.
Thank you for prompt help. You know your code better than i know mine! :-)

closed 2018-02-19 by cyberman54

olikraus, 2018-02-19:
There is already an issue regarding this topic, but the question is, shell I add extra (usually useless) code which also slows down execution to check for this? It is actually a difficult decision whether to check or not to check all the user mistakes.

cyberman54, 2018-02-19:
I wouldn't do it. But it should be noted in the documentation, that also for NULL devices a `.begin()` and a `.setFont()` is mandatory, since this may not be expected.