olikraus/u8g2 #518: Return meaningful value from begin() and initDisplay()

daleckystepan, 2018-02-22:
Hi,
let me thank you fot this great library which I'm using for wireless sensor node at home.

It would be very good if it is possible to return  meaningful value from begin() and initDisplay(). For example bool value which can indicate successful initialization and properly connceted display.

Maybe other methods can also return non void values.

Thank you

olikraus, 2018-02-22:
well, there are two doubts from my side:
 - As far as I have seen, any Arduino library should have a begin(). But I have never seen any begin() member function which returns a value.
- initDisplay could return something, but u8g2 at the moment will never fail. In fact u8g2 does not and can not read anything from the display, so there is no feedback from the display itself. U8g2 does not even know whether there is a display. As a consequence, initDisplay will always return true.

olikraus, 2018-03-24:
Is there any examples of what should there be as a return value to begin()? Is there any official docmentation?

daleckystepan, 2018-03-24:
Sorry for late responding. I missed the notification email.

I don't think there is official documentation what to return from begin function. It is only convention developed by library creators.

I'm using:
https://github.com/LowPowerLab/RFM69/blob/master/RFM69.h
which is returning bool from  initialize()

or https://github.com/adafruit/TSL2561-Arduino-Library/blob/master/TSL2561.h
which is returning boolean from begin()

olikraus, 2018-03-24:
no problem. I have added a return value for begin(). It will always return 1/true.

closed 2018-03-24 by olikraus