olikraus/u8g2 #519: drawstring a variable u8x8

Audio-Rochey, 2018-02-23:
Oli, all,

I'm hitting my head against a wall. I'm not the worlds best programmer, so if I'm asking something that's more "this is how you program" rather than "this is something wrong with the library", please point me in the right direction and I'll stay out of this thread!

I can print fixed strings just fine. Code like the following shows great
"u8x8.drawString(0,0,"  Text here ");"

But as soon as I try to use
u8x8.drawString(0,0, VariableInUse);

It falls apart.

VariableInUse is an INT, so I tried converting it in the line beforehand
String viuString= String(VariableInUse);

But then the arduino (Leonardo) compiler complains at me.

Any advice? Directions?

Many thanks

olikraus, 2018-02-23:
Well, drawString does not automatically convert a variable into a string, Maybe you could use print, which u8x8 took over from serial.print(). This is also why the documentation for u8x8.print() is here: https://www.arduino.cc/en/Serial/Print

So the full example is:
```
u8x8.setFont(u8x8_font_amstrad_cpc_extended_r);
u8x8.setCursor(0, 0);
u8x8.print(VariableInUse);  
```
Of course there are other ways for converting a variable to a string, but print() is common to Arduino users.

closed 2018-03-24 by olikraus

GrahamOB, 2018-09-30:
Thanks Oli, fixed my issue too.

olikraus, 2018-09-30:
:)

frederikheld, 2020-05-11:
For anyone else who ended up here and forgot how the degenerate string clusterf*ck in C works. 

This worked for me:

```C
    int counter = 10;
    u8x8.drawString(0, 0, String(counter).c_str());
```

You're welcome.

olikraus, 2025-11-09:
hmm, it is there: https://github.com/olikraus/u8g2/wiki/u8x8reference#print

Maybe I should mention, that it also prints the content of numeric variables....

Edit: I have mentioned, that .print() will also write numeric data from variables.