olikraus/u8g2 #520: HAL for ESP8266 NONOS/esp_open_sdk?

martin-ger, 2018-02-24:
I am intending to use the c lib with the ESP8266 NONOS/esp_open_sdk? Is there a special HAL function required/available or can I use the Arduino version?

Thx for any input.

olikraus, 2018-02-24:
ESP8266 is supported by the Arduino IDE. You need to install the ESP8266 Board Package as described here: https://learn.adafruit.com/adafruit-feather-huzzah-esp8266/using-arduino-ide

Then install U8g2 from within the Arduino IDE: https://github.com/olikraus/u8g2/wiki/u8g2install

martin-ger, 2018-02-24:
Thanks. Yes, I know that there is working Arduino support for the ESP8266.
However, for my projects I don't use the C++/Arduino environment but the plain Espressif SDK, resp. the esp_open_sdk and C. I'm not sure, whether I can use the GPIO/delay functions of the Arduino port here. 

What It actually want to do is to print out some infos on the state of the WiFi Repeater (https://github.com/martin-ger/esp_wifi_repeater) on these kind of devices: http://www.raspberrypiwiki.com/index.php/WEMOS_ESP8266_Board_with_OLED_%26_18650_Battery_Holder

olikraus, 2018-02-24:
You can create your own interface: https://github.com/olikraus/u8g2/wiki/Porting-to-new-MCU-platform

closed 2018-03-24 by olikraus