olikraus/u8g2 #521: support SSD1305 128x64

xaviergh, 2018-02-26:
Hello, I have this [display](https://www.adafruit.com/product/2719) from Adafruit that uses the SSD1305 chip, I have tested successfully using the 128x32 constructor but I can't use the whole display. Is there a way to add that extra space in code?

Thanks.

olikraus, 2018-02-26:
It is not yet implemented. I can add this. Are you willing to do some testing here?

xaviergh, 2018-02-26:
That would be great! I have the hardware ready to test, just let me know.

Thank you.

olikraus, 2018-02-26:
I have created beta release 2.22.1 for further testing. Download from here: https://github.com/olikraus/U8g2_Arduino
The zip file can be installed via the Add zip library manu of the Arduino IDE.

Especially let me know whether the  u8x8/flipmode example works.

xaviergh, 2018-02-27:
Hello, I did some testing and it seems it still take half the screen. Attached is the result of the flip example:
[image]

I ran a hello world to test unicode support with the following:

```
U8G2_SSD1305_128X64_ADAFRUIT_F_4W_HW_SPI u8g2(U8G2_R2, pin_oled_cs, pin_oled_dc, 
pin_oled_reset);
```
Setup:
```
u8g2.begin();  
u8g2.enableUTF8Print();
```
Loop:
```
u8g2.setFont(u8g2_font_ncenB08_tf);
u8g2.clearBuffer();
u8g2.setCursor(0, 15);
u8g2.print(" Hello World 1");
u8g2.setCursor(0, 40);
u8g2.print("Hello World 2");
u8g2.sendBuffer();
delay(1000);
```
Just to let you know I'm using a SAMD chip (Adafruit Feather M0) for the test.

[image]

olikraus, 2018-02-27:
ok, thanks for the feedback, I have created another beta:
eRlease 2.22.2 . Again, download from here: https://github.com/olikraus/U8g2_Arduino

xaviergh, 2018-02-28:
I tested again and it seems to work now, just a small issue, in the picture you can see that the edge is cropped. To fix it I have to add an extra space at the beginning.
[image]
Like this:
`u8g2.print(" Hello World 1");`
Instead of
`u8g2.print("Hello World 1");`

Thank you for your work!

olikraus, 2018-02-28:
strange... It is not cropped, but the text starts on the upper right and wraps over to the left side.
But initially the u8x8 flipmode example did look better. Does this work with u8x8? Do you see similar effects with u8x8? Can you run the u8x8 graphcs test or other examples?

For the example above, can you post the complete source code?

xaviergh, 2018-02-28:
I used the u8x8 and it has the same issue.
[image]
The source code is the PrintUTF8 example from full buffer except that I change the chinese text with "Hello World 2".
However something weird happened, by mistake I used the SSD1306 constructor and it works without any issue (Both u8x8 and u8g2), even tho I know that SS1305 and SS1306 are different chips.

olikraus, 2018-02-28:
Strange. So you can just use the SSD1306 constructor and your problem is solved?

olikraus, 2018-03-24:
no feedback, closing...

closed 2018-03-24 by olikraus