olikraus/u8g2 #522: Support for Winstar SSD1306 64x48 displays

matteoscordino, 2018-02-26:
I added support for these display to my own fork of u8g2 and I would be happy to open a PR.

However, there is a section of u8g2.h that I had to modify that starts with : 
`/* u8g2_d_setup.c generated code start */`

So I am not sure I should be editing it manually, but OTOH I can't easily see what would be generating that section.

I guess I'll create a PR as-is and then I can fix it before merge?

matteoscordino, 2018-02-26:
The bulk of the East Rising code is essentially working for Winstar as well, but I had to alter the u8x8_display_info_t struct to adapt the geometry. [This is the Winstar one](https://github.com/SilexEmbedded/u8g2/blob/master/csrc/u8x8_d_ssd1306_64x48_winstar.c).

olikraus, 2018-02-26:
I have created beta release 2.22.1 for further testing. Download from here: https://github.com/olikraus/U8g2_Arduino
The zip file can be installed via the Add zip library manu of the Arduino IDE.

Especially let me know whether the  u8x8/flipmode example works.

matteoscordino, 2018-02-26:
Er... that was fast =)

Unfortunately I cannot test the Arduino build, as I am working on a Silicon Labs EFR32. I was planning of submitting my port for the EFR32/EFM32 family of processors later on.

What I can do I get your code, merge with my port and test it like that.

olikraus, 2018-02-26:
> that was fast =)

hehe, I did this together with #521 

> What I can do I get your code, merge with my port and test it like that.

No problem, Let me know if you observe any issues.

matteoscordino, 2018-02-26:
A couple of comments:

- I see you have merged my code in the same file as East Rising, [here](https://github.com/olikraus/u8g2/blob/720517093d6b14b3f2adaf44669d22ff3e766c50/csrc/u8x8_d_ssd1306_64x48.c),  However, the geometry for this display is 48x64, maybe the code should be in a different file, named u8x8_d_ssd1306_64x48.c?

- to reflect the comment above, in my code I have renamed functions to be *_48x64_*

- AFAIK this display is only available in I2C versions, so having the SPI methods might not be needed?

I will push the suggested changes soon and link here

matteoscordino, 2018-02-26:
This is what I meant with the comment above: https://github.com/olikraus/u8g2/pull/523

olikraus, 2018-02-26:
> to reflect the comment above, in my code I have renamed functions to be 48x64

Well, I do not mind (although i recently tried to have only one file per controller). 

> AFAIK this display is only available in I2C versions, so having the SPI methods might not be needed?

But there might be other OLEDs which have the same geometry but a different interface. It does not increase codesize to add the other interfaces.

matteoscordino, 2018-02-26:
I see. I'll make the changes to the PR

olikraus, 2018-03-24:
can I close this issue?

matteoscordino, 2018-03-24:
Yup!

I have implemented hardware fade out and hardware horizontal scroll as
well, but there is no public API for the functions, so I'm not sure they're
worth merging...

olikraus, 2018-03-25:
ok, thanks.

closed 2018-03-25 by olikraus