olikraus/u8g2 #524: DogXl240 "no matching function"

Multichiller, 2018-02-27:
Hello :)
I use a EA DogXl240 Display with an Elegoo Uno R3 (same pinmap as Arudino Uno).
But when I try to compile the example project with my constructor, I get this response: 

> Arduino: 1.8.2 (Windows 10), Board: "Arduino/Genuino Uno"
> display:6: error: no matching function for call to U8G2_UC1611_EA_DOGXL240_1_4W_HW_SPI::U8G2_UC1611_EA_DOGXL240_1_4W_HW_SPI(const u8g2_cb_t*, int, int, int, int, int)
> 
>  U8G2_UC1611_EA_DOGXL240_1_4W_HW_SPI u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);
                                                                                                 
> 
> C:\Users\...\Documents\display\display.ino:6:120: note: candidates are:
> 
> In file included from C:\Users\...\Documents\display\display.ino:3:0:
> 
> C:\Users\...\Documents\Arduino\libraries\U8g2\src/U8g2lib.h:5592:11: note: U8G2_UC1611_EA_DOGXL240_1_4W_HW_SPI::U8G2_UC1611_EA_DOGXL240_1_4W_HW_SPI(const u8g2_cb_t*, uint8_t, uint8_t, uint8_t)
> 
>    public: U8G2_UC1611_EA_DOGXL240_1_4W_HW_SPI(const u8g2_cb_t *rotation, uint8_t cs, uint8_t dc, uint8_t reset = U8X8_PIN_NONE) : U8G2() {
> 
>            
> 
> C:\Users\...\Documents\Arduino\libraries\U8g2\src/U8g2lib.h:5592:11: note:   candidate expects 4 arguments, 6 provided
> 
> C:\Users\...\Documents\Arduino\libraries\U8g2\src/U8g2lib.h:5591:7: note: constexpr U8G2_UC1611_EA_DOGXL240_1_4W_HW_SPI::U8G2_UC1611_EA_DOGXL240_1_4W_HW_SPI(const U8G2_UC1611_EA_DOGXL240_1_4W_HW_SPI&)
> 
>  class U8G2_UC1611_EA_DOGXL240_1_4W_HW_SPI : public U8G2 {
> 
>        
> 
> C:\Users\...\Documents\Arduino\libraries\U8g2\src/U8g2lib.h:5591:7: note:   candidate expects 1 argument, 6 provided
> 
> C:\Users\...\Documents\Arduino\libraries\U8g2\src/U8g2lib.h:5591:7: note: constexpr U8G2_UC1611_EA_DOGXL240_1_4W_HW_SPI::U8G2_UC1611_EA_DOGXL240_1_4W_HW_SPI(U8G2_UC1611_EA_DOGXL240_1_4W_HW_SPI&&)
> 
> C:\Users\...\Documents\Arduino\libraries\U8g2\src/U8g2lib.h:5591:7: note:   candidate expects 1 argument, 6 provided
> 
> exit status 1
> no matching function for call to U8G2_UC1611_EA_DOGXL240_1_4W_HW_SPI::U8G2_UC1611_EA_DOGXL240_1_4W_HW_SPI(const u8g2_cb_t*, int, int, int, int, int)

 
The code:
```
#include <Arduino.h>
#include <SPI.h>
#include <U8g2lib.h>

U8G2_UC1611_EA_DOGXL240_1_4W_HW_SPI u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);

void setup(void) {
  u8g2.begin();
}

void loop(void) {
  u8g2.firstPage();
  do {
    u8g2.setFont(u8g2_font_ncenB14_tr);
    u8g2.drawStr(0,20,"Hello World!");
  } while ( u8g2.nextPage() );
  delay(1000);
}
```
Does anybody know a solution? 
Thank you

olikraus, 2018-02-27:
The full list is here: https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#uc1611-ea_dogxl240

Either use
U8G2_UC1611_EA_DOGXL240_1_4W_ ***SW*** _SPI u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);
or
U8G2_UC1611_EA_DOGXL240_1_4W_ ***HW*** _SPI u8g2(U8G2_R0, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);

Multichiller, 2018-02-28:
Thank you :) it works perfectly

closed 2018-03-24 by olikraus