olikraus/u8g2 #525: Port to Silicon Labs EFR32

matteoscordino, 2018-02-27:
Hello,

for a project I am working on I ported u8g2 to an EFR32 chip from Silicon Labs. 
At the moment I have support for HW I2C and of course the gpio/delay methods for bitbanging (which I tested for I2C too, but I guess it would work for SPI as well). 
I think it's generic enough that it would run on similar chips too (the EFM32 family is a lot more common, AFAIK, but uses the same I2C controller and GPIO)

The issue I see at the moment is that AFAIK the Gecko SDK for EFR32 chips is only available through Silicon Labs and not for free. Would it make sense to provide my code, considering that it couldn't be built unless you have access to the Gecko SDK?

How would we go about it?

olikraus, 2018-02-27:
If you already have a github project page (or something similar), then I would be happy to mention it here: https://github.com/olikraus/u8g2/wiki/Porting-to-new-MCU-platform

matteoscordino, 2018-03-24:
Hello,
I am still working on it to make it a bit less dependant on the Thread SDK which I am using now, since it's just this SDK that is paid for (while the basic one is available even on GitHub)

olikraus, 2018-03-24:
ok, no problem. Thanks for the feedback.

olikraus, 2018-04-14:
closing... feel free to reopen if required...

closed 2018-04-14 by olikraus