olikraus/u8g2 #526: Adding a custom font

matteoscordino, 2018-02-27:
Hello,

I need to use a specific font that is provided by a 3rd party as ttf. I successfully converted it to a bdf rasterised font, so I'm fine with using either format.

However, I can't find any documentation about how to use a custom font with u8g2. Is there any documentation about it? Otherwise, could you provide a few pointers?

Thanks in advance

olikraus, 2018-02-27:
There should be several issues here with the same question. I also added some hints to the U8g2 FAQ: https://github.com/olikraus/u8g2/blob/master/doc/faq.txt#L179

Let me know if you need more specific help here.

matteoscordino, 2018-02-28:
Oh, thanks, the documentation in the FAQ is exactly what I needed!
I was not even aware that .txt existed, I kept looking thought the Wiki and couldn't find it.

Thank you

closed 2018-02-28 by matteoscordino

eaglewen2020, 2021-04-12:
Hi,
question...in your FAQ is written:

Q: How can I generate my own font?
A: The font must be available in bdf file format. Then use bdfconv to generate
the font data. The font data can be pasted into an existing file of your project.
There is also a nice Windows Bitmap Font Editor "Fony" (http://hukka.ncn.fi/?fony) 
which can export .bdf files.

I have exported a new font to bdf and converted it to a .c file with bdfconv.exe

How do I use this in my ino file. I tried to include it as a specific file, I tried to declare it in your u8g2lib...so siple question...how do you enable the use of this new font data....just paste it, but where?

Thanks
Werner