olikraus/u8g2 #527: SSD1322 Support

Zborex, 2018-02-27:
Hello would You like to implement support for ssd1322?
I'm looking for help in setting up an 2x ssd1322 oled 256x128 pix
I can provide You a support. I would like to use it in MBED

Regards

Zborex, 2018-02-27:
edit:
Please close i found the solution in FAQ ;)

olikraus, 2018-02-28:
What is your solution then? This display is probably not yet supported... But I think it could be done, based on one of the existing devices.

Zborex, 2018-02-28:
I think i will use 2 256x64 while one rotated, i will let You know if it will work

Zborex, 2018-02-28:
I worked today to try to solve it, but I have problems on setting up the display through the Clib, especially in making the uC specific code implementation for the Mbed software. I want to use lpc1768 with hardware SPI

olikraus, 2018-02-28:
So you did follow the instructions form here? https://github.com/olikraus/u8g2/wiki/Porting-to-new-MCU-platform

closed 2018-03-24 by olikraus