olikraus/u8g2 #529: Add proper kerning for RTL scripts(persian, arabic, ...)

IMAN4K, 2018-03-02:
Hi @olikraus 
Thanks for your great work, it's precious!

I've followed issues and works for supporting kerning for RTL languages in these threads:
* https://github.com/olikraus/u8g2/issues/234
* https://github.com/olikraus/u8g2/issues/42
* https://github.com/olikraus/u8g2/issues/39
* https://github.com/olikraus/u8g2/issues/14
* https://github.com/olikraus/U8glib_Arduino/issues/14

I'm going to add this feature to `u8g2` but first need to know what part of `u8g2` needs modification and what is the right way to do it.
Since adding manual kerning as your last work will not solve the issue: https://github.com/olikraus/u8g2/issues/42#issuecomment-241196338
So if possible please help me to start.

Regards.
Iman.

olikraus, 2018-03-02:
I think there are two different problems.

1. Combining two glyphs into one under specific conditions #234 
2. Reducing the size between glyphs #42

Which problem do you want to solve/improve? Both of the problems or only one of the problems?
Basically we could leave problem one to the writer of the text. That means, U8g2 can expect the correct utf-8 code of the combind glyph. So the only problem is the kerning itself. I think the solution in #42 should be good enough. In fact my suggestion would be to give it a try with a bigger portion of text and enhance the same if required.

IMAN4K, 2018-03-02:
At first glance, if i put chars with these unicodes:

| Unicode  | Char |
| ------------- | ------------- |
| 0xFEB3  | ﺳ  |
| 0xFEE0  | ﻠ  |
| 0xFE8E  | ﺎ  |
| 0xFEE1  | ﻡ  |
| 0xFEA9  | د  |
| 0xFEE7  | ﻧ  |
| 0xFEF4  | ﻴ  |
| 0xFE8E  | ﺎ  |

The string is rendered properly(kerning table is required)!
```
const uint16_t my_kerning_table[] = {
	/* first char, second char, gap reduction value */
	0x646, 0x627, 4, /* ن and ا */
	0x627, 0x647, 5, /* ا and ه */
	0x647, 0x62C, 6, /* ه and ج */
	0x645, 0x627, 3, /* م and ا */
	0x627, 0x644, 4, /* ا and ل */
	0x644, 0x633, 2, /* ل and س */

	/* this line terminates the table */
	0xffff, 0xffff, 0xffff
};

void setup() {
	LCD.begin();
	LCD.setFont(u8g2_font_unifont_t_arabic);
	LCD.drawExtUTF8(80, 64 / 2, 1, my_kerning_table, "ﺳﻠﺎﻡ دﻧﻴﺎ");
	LCD.sendBuffer();
}
```
[image]

IMAN4K, 2018-03-02:
More interesting :)
We've right text in display with even empty kerning table (but without it problem exists!)
Should dive into `drawExtUTF8(...)` source, see what's going on.
```
const uint16_t my_kerning_table[] = {
	/* this line terminates the table */
	0xffff, 0xffff, 0xffff
};

void setup() {
	LCD.begin();
	LCD.setFont(u8g2_font_unifont_t_arabic);
	LCD.drawExtUTF8(80, 64 / 2, 1, my_kerning_table, "ﻤﺎﮬﻲ ﺧﻮﺐ");
	LCD.sendBuffer();
}
```
[image]

IMAN4K, 2018-03-07:
@olikraus 
For now we provide exact unicode for proper text display to `drawExtUTF8(...)` with empty kerning table.
I'm thinking to have a lookup table to replace the correct unicodes.
e.g
If we have unicode `0x0633:س` followed by `0xFEDD:ﻝ` then we should replace first one with `0xFEB3:ﺳ` and so forth.

olikraus, 2018-03-07:
ok, but i think this should be done during typing in the string into Arduino IDE.
I think this is not something for u8g2.

IMAN4K, 2018-03-07:
Adding Kerning for `RTL` scripts make `u8g2` even better!
So if you have no problem with an **optional** implementation(adding a control macro):
`#define U8G2_ENABLE_RTL_KERNING`
Let me know to commit in the next few days.

olikraus, 2018-03-07:
I tried already some time back. The kerning tables will be VERY huge. That is why I added `drawExtUTF8`.
Is drawExtUTF8 not sufficient? You can nicely construct the kerning table exactly for your need.

olikraus, 2018-03-24:
no further feedback, closing...

closed 2018-03-24 by olikraus