olikraus/u8g2 #530: ESP8266 Arduino userInterfaceSelectionList WDT Reset

microdimmer, 2018-03-02:
Hello!
Is there need additional GPIO tuning to use userInterfaceSelectionList on NodeMCU 1.0?
Trying to run "SelectionList" example even with NULL device (U8G2_NULL u8g2(U8G2_R0)) and catching Soft WDT reset.
In my project i use ST7920 and `U8G2_ST7920_128X64_F_SW_SPI u8g2(U8G2_R0,12,13,15);`

Trying this
`u8g2.begin(5, 4, 2, U8X8_PIN_NONE, U8X8_PIN_NONE,  U8X8_PIN_NONE);`
and this
`u8g2.begin();`
and this
`u8g2.begin(U8X8_PIN_NONE, U8X8_PIN_NONE, U8X8_PIN_NONE, U8X8_PIN_NONE, U8X8_PIN_NONE,  U8X8_PIN_NONE);`

ctx: cont 
sp: 3ffef960 end: 3ffefbc0 offset: 01b0

>>>stack>>>
0x40201e0f: u8g2_UserInterfaceSelectionList at D:\arduino\projects\libraries\U8g2\src\clib/u8g2_selection_list.c line 268
0x40203a3c: loop at D:\arduino\projects\U8g2lib_menu\SelectionList/SelectionList.ino line 108
0x402039ee: setup at D:\arduino\projects\U8g2lib_menu\SelectionList/SelectionList.ino line 86
0x40203f74: loop_wrapper at C:\Users\1\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2\cores\esp8266/core_esp8266_main.cpp line 57
0x40100108: cont_norm at C:\Users\1\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2\cores\esp8266/cont.S line 109  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v0c897c37
~ld

olikraus, 2018-03-24:
I am unable to support here. Can we close the issue?

microdimmer, 2018-03-25:
Ok, some of examples didnt working on esp8266, on arduino uno working good.

rosarior, 2018-05-28:
This is caused by SelectionList (and userInterfaceSelectionList) being blocking operations. 

> This is one of the most critical differences between the ESP8266 and a more classical Arduino microcontroller. The ESP8266 runs a lot of utility functions in the background – keeping WiFi connected, managing the TCP/IP stack, and performing other duties. Blocking these functions from running can cause the ESP8266 to crash and reset itself. To avoid these mysterious resets, avoid long, blocking loops in your sketch.

https://learn.sparkfun.com/tutorials/esp8266-thing-hookup-guide/using-the-arduino-addon

Since the GUI methods block the main loop, the watchdog timer reaches the end of its count and resets the microcontroller. 

Until the GUI methods are updated to be non-blocking they will not work on an ESP8266 microcontroller.

rosarior, 2018-05-28:
Another solution could be to add a `yield()` statement inside the `for (;;)` loops allowing the ESP8266 cooperative multitasking to continue handling other tasks. 

For compatibility with other microcontrollers (although I think newer versions already have this conditional)

    #if not defined(ESP8266)
        #define yield()
    #endif

olikraus, 2018-05-29:
yield() should be called. I think this is not ESP8266 specific: https://www.arduino.cc/en/Reference/SchedulerYield

olikraus, 2018-08-04:
maybe implement this together with #323

enlinea777, 2018-08-16:
I was trying to implement U8g2, and I found the typical WDT error.
So, implement a few
yield ():
in calls to generate output buffers and others while and for
with what I got that the screens worked in the ESP8266 without it being blocked by such a long loop.

I hope you implement it.

-Note: I do not know how to collaborate but I made a fork of the project and it works very well with esp8266
[image]

microdimmer, 2018-08-16:
Thank you for your project!

olikraus, 2018-08-23:
added to the Arduino gpio and delay procedure

closed 2018-08-23 by olikraus