olikraus/u8g2 #531: Can multiple instances share a buffer?

pastaclub, 2018-03-02:
I want to use multiple displays with the library. The displays have different controller chips, so I need to use different constructors. I can simply declare two instances of u8g2, but that will use up a lot of memory.

The way I see it, the page buffer is only used during drawing, so as long as I only draw on one display at a time, there is no need to have multiple buffers. Is there any way to tell the library to share the same buffer between two instances?

olikraus, 2018-03-03:
Background: The C++ constructor will call the C setup procedure in this file: https://github.com/olikraus/u8g2/blob/master/csrc/u8g2_d_setup.c
The setup procedure will call the memory allocation procedure from here: https://github.com/olikraus/u8g2/blob/master/csrc/u8g2_d_memory.c

You could modify u8g2 in such a way, that the smaller display uses the buffer of the larger display.

However, in your case if the resolution is the same, then they already share the buffer. We could crosscheck this, once you post the C++ constructor for both displays.

pastaclub, 2018-03-05:
I am instantiating them from an .ino file:

```
U8G2_SSD1306_128X64_NONAME_1_HW_I2C oled(U8G2_R0);
U8G2_SSD1309_128X64_NONAME2_1_HW_I2C oled2(U8G2_R0);
```

The second one uses up 223 additional bytes of global variables RAM, so it seems they are not sharing the buffer. You wrote "C++ constructor" - is it going to make a difference if I instantiate them from a .cpp file instead of an .ino file?

(I am not sure yet whether the 2nd constructor is the correct one since that display will still take a few weeks to arrive in the mail, but I'm sure they have the same resolution).

olikraus, 2018-03-05:
It should be
```
U8G2_SSD1306_128X64_NONAME_1_HW_I2C oled(U8G2_R0);
U8G2_SSD1309_128X64_NONAME_1_HW_I2C oled2(U8G2_R0)
```

.ino files are cpp files, there is no difference. Each display requires space for the local variables (this are the 223 bytes) and some space for the frame buffer page. This frame buffer page can be shared, but the 223 bytes for the object will always be there. For example it contains the I2C address, which should be different in your case.

pastaclub, 2018-03-06:
First of all, thanks for your fantastic support!

I'm already using multiple displays of the _same_ kind (even same I2C address, thorugh a PCA9548 I2C-Mux). For that I only use one constructor and it works very well. I simply render one display at a time, and in the page loop I draw the content for the respective display. The library doesn't even know that there is more than one.

I expected that I could also add a different display and merely the code behind the scenes would differ. It's sad to hear that this uses so much additional memory. While 223 bytes sounds tiny these days, that's another 10% of the total RAM in an ATmega328. I'm wondering what exactly that much memory is used for (the I2C address is one byte, the configuration of pins and protocols to use should also fit into a few bytes).

Do you see any way to have just one instance of the class and somehow be able to switch between SSD1306 and SSD1309 code (e.g. by overwriting a few member variables)?

olikraus, 2018-03-06:
> (the I2C address is one byte, the configuration of pins and protocols to use should also fit into a few bytes).

sure... There are multiple issues. For example the display information struction (which is static) is copied from flash to RAM for ATmega systes. Other systems (ARM) leave this in Flash and access this directly. Of course I could do the same, but this would require extra effort only for ATmega.

Structs are here: https://github.com/olikraus/u8g2/blob/master/csrc/u8x8.h#L316
and here: https://github.com/olikraus/u8g2/blob/master/csrc/u8g2.h#L284

For sure, reuse is thinkable, but it will add extra restrictions to the users of U8g2.  Others could be shared without problem, like the font decode temp data.

closed 2018-03-24 by olikraus