olikraus/u8g2 #532: [Font request] new font for persian chars

IMAN4K, 2018-03-03:
It would be much appreciated if we could have one of the standard `Farsi` font as `8x8`:

* [Samim.zip](https://github.com/olikraus/u8g2/files/1777660/samim.zip)

IMAN4K, 2018-03-03:
I've created two font type with options:
```
./otf2bdf -r 72 -p 8 Samim_8.ttf -o Samim_8.bdf
./bdfconv -f 1 -m '32-255' -n GanjNameh_8 -o GanjNameh_8.c GanjNameh_8.bdf
```
Results:
```
const uint8_t Samim_8[1170] =
"\177\0\2\2\3\4\3\4\4\10\12\377\376\7\376\7\377\1&\2h\4y \5\0\242\2!\7\271\242"
"\62T\0\42\6\222\366\62\2#\14=b\253\201\32\252\32\252\224\0$\11\273b\227\206\222j\10%\15"
">\342\27S\227L#\231IE\1&\17>\242\247Q\15RF\31%\225A\134\1'\5\21\266\42("
"\10B\236\226T\63\5)\10B\236\22S\227\2*\10$.\233F\214\3+\11$f\233F\314 \1"
",\5\21\236\42-\5\12\352\42.\5\211\242\22/\11;\342Z\305*F\0\60\12<b'\223\64S"
"\245\0\61\6\272b\267z\62\13<b\247b\6\31\304r\4\63\14<b\67b\6I\243\14\206\0\64"
"\13<b\17\242j\311\241A\0\65\11\273b\63b\234q\1\66\12<b\253b\6C\312\64\67\15<"
"b\63\62\310 f\220A\214\0\70\12<b\67b\252T\224i\71\12<b\247\242\231\66H\12:\6"
"\241\242\22\3;\10\62\236\226\211\224\0<\10$f\17\222\215\4=\10\34j\63\62\32\1>\11$f"
"\243Q\6#\2?\11;\342\62[\31&\0@\17G\336\273A\6\61Y\225\252\254\230\251\11A\15="
"b\233Q\6)\246\270\62\320 B\13\274b\263\242\34)\312\221\0C\14=b\273\62\310(\243\14\63"
"\34D\11\274\242\263\242\347H\0E\11\273\42\63b\34\61\16F\11\273\42\63b\34\261\4G\14=\242"
"\273\62\310(n\220b\34H\11\274\242\23\315\61\315\0I\6\271\242r\4J\7\302\235\326\327\0K\12"
"\274b\23U\222f*\3L\7\273\42\23{\16M\15\275\342\223\201\6k\244\221VR\5N\11\274\242"
"\23\327R\247\14O\15>\242;\253\214\64\322\250\314`\1P\11\273b\63\222\65d\4Q\16F\236;"
"\253\214\64\322\250\314`\303\4R\12\274b\263\242J\253g\0S\14<\42\67\62\310H\243\14F\2T"
"\15<\42\63b\6\31d\220A\6\11U\10\274\242\23}\216\4V\16=b\223\201\254b\212)\203\214"
"\42\0W\16?\342\23\243\214VR\225)k\220\0X\14=b\23\253\250Q\6\32\244RY\15<\42"
"\23S\245A\6\31d\220\0Z\14<b\63\62\210\31\304\14\342\10[\6\311\332\362\0\134\12;\342\22"
"\63\210\65\210\1]\7J\332\242\372\32^\10$*\33e\222\1_\6\14\32\63\2`\5\21{#a"
"\11,\42\267A\32\61\15b\11\273b\23\343H\326\10c\12,\42\67\62\310 #\5d\12<b\17"
"\62H#\232ie\10,\42\247\342\330hf\11;\342\246\322\212-\0g\13<\32\67b\222\31\214\70"
"\4h\11\273b\23\343H\256\0i\7\271\242\222\206\0j\7I\232\222\306\10k\11\273\42\23\253\25U"
"\1l\6\271\242r\4m\13\256\342\63T\251\242\212*\6n\7\253b\63\222+o\10,b\67\242\231"
"\6p\11\273Z\63\222\65d\4q\12<Z\67\242\231\66\310 r\7\252\342\262Z\0s\10+\42\267"
"\201\34\1t\10\63\342\226V,\5u\7\253b\223\134#v\11,\42\23e\222\26\0w\12.\242\23"
"\223\225V\331\2x\12,\42\23\223\6Q\305\0y\13<\32\23e\222\32\304(\1z\10+\42\63S"
"L\3{\13K\332\232b\225A\254A\0|\6I\33\363\0}\13K\332\222A\254A\212\225\2~\6"
"\14j\63\2\240\5\0\242\2\241\7\271\232\222\206\0\242\11\273b\227F,W\2\243\13<b\267\62\210"
"\63\203\70\2\244\10$f\63Rk\10\245\13<b\23S\245A\134\63\1\246\7I\33\263\301\0\247\14"
"<\42\67\62R\61i\64\22\0\250\5\212z#\251\14>\342{\251\250\242\212i\305\5\252\6\42\352r"
"\10\253\7#\42\267T\22\254\7\24j\63\62\10\255\5\12\352\42\256\14>\342{\251\264\254\244\62\210\13"
"\257\6\14>\63\2\260\6\232\362\242\6\261\12,b\233F\314 \215\0\262\10#\352\242AJ\3\263\10"
"#\352\242AJ\3\264\5\21{#\265\11\273Z\223\134CF\0\266\10D^w\214t:\267\5\211\256"
"\22\270\6\22\232\222\4\271\6\42\352\242\32\272\7#\352V\251\12\273\6\242\42s\10\274\15>\242\17g"
"\227L\32\214\244\22\0\275\17>\242\217\222\312 e\60b\6)&\65\276\17>\242\17g\6)&\65"
"\64\30I%\0\277\11;\332\232a*\63\20\0\0\0";

const uint8_t GanjNameh_8[169] U8G2_FONT_SECTION("GanjNameh_8") =
"\26\0\2\2\3\3\1\2\4\5\6\0\377\0\0\5\377\0\0\0\0\0\220 \4@\23!\5i\225\25"
"(\7\362\64U\246\0)\7\362\24\325R\0*\7\134\231d%\1+\10e\333\70dF\21-\5\312"
"\27\1.\5I\225\0/\7k\327\24\253\10:\5Y\225\24=\6\325\233G\0[\6\362\224\325\22\134"
"\10k\227\14b\15\2]\6\362\24\325\32{\11\363\30\231A\212I\1|\5i\23%}\10\363\70\225"
"d\15\2\240\4@\23\253\7\134\271\244b\12\273\10\134\231\24\223J\0\327\10d\233\230\244\212\1\367\7"
"\333\271\264\22\0\0\0\0";
```
But nothing is shown on display!

olikraus, 2018-03-04:
You have to select the correct unicode range for arabic glyphs

```
./otf2bdf -r 72 -p 8 Samim_8.ttf -o Samim_8.bdf
./bdfconv -f 1 -m '32-128,$600-$6ff,$FB50-$FBB1,$FE70-$FEFF' -n Samim_8 -o Samim_8.c Samim_8.bdf
```

IMAN4K, 2018-03-07:
Here are the fonts data, thanks to @olikraus : [persian_fonts.zip](https://github.com/olikraus/u8g2/files/1788764/persian_fonts.zip)
Available fonts in `8pt`, `10pt`, `12pt`, `14pt` and `16pt`:
* Samim & Samim-FD ([LICENSE](https://github.com/rastikerdar/samim-font/blob/master/LICENSE))
* GanjNamehSans ([LICENSE](https://github.com/font-store/GanjnamehFont/blob/master/OFL.txt))
* IranianSans ([LICENSE](https://github.com/nekofar/iranian-sans-fontface/blob/master/LICENSE))

I would prefer `Samim-FD@16pt` for digits and `Samim@12pt` for scripts, so if possible, adding these two would be nice.
Preview:
[image]

olikraus, 2018-03-07:
reopen this for adding the requested fonts.

olikraus, 2018-04-14:
Samim: requires hinting and forced auto hinting. 8pt does not look good a at all. 
   Sizes: 10pt, 12pt, 14pt, 16pt
Samim-FD, same as Samim
GanjNamehSans-Regular: 8pt does not make sense, hinting on, but forced auto hinting should be off!
   Sizes: 10pt, 12pt, 14pt, 16pt
IranianSans: ok, 8pt looks almost ok. hinting on, forced auto hinting has no influence
   Sizes: 8pt, 10pt, 12pt, 14pt, 16pt

olikraus, 2018-04-14:
done: https://github.com/olikraus/u8g2/wiki/fntgrppersian

closed 2018-04-14 by olikraus

ahmadelwaly2016, 2020-06-11:
how i can use those fonts with **20 pt** or **24 pt** ?

olikraus, 2020-06-11:
You can't. 
The size of the fonts are fixed because u8g2 only supports fixed bitmap fonts.
Do not confuse embedded system programming with windows / linux systems where you have enough resources and computing power to scale fonts on the fly.

You can, however provide a 20pt or 24pt font in bdf format. Such a pre scaled font can be included without problems.

ahmadelwaly2016, 2020-06-12:
Can we add this fonts to the library http://www.fonts-schriftarten.de/download-7021-arabicnaskhssk.zip 
with 18 pt 24 pt  36 pt  48 pt and 60 pt ?

ahmadelwaly2016, 2020-06-12:
and make it supported by the library like those fonts (Samim-FD, same as Samim, GanjNamehSans-Regular,IranianSans) it will be a greatful from you , thanks oliver.

olikraus, 2020-06-13:
There are two github issues. Let's continue the discussion here: #1220

> Can we add this fonts to the library http://www.fonts-schriftarten.de/download-7021-arabicnaskhssk.zip
with 18 pt 24 pt 36 pt 48 pt and 60 pt ?

Yes, but please provide bdf files as described in #1220