olikraus/u8g2 #533: STM2F10X MCU is not working with T6963C LCD through 8080 interface

lwFlyFish, 2018-03-08:
1、GPIO and Delay
uint8_t u8x8_gpio_and_delay_stm32(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
{
    GPIO_InitTypeDef    GPIO_Initstrutrue;

    switch (msg)
    {
    case U8X8_MSG_GPIO_AND_DELAY_INIT:	// called once during init phase of u8g2/u8x8
        RCC_APB2PeriphClockCmd(RCC_APB2Periph_GPIOG, ENABLE);               // CS
        GPIO_Initstrutrue.GPIO_Pin = GPIO_Pin_12;
        GPIO_Initstrutrue.GPIO_Mode = GPIO_Mode_Out_PP;
        GPIO_Initstrutrue.GPIO_Speed = GPIO_Speed_50MHz;
        GPIO_Init(GPIOG, &GPIO_Initstrutrue);

        RCC_APB2PeriphClockCmd(RCC_APB2Periph_GPIOF, ENABLE);               // RS
        GPIO_Initstrutrue.GPIO_Pin = GPIO_Pin_0;
        GPIO_Initstrutrue.GPIO_Mode = GPIO_Mode_Out_PP;
        GPIO_Initstrutrue.GPIO_Speed = GPIO_Speed_50MHz;
        GPIO_Init(GPIOF, &GPIO_Initstrutrue);

        RCC_APB2PeriphClockCmd(RCC_APB2Periph_GPIOD, ENABLE);               // WR/SCL
        GPIO_Initstrutrue.GPIO_Pin = GPIO_Pin_5;
        GPIO_Initstrutrue.GPIO_Mode = GPIO_Mode_Out_PP;
        GPIO_Initstrutrue.GPIO_Speed = GPIO_Speed_50MHz;
        GPIO_Init(GPIOD, &GPIO_Initstrutrue);

        RCC_APB2PeriphClockCmd(RCC_APB2Periph_GPIOD, ENABLE);               // RD
        GPIO_Initstrutrue.GPIO_Pin = GPIO_Pin_4;
        GPIO_Initstrutrue.GPIO_Mode = GPIO_Mode_Out_PP;
        GPIO_Initstrutrue.GPIO_Speed = GPIO_Speed_50MHz;
        GPIO_Init(GPIOD, &GPIO_Initstrutrue);

        RCC_APB2PeriphClockCmd(RCC_APB2Periph_GPIOA, ENABLE);               // LCD_Backlight
        GPIO_Initstrutrue.GPIO_Pin = GPIO_Pin_8;
        GPIO_Initstrutrue.GPIO_Mode = GPIO_Mode_Out_PP;
        GPIO_Initstrutrue.GPIO_Speed = GPIO_Speed_50MHz;
        GPIO_Init(GPIOA, &GPIO_Initstrutrue);

        RCC_APB2PeriphClockCmd(RCC_APB2Periph_GPIOD, ENABLE);               // D0
        GPIO_Initstrutrue.GPIO_Pin = GPIO_Pin_14;
        GPIO_Initstrutrue.GPIO_Mode = GPIO_Mode_Out_PP;
        GPIO_Initstrutrue.GPIO_Speed = GPIO_Speed_50MHz;
        GPIO_Init(GPIOD, &GPIO_Initstrutrue);

        RCC_APB2PeriphClockCmd(RCC_APB2Periph_GPIOD, ENABLE);               // D1
        GPIO_Initstrutrue.GPIO_Pin = GPIO_Pin_15;
        GPIO_Initstrutrue.GPIO_Mode = GPIO_Mode_Out_PP;
        GPIO_Initstrutrue.GPIO_Speed = GPIO_Speed_50MHz;
        GPIO_Init(GPIOD, &GPIO_Initstrutrue);

        RCC_APB2PeriphClockCmd(RCC_APB2Periph_GPIOD, ENABLE);               // D2
        GPIO_Initstrutrue.GPIO_Pin = GPIO_Pin_0;
        GPIO_Initstrutrue.GPIO_Mode = GPIO_Mode_Out_PP;
        GPIO_Initstrutrue.GPIO_Speed = GPIO_Speed_50MHz;
        GPIO_Init(GPIOD, &GPIO_Initstrutrue);

        RCC_APB2PeriphClockCmd(RCC_APB2Periph_GPIOD, ENABLE);               // D3
        GPIO_Initstrutrue.GPIO_Pin = GPIO_Pin_1;
        GPIO_Initstrutrue.GPIO_Mode = GPIO_Mode_Out_PP;
        GPIO_Initstrutrue.GPIO_Speed = GPIO_Speed_50MHz;
        GPIO_Init(GPIOD, &GPIO_Initstrutrue);

        RCC_APB2PeriphClockCmd(RCC_APB2Periph_GPIOE, ENABLE);               // D4
        GPIO_Initstrutrue.GPIO_Pin = GPIO_Pin_7;
        GPIO_Initstrutrue.GPIO_Mode = GPIO_Mode_Out_PP;
        GPIO_Initstrutrue.GPIO_Speed = GPIO_Speed_50MHz;
        GPIO_Init(GPIOE, &GPIO_Initstrutrue);

        RCC_APB2PeriphClockCmd(RCC_APB2Periph_GPIOE, ENABLE);               // D5
        GPIO_Initstrutrue.GPIO_Pin = GPIO_Pin_8;
        GPIO_Initstrutrue.GPIO_Mode = GPIO_Mode_Out_PP;
        GPIO_Initstrutrue.GPIO_Speed = GPIO_Speed_50MHz;
        GPIO_Init(GPIOE, &GPIO_Initstrutrue);

        RCC_APB2PeriphClockCmd(RCC_APB2Periph_GPIOE, ENABLE);               // D6
        GPIO_Initstrutrue.GPIO_Pin = GPIO_Pin_9;
        GPIO_Initstrutrue.GPIO_Mode = GPIO_Mode_Out_PP;
        GPIO_Initstrutrue.GPIO_Speed = GPIO_Speed_50MHz;
        GPIO_Init(GPIOE, &GPIO_Initstrutrue);

        RCC_APB2PeriphClockCmd(RCC_APB2Periph_GPIOE, ENABLE);               // D7
        GPIO_Initstrutrue.GPIO_Pin = GPIO_Pin_10;
        GPIO_Initstrutrue.GPIO_Mode = GPIO_Mode_Out_PP;
        GPIO_Initstrutrue.GPIO_Speed = GPIO_Speed_50MHz;
        GPIO_Init(GPIOE, &GPIO_Initstrutrue);

        break;							// can be used to setup pins
    case U8X8_MSG_DELAY_NANO:			// delay arg_int * 1 nano second
        bsp_DelayUS(arg_int);
        break;

    case U8X8_MSG_DELAY_100NANO:		// delay arg_int * 100 nano seconds
        bsp_DelayUS(100 * arg_int);
        break;

    case U8X8_MSG_DELAY_10MICRO:		// delay arg_int * 10 micro seconds
        bsp_DelayMS(10 * arg_int);
        break;

    case U8X8_MSG_DELAY_MILLI:			// delay arg_int * 1 milli second
        bsp_DelayMS(arg_int);
        break;

    case U8X8_MSG_DELAY_I2C:				// arg_int is the I2C speed in 100KHz, e.g. 4 = 400 KHz
        break;							// arg_int=1: delay by 5us, arg_int = 4: delay by 1.25us
    case U8X8_MSG_GPIO_D0:				// D0 or SPI clock pin: Output level in arg_int
                                        //case U8X8_MSG_GPIO_SPI_CLOCK:
        if (arg_int) { GPIO_SetBits(GPIOD, GPIO_Pin_14); }
        else { GPIO_ResetBits(GPIOD, GPIO_Pin_14); }
        break;

    case U8X8_MSG_GPIO_D1:				// D1 or SPI data pin: Output level in arg_int
                                        //case U8X8_MSG_GPIO_SPI_DATA:
        if (arg_int) { GPIO_SetBits(GPIOD, GPIO_Pin_15); }
        else { GPIO_ResetBits(GPIOD, GPIO_Pin_15); }
        break;

    case U8X8_MSG_GPIO_D2:				// D2 pin: Output level in arg_int
        if (arg_int) { GPIO_SetBits(GPIOD, GPIO_Pin_0); }
        else { GPIO_ResetBits(GPIOD, GPIO_Pin_0); }
        break;

    case U8X8_MSG_GPIO_D3:				// D3 pin: Output level in arg_int
        if (arg_int) { GPIO_SetBits(GPIOD, GPIO_Pin_1); }
        else { GPIO_ResetBits(GPIOD, GPIO_Pin_1); }
        break;

    case U8X8_MSG_GPIO_D4:				// D4 pin: Output level in arg_int
        if (arg_int) { GPIO_SetBits(GPIOE, GPIO_Pin_7); }
        else { GPIO_ResetBits(GPIOE, GPIO_Pin_7); }
        break;

    case U8X8_MSG_GPIO_D5:				// D5 pin: Output level in arg_int
        if (arg_int) { GPIO_SetBits(GPIOE, GPIO_Pin_8); }
        else { GPIO_ResetBits(GPIOE, GPIO_Pin_8); }
        break;

    case U8X8_MSG_GPIO_D6:				// D6 pin: Output level in arg_int
        if (arg_int) { GPIO_SetBits(GPIOE, GPIO_Pin_9); }
        else { GPIO_ResetBits(GPIOE, GPIO_Pin_9); }
        break;

    case U8X8_MSG_GPIO_D7:				// D7 pin: Output level in arg_int
        if (arg_int) { GPIO_SetBits(GPIOE, GPIO_Pin_10); }
        else { GPIO_ResetBits(GPIOE, GPIO_Pin_10); }
        break;

    case U8X8_MSG_GPIO_E:				// E/WR pin: Output level in arg_int
        if (arg_int) { GPIO_SetBits(GPIOD, GPIO_Pin_5); }
        else { GPIO_ResetBits(GPIOD, GPIO_Pin_5); }
        break;

    case U8X8_MSG_GPIO_CS:				// CS (chip select) pin: Output level in arg_int
        if (arg_int) { GPIO_SetBits(GPIOG, GPIO_Pin_12); }
        else { GPIO_ResetBits(GPIOG, GPIO_Pin_12); }
        break;

    case U8X8_MSG_GPIO_DC:				// DC (data/cmd, A0, register select) pin: Output level in arg_int
        if (arg_int) { GPIO_SetBits(GPIOF, GPIO_Pin_0); }
        else { GPIO_ResetBits(GPIOF, GPIO_Pin_0); }
        break;

    case U8X8_MSG_GPIO_RESET:			// Reset pin: Output level in arg_int
        break;
    case U8X8_MSG_GPIO_CS1:				// CS1 (chip select) pin: Output level in arg_int
        break;
    case U8X8_MSG_GPIO_CS2:				// CS2 (chip select) pin: Output level in arg_int
        break;
    case U8X8_MSG_GPIO_I2C_CLOCK:		// arg_int=0: Output low at I2C clock pin
        break;							// arg_int=1: Input dir with pullup high for I2C clock pin
    case U8X8_MSG_GPIO_I2C_DATA:			// arg_int=0: Output low at I2C data pin
        break;							// arg_int=1: Input dir with pullup high for I2C data pin
    case U8X8_MSG_GPIO_MENU_SELECT:
        u8x8_SetGPIOResult(u8x8, /* get menu select pin state */ 0);
        break;
    case U8X8_MSG_GPIO_MENU_NEXT:
        u8x8_SetGPIOResult(u8x8, /* get menu next pin state */ 0);
        break;
    case U8X8_MSG_GPIO_MENU_PREV:
        u8x8_SetGPIOResult(u8x8, /* get menu prev pin state */ 0);
        break;
    case U8X8_MSG_GPIO_MENU_HOME:
        u8x8_SetGPIOResult(u8x8, /* get menu home pin state */ 0);
        break;
    default:
        u8x8_SetGPIOResult(u8x8, 1);			// default return value
        break;
    }
    return 1;
}

2、
u8g2_Setup_t6963_240x128_1(&u8g2, U8G2_R0, u8x8_byte_8bit_8080mode, u8x8_gpio_and_delay_stm32);

void main(void)
{
    u8g2_InitDisplay(&u8g2);
    u8g2_SetPowerSave(&u8g2, 0); // wake up display
    u8g2_ClearBuffer(&u8g2);
    u8g2_SetFontMode(&u8g2, 1);
    u8g2_SetFontDirection(&u8g2, 0);
    u8g2_SetFont(&u8g2, u8g2_font_helvB18_te);
    u8g2_DrawStr(&u8g2, 0, 24, "hello world");

    while(1);
}

3、There is no display on the screen。

olikraus, 2018-03-08:
you must encapsulate the drawing commands into the picture loop:
```
u8g2_InitDisplay(&u8g2);
u8g2_SetPowerSave(&u8g2, 0); // wake up display
u8g2_FirstPage(&u8g2)
do
{
  u8g2_SetFontMode(&u8g2, 1);
  u8g2_SetFontDirection(&u8g2, 0);
  u8g2_SetFont(&u8g2, u8g2_font_helvB18_te);
  u8g2_DrawStr(&u8g2, 0, 24, "hello world");
} while( u8g2_NextPage(&u8g2) );
```

as an alternative (more RAM consuming), you can work with a full buffer (but then you need a different constructor call):
```
u8g2_Setup_t6963_240x128_f(&u8g2, U8G2_R0, u8x8_byte_8bit_8080mode, u8x8_gpio_and_delay_stm32);
```

And of course you need to send the data to the display (which was also missing in your code)
```
u8g2_InitDisplay(&u8g2);
u8g2_SetPowerSave(&u8g2, 0); // wake up display
u8g2_ClearBuffer(&u8g2);
u8g2_SetFontMode(&u8g2, 1);
u8g2_SetFontDirection(&u8g2, 0);
u8g2_SetFont(&u8g2, u8g2_font_helvB18_te);
u8g2_DrawStr(&u8g2, 0, 24, "hello world");
u8g2_SendBuffer(&u8g2);    // transfer data from uC memory to display 
```

lwFlyFish, 2018-03-08:
Very Thank you!

closed 2018-03-24 by olikraus