olikraus/u8g2 #538: [Question] u8g2 with MSP430F5529

Multichiller, 2018-03-12:
Hey :) 
I set up the function for my DOGXL240-7 with an Arduino, but I have to use a Texas Instruments MSP430F5529. I already adapted the code execpt the fourth arguments of the setup procedure. 
Is there a example for a Texas Instruments controller which I can use to adapt? Because I have no idea how to do it. (I already read the guide.)

- 4-wire SPI
- C-code
- Displaycontroller: UC1611s

Thank you!

olikraus, 2018-03-12:
You read https://github.com/olikraus/u8g2/wiki/Porting-to-new-MCU-platform?

As an intermediate solution you could use the existing "u8x8_byte_4wire_sw_spi" procedure as 4th argument. Once this works you can replace this procedure with a faster version which used SPI subsystem of your controller.

Multichiller, 2018-03-15:
Sorry, I am not very familiar with C-Programming...
How can I define my pins (MOSI, CS, .....), if I use "u8x8_byte_4wire_sw_spi" as fourth argument?
Does this mean that I can use the setup procedure like this?
```
#include "u8g2.h"

u8g2_t u8g2;
u8g2_Setup_uc1611_ea_dogxl240_1(&u8g2, U8G2_R0, u8x8_byte_4wire_sw_spi, u8x8_byte_4wire_sw_spi);
u8g2_InitDisplay(&u8g2); 
u8g2_SetPowerSave(&u8g2, 0); 

void draw(u8g2_t *u8g2)
{
    u8g2_SetFontMode(u8g2, 1);	
    u8g2_SetFontDirection(u8g2, 0);
    u8g2_SetFont(u8g2, u8g2_font_6x12_tr);
    u8g2_DrawStr(u8g2, 0, 20, "Hello World");
}

```
With the Arduino I always pictured a variabel like this:
```
 u8g2.setFont(u8g2_font_crox5hb_tf); //battery in %
 u8g2.setCursor(170, 25);
 u8g2.print(battery); 
```
Is the print function is a function from the arduino software or can I use them aswell to draw a variabel?
Thank you  :)

olikraus, 2018-03-15:
1) You need to write a gpio callback, where you set the pins
2) Probably you will have the c interface only. This means the Arduino print is not available for your.

closed 2018-03-24 by olikraus