olikraus/u8g2 #539: vertical line misaligned in SSD1322 on Arduino Due full-buffer mode.

tinyboxxx, 2018-03-14:
thanks for your time.

[image]
[image]
[image]
[image]

I use U8G2_SSD1322_NHD_256X64_F_4W_HW_SPI.

It seems is a problem with the width of the screen, it will move the left line to the end of the previous row.

FYI, if I run in page buffer mode, last few pixels will be lost. this is what I run fps, you can see right part is blank.
[image]

olikraus, 2018-03-14:
Did you activate 16 bit mode?

tinyboxxx, 2018-03-14:
Thanks for your quick reply.
I didn't realize that a comment is an option, It works great now!
[image]

Love your work! appreciated!

closed 2018-03-14 by tinyboxxx

olikraus, 2018-03-15:
Looks good!