olikraus/u8g2 #540: Use different buffers for 2 similar displays?

rnair510, 2018-03-15:
Is it possible to use different buffers for 2 equivalent displays? I am running a couple of ST7565 of an ArduinoMega and for various external reasons I need to draw multiple things into each displays buffer and then send them out individually. Since each display needs to show different content I'd like to not have to share a buffer but dont see any option to skip buffer sharing.

olikraus, 2018-03-15:
You have to change the library. Let me assume, that you use the dogm128 display in full buffer mode, then the following function in u8g2_d_setup.c is called:
```
void u8g2_Setup_st7565_ea_dogm128_f(u8g2_t *u8g2, const u8g2_cb_t *rotation, u8x8_msg_cb byte_cb, u8x8_msg_cb gpio_and_delay_cb)
{
  uint8_t tile_buf_height;
  uint8_t *buf;
  u8g2_SetupDisplay(u8g2, u8x8_d_st7565_ea_dogm128, u8x8_cad_001, byte_cb, gpio_and_delay_cb);
  buf = u8g2_m_16_8_f(&tile_buf_height);
  u8g2_SetupBuffer(u8g2, buf, tile_buf_height, u8g2_ll_hvline_vertical_top_lsb, rotation);
}
```

You can see that u8g2_m_16_8_f is called, which allocates the buffer. Let me assume you have four DOGM128 display, then you could modify this prozedure in the following way:

```
void u8g2_Setup_st7565_ea_dogm128_f(u8g2_t *u8g2, const u8g2_cb_t *rotation, u8x8_msg_cb byte_cb, u8x8_msg_cb gpio_and_delay_cb)
{
  uint8_t tile_buf_height = 8;
  static uint8_t buf[1024 * 4];
  static uint8_t cnt = 0;
  u8g2_SetupDisplay(u8g2, u8x8_d_st7565_ea_dogm128, u8x8_cad_001, byte_cb, gpio_and_delay_cb);
  u8g2_SetupBuffer(u8g2, buf+1024*(uint16_t)cnt, tile_buf_height, u8g2_ll_hvline_vertical_top_lsb, rotation);
  cnt++;
}
```

if have not tested the modification....

rnair510, 2018-03-15:
Thanks for the fast response. With the potential of affecting other displays down the road I've moved my drawing logic up the stack and buffered them elsewhere before individually refreshing each display. Thanks for maintaining such a great library.

closed 2018-04-15 by olikraus