olikraus/u8g2 #542: Bitmap cropped when using two displays

kaichan, 2018-03-16:
Thanks for maintaining this great library.
I tried to use two SSD1306 displays to display bitmaps.

```
U8G2_SSD1306_64X48_ER_1_4W_SW_SPI u8g2b(U8G2_R0, 10 , 9 ,  12, 11, 13);
U8G2_SSD1306_64X48_ER_1_4W_SW_SPI u8g2(U8G2_R0, 3 , 5 , 6, 7,  8);  
```

it works but with a tiny problem which there are 8 lines of pixels missing at the bottom of the 2nd display.
drawXBMP() is being used.

Any suggestion is welcomed.
[image]

olikraus, 2018-03-16:
difficult without knowing your code. Can you post the part of your code which generates these pictures?

My assumption is: You use a ```_1_``` constructor together with the pricture loop, but something went wrong in the while condition. Maybe during copy&paste you missed to exchange all  u8g2 to u8g2b

kaichan, 2018-03-17:
Thanks, i have simplified the code a bit and here it is

```
#include<Arduino.h>
#include<U8g2lib.h>
 
#ifdef U8X8_HAVE_HW_SPI
#include<SPI.h>
#endif
#ifdef U8X8_HAVE_HW_I2C
#include<Wire.h>
#endif

U8G2_SSD1306_64X48_ER_1_4W_SW_SPI u8g2b(U8G2_R0, /*grey d0 clock=*/10 , /*purple d1 data=*/9 , /*yellow  cs=*/ 12, /*green dc=*/ 11, /*blue reset=*/ 13);
U8G2_SSD1306_64X48_ER_1_4W_SW_SPI u8g2(U8G2_R0, /* grey d0 clock=*/3 , /*white d1 data=*/5 , /* red cs=*/ 6, /*orange dc=*/ 7, /*black reset=*/ 8);

#define eye_width 64
#define eye_height 48
static const unsigned char eye_bits[] PROGMEM = {
   0xff, ...... 0xff}; //removed the content for readability 

#define eye2_width 64
#define eye2_height 48
static const unsigned char eye_bits2[] PROGMEM = {
0xff, .......... 0xff, 0xff }; //removed the content for readability 
 
void draw(void) {
 int r = random(1,100);
 if (r%2 == 0)
  u8g2.drawXBMP( 0, 0, eye_width, eye_height, eye_bits2);
 else
  u8g2.drawXBMP( 0, 0, eye_width, eye_height, eye_bits);
}

void draw2(void) {  
 int r = random(1,100);
 if (r%2 == 0)
  u8g2b.drawXBMP( 0, 0, eye_width, eye_height, eye_bits2);
 else
  u8g2b.drawXBMP( 0, 0, eye_width, eye_height, eye_bits);
}
 
void setup(void) { 
 u8g2.begin();
 u8g2b.begin();
}
 
void loop(void) {
 u8g2.firstPage();
 u8g2b.firstPage();
 int i = 0;
 bool np = false;
 do {
  if (i % 2 == 0)
  {
    draw();
    np = u8g2.nextPage();
  }
  else
  {
    draw2();
    np = u8g2b.nextPage();
  }
  i++;
 } while( np );
 
 delay(1000);
}
```

olikraus, 2018-03-17:
I understand the idea, but it probably doesn't work because of the buffer sharing. Better execute the loops one after the other.

kaichan, 2018-03-17:
Thanks for the idea, sure it is about the buffer sharing. I fixed a bug in my code and it's now working.

closed 2018-04-15 by olikraus