olikraus/u8g2 #543: Displacement problem: x coordinates shifted to the right by 2 pixels

bstrobel, 2018-03-16:
Hello,

I use the u8g2 C lib on a STM32 Nucleo board with a STM32F411RE and a .96" 128x64 OLED with a SH1106 controller.

I use Atollic TrueStudio 9.0.0 which uses GCC as the compiler. The same problem appears also with the latest version of SW4STM32.

I checked out the master branch yesterday to be sure my problem is not already solved.

Now to my problem. This code:
```c
	u8g2_ClearBuffer(&u8g2);
	u8g2_DrawStr(&u8g2, 0, 12, "NUCLEO-64 F411RE");
	u8g2_DrawStr(&u8g2, 0, 24, u8x8_u8toa(SystemCoreClock / 1000000, 2));
	u8g2_DrawStr(&u8g2, 20, 24, "MHz");
//	u8g2_DrawHLine(&u8g2, 0, 0, 128);
//	u8g2_DrawHLine(&u8g2, 0, 63, 128);
//	u8g2_DrawVLine(&u8g2, 0, 0, 64);
//	u8g2_DrawVLine(&u8g2, 127, 0, 64);

	for (u8g2_uint_t px = 0; px < 128; px++)
	{
		u8g2_DrawPixel(&u8g2, px, 0);
		u8g2_DrawPixel(&u8g2, px, 63);
	}
	for (u8g2_uint_t py = 0; py < 64; py++)
	{
		u8g2_DrawPixel(&u8g2, 0, py);
		u8g2_DrawPixel(&u8g2, 127, py);
	}
	u8g2_SendBuffer(&u8g2);
```

leads to this display:

[image]

As you can see the 4 lines should exactly be on the edges of the display area. As this is true for the horizontal lines, in the picture you can see how the vertical lines are displaced by 2 pixels to the right. The horizontal lines also don't start on the very left but 2 pixels to the right. There are also some garbage pixels on the very left.

The same problem appears if I use `u8g2_DrawHLine` and `u8g2_DrawVLine` instead of `u8g2_DrawPixel`.

Please note that strangely enough the text output seems not to be shifted to the right.

olikraus, 2018-03-16:
Probably this is not a SH1106, but instead a SSD1306 controller.

bstrobel, 2018-03-16:
Yes, that solved it! I only changed this line from that
```c
u8g2_Setup_sh1106_128x64_noname_f(&u8g2, U8G2_R0, u8g2_byte_4wire_hw_spi_stm32, u8g2_gpio_and_delay_stm32);
```
to this
```c
u8g2_Setup_ssd1306_128x64_noname_f(&u8g2, U8G2_R0, u8g2_byte_4wire_hw_spi_stm32, u8g2_gpio_and_delay_stm32);
```
and it works.

As a background: I bought a couple of SPI capable and another bunch of I2S capable displays from the same Ebay seller. He claimed the SPI displays would have a SH1106 and the I2S ones a SSD1306. Seems he was wrong... :)

Many thanks for the help and the lib. I really like your it. Great work!

closed 2018-03-16 by bstrobel

olikraus, 2018-03-16:
> He claimed the SPI displays would have a SH1106 and the I2S ones a SSD1306

Yes, I heard this from many users... :-(  
These seller do not know, what they sell...