olikraus/u8g2 #544: Nothing displayed on ST7920 LCD

TTTTTAN, 2018-03-20:
Hi :) I am using u8g2 lib in my UNO-based project, but my UNO doesn't display anything on the ST7920 LCD screen, however when running GraphicTest example, it turned out normal.

(Once I was building another project using ST7920 LCD and found that when i removed all C library functions such as `strcat` from the draw function, things could be normally displayed. Is my problem related to the standard library?)

Could anyone help solve my problem? Thx

My codes go as follow, written in C++: 
```
U8G2_ST7920_128X64_1_SW_SPI u8g2(U8G2_R0, LCD_E_PIN, LCD_R_W_PIN, LCD_RS_PIN, LCD_RST_PIN);

void loop {
	u8g2.firstPage();
	do {
		draw();
	} while (u8g2.nextPage());
}

void draw() {
	u8g2.setFont(u8g2_font_profont12_mr);

	u8g2.setDrawColor(1);

	String ts("Temperature: ");
	ts += tempr;
	ts += "C";
	u8g2.setCursor(2, 12); u8g2.print(ts);

	ts = "Humidity: ";
	ts += hum;
	ts += "%";
	u8g2.setCursor(2, 26); u8g2.print(ts);

	ts = "Soil Moisture: ";
	if (soilMois > 920) ts += "V.Dry";
	else if (soilMois > 500) ts += "Dry";
	else if (soilMois > 200) ts += "Med.";
	else if (soilMois > 80) ts += "Wet";
	else ts += "V.Wet";
	u8g2.setCursor(2, 40); u8g2.print(ts);

	u8g2.drawLine(2, 44, 126, 44);
	u8g2.setCursor(2, 58);
	switch (curSetting) {
	case SETTING_TEMPR_MAX:
		ts = "Max tempr: ";
		ts += temprMax;
		u8g2.print(ts);
		break;
	case SETTING_TEMPR_MIN:
		ts = "Min tempr: ";
		ts += temprMin;
		u8g2.print(ts);
		break;
	case SETTING_HUM_MIN:
		ts = "Min humdt: ";
		ts += humMin;
		u8g2.print(ts);
		break;
	case SETTING_SOIL_MOIS_MAX:
		ts = "Max soil mosi: ";
		ts += soilMoisMax;
		if (soilMois > 920) ts += "V.Dry";
		else if (soilMois > 500) ts += "Dryt";
		else if (soilMois > 200) ts += "Med.";
		else if (soilMois > 80) ts += "Wet";
		else ts += "V.Wet";
		u8g2.print(ts);
		break;
	}

	if (tempr > temprMax || tempr < temprMin || hum < humMin || rain > 960) {
		u8g2.setCursor(2, 58);
		uint8_t tc = u8g2.getDrawColor();
		u8g2.setDrawColor(1);
		u8g2.drawBox(2, 46, 124, 14);
		u8g2.setDrawColor(0);
		if (tempr > temprMax)u8g2.print("Attention! Hot!");
		else if (tempr < temprMin)u8g2.print("Attention! Cold!");
		else if (hum < humMin)u8g2.print("Attention! Dry!");
		else if (rain > 960) u8g2.print("Attention! Hvy Rain!");
		u8g2.setDrawColor(tc);
	}

	if (muted) {
		u8g2.setCursor(56, 12);
		u8g2.print("M");
	}
}
```

olikraus, 2018-03-20:
Code looks ok from my side.

TTTTTAN, 2018-03-20:
But the problem still exists however i edit it. By the way I am using an UNO-compatible [Seeeduino v4.2 ](http://wiki.seeed.cc/Seeeduino_v4.2/)instead of Arduino UNO,

olikraus, 2018-03-20:
Maybe something is wrong with "String ts"

TTTTTAN, 2018-03-20:
That makes sense. I'll try another method to set display strings, if the situation changed I'll inform you, thanks.

TTTTTAN, 2018-03-20:
@olikraus  what a **stupid and absurd** mistake i've made!!! I forgot `begin()` in the setup function! I'm soooooo sorry for disturbing you because of that. Anyway, thank you very much, u8g2 is an excellent library!

closed 2018-03-20 by TTTTTAN

olikraus, 2018-03-20:
No problem, Glad that things work.