olikraus/u8g2 #545: Open Iconic Font on Arduino?

Ellokah, 2018-03-21:
Hi there,

I was just wondering if the **open iconic fonts** are also available for **Arduino**? Im asking because I always get the compiler error: 

> error: 'u8g2_font_open_iconic_all_1x_t' was not declared in this scope

when using the command `u8g2.setFont(u8g2_font_open_iconic_all_1x_t);` in the Arduino Web-Editor. (I manually uploaded the recent .zip library files from [here](https://github.com/olikraus/U8g2_Arduino). In the Font reference it's stated that open iconic should be available by v.2.22 and higher.

What am I overlooking here? Thanks for your help.

Oliver

olikraus, 2018-03-21:
Yes, it is not yet released. Official release is 2.21. It is actually assigned to 2.22 release. But it is already implemented and I can create a beta release 2.22.something for you Need to reach my lab first...

olikraus, 2018-03-21:
I have created beta 2.22.4, which should include the iconic font:
https://github.com/olikraus/U8g2_Arduino

Ellokah, 2018-03-21:
Wow, Oli!

Thank you very much for your fast reply and the special treatment Beta upload. It works fine for me!

Sorry for my hasty inquiry. Didn't know that it was not released yet. I think I was a little bit confused because this Arduino lib was already labeled with "v2.22". Anyway... Thank you so much for your help. You got an email from me ;)

Kind regards,

Oliver

olikraus, 2018-03-21:
I think I was confused by myself (it is already late). The latest stable release is 2.21.8.
But I had already created some beta releases for 2.22. The first beta release which included the iconic font is 2.22.4, which I created for you.

closed 2018-04-15 by olikraus

sarwarislammoon, 2018-08-03:
Hi
I am new with GLCDs can u please tell me how can i print custom symbols like wifi in my lcd . I am using ST7565 lcd , i can write text on it using U8g2 , but i am very much confused with . 

after a lot of trial and error with U8g2font.c I found that following code print wifi  symbol , but is there any simple way to print it ?  or at least how can I  find strings corrospond with symbol . 

thanks in advance . 

code :
u8g2.setFont(u8g2_font_open_iconic_www_1x_t);
u8g2.setCursor(90,63);  
u8g2.print("Q\23\210\24");