olikraus/u8g2 #547: Support for WiseChip 0.42 OLED SH1106

feistyz, 2018-03-24:
Hi,

Would it be possible to add support for the [WiseChip 0.42 SH1106 display](http://www.wisechip.com.tw/en/product-549063/0-42-inch-OLED-Display.html)?

I can get it mostly working with the generic U8G2_SH1106_128X64_NONAME_F_4W_HW_SPI class except for having the coordinate system be in the right place. Hopefully, this would be a simple addition.

Thanks!

olikraus, 2018-03-24:
You have a datasheet? Picture of HelloWorld Example?
Are you sure it is SH1106?

feistyz, 2018-03-24:
Yes, it's the SH1106. Do you need the datasheet for the display or sh1106? The HelloWorld Example is off the screen and therefore doesn't show but the graphics test works. Here are some photos:

[image]
[image]
[image]

olikraus, 2018-03-24:
The datasheet of the display. It should explain the offset.

feistyz, 2018-03-24:
Here you go
[WiseChip 0.42.pdf](https://github.com/olikraus/u8g2/files/1844656/WiseChip.0.42.pdf)

olikraus, 2018-03-24:
ok, from page 2:
y: 12 to 51 (40)
x: 30 to 101 (72)

page 14: 
display offset 0x0d3 0x0c

feistyz, 2018-03-24:
That seems right based on what I am visually seeing. 

I am not as familiar with the library internals but can I plug these values into the 128x64 code and see if it works or do you need to generate a new set of classes?

olikraus, 2018-03-24:
I will generate a new class soon.

feistyz, 2018-03-24:
Thanks much!

olikraus, 2018-03-24:
new constructor will be: U8G2_SH1106_72X40_WISE_1_4W_HW_SPI

feistyz, 2018-03-24:
That was fast. Thanks!

So the width is correct but the height is wrong. See hello world below. Only 8 rows are visible.
[image]

olikraus, 2018-03-25:
Strange. Can you send the complete code for this example?
What about other examples? Can you test u8x8 example? Will it work?

feistyz, 2018-03-25:
I take it back. My mistake. I was running full buffer examples instead of page examples. When I run the page examples Hello World. Here is what it and the Flip example look like:

[image]
[image]
[image]
[image]
[image]

feistyz, 2018-03-25:
BTW, Is it possible to have it be a full buffer?

Here is a video of the flip app. Might be easier to see what it is doing. 
[VIDEO0452.zip](https://github.com/olikraus/u8g2/files/1845074/VIDEO0452.zip)

olikraus, 2018-03-25:
> BTW, Is it possible to have it be a full buffer?

It should be there and working.

> Here is a video of the flip app. Might be easier to see what it is doing.

I will have a look, but I assume it is the y offset in flip mode 1 which is still not correct. 

I have created a new release for download: 2.22.7 
Please download from here: https://github.com/olikraus/U8g2_Arduino

If 2.22.7 is still incorrect, can you locate file u8x8_d_sh1106_72x40.c on your hard disk and modify the offset 24 to something else:
https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_sh1106_72x40.c#L102

What will be the correct value for your display?

olikraus, 2018-03-25:
After seening your video, maybe the offset should 0 instead of 24... Hope you can modify the code to figure out the correct offset in flip mode 1.

feistyz, 2018-03-26:
Sorry for the delay. 
2.22.7 with 24 didn't work. 
Zero for the offset didn't work either.

Trial and error yields either -12 or 52.

Here are the videos (names correspond to offsets):
[flip.zip](https://github.com/olikraus/u8g2/files/1846331/flip.zip)

olikraus, 2018-03-26:
Thanks, looks good. I have create beta 2.22.8 with value 52.

feistyz, 2018-03-27:
Thanks for adding the display!

olikraus, 2018-03-27:
No problem

closed 2018-03-27 by olikraus