olikraus/u8g2 #548: How to refresh the display of part of the LCD?

lwFlyFish, 2018-03-25:
I use a 240*128 LCD.And I draw a frame by "drawFrame(20, 20, 20, 20)"
I want to update only the frame , but I found that in the process of actual operation function will (0, 20) (0 40) (240) (240, 40) between the area clear.
And I can't update the data in the original location.
What should I do?

olikraus, 2018-03-25:
> I want to update only the frame , but I found that in the process of actual operation function will (0, 20) (0 40) (240) (240, 40) between the area clear.

I do not understand this sentence. What do you mean?

> And I can't update the data in the original location.

Why not?

> What should I do?

What exactly do you want to do?

lwFlyFish, 2018-03-25:
sorry,I need the  to display three rows.

u8g2.setFont(u8g2_font_ncenB14_tr);
u8g2.drawStr(0,15,"Hello World!");
u8g2.drawStr(50,15,"World");
u8g2.drawStr(100,15,"!");

u8g2.drawStr(0,50,"time:");
u8g2.drawStr(50,50,"2018-03-28");
u8g2.drawStr(100,50,"10:00：03");

u8g2.drawStr(0,100,"start!");

I want to change “2018“ to ”2019“.can I only refresh the 4 char?

olikraus, 2018-03-26:
Yes, but you must use full page mode.

Example: https://github.com/olikraus/u8g2/blob/master/sys/arduino/u8g2_full_buffer/FontUsage/FontUsage.ino

lwFlyFish, 2018-03-26:
Yes, I tried to write the program like this.
```
 u8g2_SetFont(&u8g2, u8g2_font_test_tf1);
u8g2_DrawUTF8(&u8g2, (120 - nWidth / 2), (65 + 99) / 2, "滑移系数检测仪");
u8g2_SendBuffer(&u8g2);

u8g2_DrawUTF8(&u8g2, (120 - nWidth1 / 2), (65 + 99) / 2, "系数滑移检测");
u8g2_SendBuffer(&u8g2);
```
Chinese has changed,But some points haven't been cleared. And refreshing the screen is very slow.

[image]

lwFlyFish, 2018-03-26:
sorry,Update the code
u8g2_SetFont(&u8g2, u8g2_font_test_tf1);
u8g2_DrawUTF8(&u8g2, 20, 80, "滑移系数检测仪");
u8g2_SendBuffer(&u8g2);

u8g2_DrawUTF8(&u8g2, 20, 80, "系数滑移检测");
u8g2_SendBuffer(&u8g2);

olikraus, 2018-03-26:
You need to clear the background first with the drawBox command. For speed up use font mode 1 (transparent mode).

lwFlyFish, 2018-03-27:
it is ok

closed 2018-04-15 by olikraus