olikraus/u8g2 #550: bdf decoder documentation

olikraus, 2018-03-26:
https://www.mikrocontroller.net/topic/443768

NikolaiPrzewalski, 2018-04-05:
[2018-04-05 U8G2_Doku.pdf](https://github.com/olikraus/u8g2/files/1880115/2018-04-05.U8G2_Doku.pdf)

Ich habe mir die Freiheit genommen, das Ergebnis unserer Diskussion (noch auf deutsch) zusammenzufassen.

olikraus, 2018-04-05:
Wow, nicht schlecht. Danke dafür. Sieht nach einer Abschlussarbeit aus ;-)

olikraus, 2018-04-05:
> Diese Art der Laulängenkodierung sieht nicht aus, als hätte sie sich jemand spontan ausgedacht.

Doch, hab ich...

olikraus, 2018-04-05:
> Die Datei u8g2_fonts.c blo nicht Eclipse oder EmBitz onen!

hehe, ich benutze "scite"... da gehts... Aber klar, die Datei ist ja automatisch erzeugt.

NikolaiPrzewalski, 2018-04-05:
Ich sehe: Ich kann das angehängte Dokument nachträglich ändern. Das ist praktisch. Dann halte ich es auf dem aktuellen Stand.
Eine Abschlussarbeit ist das nicht. Nur ein mittelgroßes Hobbyprojekt. Die U8G2-Schriften füllen dort einfach eine real vorhandene Lücke.

olikraus, 2018-04-05:
wow, ich bin beeindruckt. Gute Dokumentation für ein Hobbyprojekt.

NikolaiPrzewalski, 2018-04-06:
{| class="wikitable"
|-
! Name                                          !! Regierungszeit                       !! Abstammung
|-
| [[Stephan II. (Bayern)|Stephan&nbsp;II.]]     || 1347–1375 Herzog von Bayern          || Sohn Kaiser Ludwigs&nbsp;IV.
|-
| [[Friedrich (Bayern)|Friedrich]]              || 1375–1393 Herzog von Bayern-Landshut || Sohn Stephans&nbsp;II.
|-
| [[Heinrich XVI. (Bayern)|Heinrich&nbsp;XVI.]] || 1393–1450 Herzog von Bayern-Landshut || Sohn Friedrichs
|-
| [[Ludwig IX. (Bayern)|Ludwig&nbsp;IX.]]       || 1450–1479 Herzog von Bayern-Landshut || Sohn Heinrichs&nbsp;XVI.
|-
| [[Georg (Bayern)|Georg]]                      || 1479–1503 Herzog von Bayern-Landshut || Sohn Ludwigs&nbsp;IX.
|}

NikolaiPrzewalski, 2018-04-06:
{| 
|-
! Name                                          !! Regierungszeit                       !! Abstammung
|-
| [[Stephan II. (Bayern)|Stephan&nbsp;II.]]     || 1347–1375 Herzog von Bayern          || Sohn Kaiser Ludwigs&nbsp;IV.
|-
| [[Friedrich (Bayern)|Friedrich]]              || 1375–1393 Herzog von Bayern-Landshut || Sohn Stephans&nbsp;II.
|-
| [[Heinrich XVI. (Bayern)|Heinrich&nbsp;XVI.]] || 1393–1450 Herzog von Bayern-Landshut || Sohn Friedrichs
|-
| [[Ludwig IX. (Bayern)|Ludwig&nbsp;IX.]]       || 1450–1479 Herzog von Bayern-Landshut || Sohn Heinrichs&nbsp;XVI.
|-
| [[Georg (Bayern)|Georg]]                      || 1479–1503 Herzog von Bayern-Landshut || Sohn Ludwigs&nbsp;IX.
|}

NikolaiPrzewalski, 2018-04-06:
{| 
|-
! Name                                          !! Regierungszeit                       !! Abstammung
|-
| [[Stephan II. (Bayern)|Stephan&nbsp;II.]]     || 1347–1375 Herzog von Bayern          || Sohn Kaiser Ludwigs&nbsp;IV.
|-
| [[Friedrich (Bayern)|Friedrich]]              || 1375–1393 Herzog von Bayern-Landshut || Sohn Stephans&nbsp;II.
|-
|}

NikolaiPrzewalski, 2018-04-08:
Ich denke, die Beschreibung ist auf einem guten Stand. Ich schaue in den nächsten Tagen noch einmal nach, ob ein grober Schnitzer drin ist, dann kannst Du sie ins Wiki kopieren und damit machen, wozu Du Lust hast.

olikraus, 2018-04-08:
Hört sich gut an :-)

olikraus, 2018-04-08:
Ich hab Dich zum Projekt Mitarbeiter ernannt (nur für den Fall, dass Du selber etwas editieren willst) und mal angefangen eine Seite aufzusetzen:

https://github.com/olikraus/u8g2/wiki/u8g2fontformat

olikraus, 2018-04-09:
Woher bekommst du denn die "Glyph-Bounding-Box"?
Meinst du mit "Font-Bounding-Box" die vier werte aus dem Header?

NikolaiPrzewalski, 2018-04-09:
[image]

Mit "Glyph-Bounding-Box" meine ich "glyph bitmap width", "glyph bitmap height", "glyph bitmap x offset" und "glyph bitmap y offset". Zusammen genommen kann man sie ja als Rechteck, das gegenüber einem Punkt auf der Basislinie verschoben ist, interpretieren. Ich habe das mal für ein paar Zeichen skizziert.

Meine Erwartung war, daß die Byte 9 bis 12, die ja auch ein Rechteck beschreiben, das umhüllende Rechteck beschreiben, das sich ergibt, wenn man alle Zeichen übereinandergedruckt hätte.

Allerdings besteht da ein Unterschied im Y-Offset, der ungefähr dem Abstand der Basislinie von der Oberkante entspricht.

olikraus, 2018-04-09:
hmm... schwierige Frage. 
An sich ist das schon das "einhüllende" Rechteck. Das wird hier berechnet (über alle glyphs): https://github.com/olikraus/u8g2/blob/master/tools/font/bdfconv/bdf_font.c#L237

Man müsste sich mal ausgeben lassen, welches Zeichen den y Wert verursacht hatte (enc_y).

Die Berechnung der Einhüllenden findet hier statt: https://github.com/olikraus/u8g2/blob/master/tools/font/bdfconv/bdf_glyph.c#L376

NikolaiPrzewalski, 2018-04-09:
Ich habe mal zwei Beispiele nachgerechnet:

[Font_Outline.txt](https://github.com/olikraus/u8g2/files/1890254/Font_Outline.txt)

(Die Zeichen-Interpretationen in der Windows-Konsole stimmen nicht immer, aber der Zeichencode natürlich schon.)

Was mir spontan auffällt: Für die Offsets wird das Maximum genutzt. Spontan finde ich keine Stelle, wo das Vorzeichen gedreht wird. Wenn Höhe und Breite immer positiv sind, ist beim umhüllenden Rechteck zweier Rechtecke doch normalerweise das Minimum für den Offset verantwortlich?

    /* Umhuellende Box zweier Einzel-Boxen berechnen */
    static inline Rect_t rect_outline(Rect_t R1, Rect_t R2)
    {
        Rect_t R =
        {
            .x = MIN(R1.x, R2.x),
            .y = MIN(R1.y, R2.y),
            .w = MAX(R1.x + R1.w, R2.x + R2.w) - MIN(R1.x, R2.x),
            .h = MAX(R1.y + R1.h, R2.y + R2.h) - MIN(R1.y, R2.y),
        };
        return R;
    }

Was ich zumindest sehe:  Die Höhe wird bei den 8x8-Fonts offensichtlich manuell gesetzt, ist also wohl Absicht: https://github.com/olikraus/u8g2/blob/master/tools/font/bdfconv/bdf_font.c#L281

olikraus, 2018-04-09:
Anbei der output von bdfconv.

```
./bdfconv -v -f 1 -m '32-255' tmp.bdf -o fub30_tf.c  -n fub30_tf -d ../bdf/helvB18.bdf 
Parse File tmp.bdf: 243 glyph(s) found
Map: map_cmd_list='32-255'
Map: exclude=0 from=32/$20 to=255/$ff map=32/$20
Reduce: Start
Reduce: End
CalculateMaxBBX: x=-1, y=-11, w=41, h=54
CalculateMaxBBX: Encodings x=96, y=124, w=37, h=197
bf_CalculateMinMaxDWidth: dx_min=9, dx_max=42
bf_CalculateMinMaxDWidth: x_min=-1, x_max=10
bf_CalculateMaxBitFieldSize: bbx.x=5, bbx.y=6, bbx.w=6, bbx.h=6, dwidth=7
RLE Compress: best zero bits 5, one bits 4, total bit size 70640
RLE Compress: Font code generation, selected glyphs=189, total glyphs=243
RLE Compress: 'A' pos = 1378, 'a' pos = 2959
RLE Compress: Font size 8857
Draw TGA, line height 54
Draw TGA, xmax 833
bf_WriteU8G2CByFilename: Write file 'fub30_tf.c'
```

Entscheidend sind diese beiden Zeilen:

CalculateMaxBBX: x=-1, y=-11, w=41, h=54
CalculateMaxBBX: Encodings x=96, y=124, w=37, h=197

Das stimmt schon mal überein. Auch im font mode 2 (8x8) ändert sich da nicht viel.

Wenn wir uns mal den fub30_tf anschauen:
https://github.com/olikraus/u8g2/blob/master/tools/font/build/single_font_files/u8g2_font_fub30_tf.c

Das fragliche y offset ist: \365
Das ist in dezimal: 3*8*8+6*8+5 = 245 bzw 245-256 = -11

Entspricht also dem erwarteten Wert. Oder sehe ich da was falsch? Anders gesagt, wie kommst du auf -42?

olikraus, 2018-04-09:
> Entschuldige die Mühe.

Kein Problem

> Was bleibt, ist die Frage: Warum ist bei dem 8x8-Font die Glyph-Boundingbox größer als die Font-BoundingBox ? Das scheint ja eine bewußte Design-Entscheidung gewesen zu sein.

Auch eine gute Frage. Tatsache ist, dass damit die Fonts für das 8x8 Interface berechnet werden. Er muss also in das 8x8 Raster passen.

olikraus, 2018-04-15:
Nochmal vielen Dank für die Dokumentation. Die neue Wiki Seite ist nun auch auf der Wiki-Home Page verlinkt. Ich werde mal das Ticket hier schließen.

Danke!

closed 2018-04-15 by olikraus

RobotGB, 2018-09-18:
Hello Oliver and Nikolai. Thank you very much for the documentation.
I would like to know more about how the bits are arranged in the bitmap format.
For example, what does m0 and m1 mean?
_Hallo Oliver und Nikolai. Vielen Dank für die Dokumentation.
Ich würde gerne mehr darüber wissen, wie die Bits im Bitmap-Format angeordnet sind.
Zum Beispiel, was bedeutet m0 und m1?_

> Glyph bitmaps are 1-Bit horizontally packed bitmaps with tight fit bounding box (see image above). They are compressed by an ad-hoc run length algorithm. The bit array has
> m0 Bits (see font header) denoting the number of zeros
> m1 Bits (see font header) denoting the number of ones
> n Bits == 1 (to be counted) denoting the number of repetitions of the sequence and
> 1 Bit == 0 as stop marker for each sequence.
> Run lengths go beyond lines. Glyph bitmaps don't contain end markers, since their widths and heights are known.

I have an example glyph: the '!' character in Unifont.
After looking through the data the glyph data of '!' looks like this:
_Ich habe ein Beispielzeichen: das '!' Charakter in Unifont.
Nach dem Durchsehen der Daten die Glyphendaten von '!' sieht aus wie das:_
`0b00100001` = the unicode number (0x21) of the character
`0b00001110` ┐
`0b00001000` │
`0b00000110` │
`0b01111101` │
`0b01100100` │
`0b01110001` │
`0b11111001` │
`0b00011110` ├ Could you please explain how the compressed bitmap is arranged?
`0b00100000` │ Könntest du bitte erklären, wie die komprimierte Bitmap angeordnet ist?
`0b00100000` │
`0b00101110` │
`0b00011110` │
`0b00001000` │
`0b00000000` ┘

I hope Google Translate worked well. Thank you very much for your help.
_Ich hoffe Google Übersetzer hat gut funktioniert. Vielen Dank für Ihre Hilfe._