olikraus/u8g2 #551: Can't use u8g2 on Arduino - Development Board is STM32F103C8T6, 4W HW SPI interface, SH1106

a0908896067, 2018-03-28:
Arduino IDE version : 1.8.5
Arduino core for ST: from STM32duino.com, version=0.1.2 
DEV. Board: STM32F103C8T6
OLED driver: SH1106

I used Arduino & u8g2 on ESP32 board for a while without any problems but I have problems to use similar way  on new low cost ARM board with STM32F103C8T6. I used the following test program and see the result from logic analyzer, I found the issue is from SPI clock speed. SH1106's max SCK speed is 1MHz and default SCK settings for STM32F103 board is 4.6MHz. If I don't use u8g2 lib, I have no problem to change SPI SCK clock speed. But if I use u8g2 lib, then the SPI clock's speed is always 4.6MHz, can some one tell me how to fix this issue. Thanks in advance. 

Rgds,
Michael

#include <Arduino.h>
#include <U8g2lib.h>
//#include <U8x8lib.h>
#include <SPI.h>

#define Enable_U8G2

#define SPI_RESET PB0
#define SPI_CS1   PA4                // MEGA 2560 PIN 48
#define SPI_DC    PB1                // MEGA 2560 PIN 49
#define SPI_CLK   PA5
#define SPI_MOSI  PA7

U8G2_SH1106_128X64_NONAME_1_4W_HW_SPI u8g2(U8G2_R0, SPI_CS1, SPI_DC, SPI_RESET);
 
void setup(void) { 
  Serial.begin(115200);
  Serial.println(F("Testing SPI CLK speed"));
  
  // Setup SPI 1
  SPI.begin(); //Initialize the SPI_1 port.
  SPI.setBitOrder(MSBFIRST); // Set the SPI_1 bit order
  SPI.setDataMode(SPI_MODE0); //Set the  SPI_2 data mode 0
  SPI.setClockDivider(SPI_CLOCK_DIV64);      // DIV16=4.6MHz, DIV64=1.15MHz 
  pinMode(SPI_CS1, OUTPUT);
  digitalWrite(SPI_CS1,HIGH);
  
  // reset SH1106
  pinMode(SPI_RESET,OUTPUT);
  digitalWrite(SPI_RESET,HIGH);
  delay(500);
  digitalWrite(SPI_RESET,LOW);
  delay(500);
  digitalWrite(SPI_RESET,HIGH);

#ifdef Enable_U8G2
  u8g2.begin();
#endif
}

uint8_t i=1;
  
void loop(void) 
{
    Serial.println(i++);
#ifndef Enable_U8G2
    SPI.write(i);
    delay(5);
#else
    u8g2.drawBox(0,0,127,63);
    u8g2.sendBuffer();
    delay(1000);
    
    u8g2.clearDisplay();
    u8g2.sendBuffer();
    delay(1000);

    u8g2.drawStr(16,13,"YTT Workshop");
    u8g2.sendBuffer();
    delay(1000);
#endif
}

olikraus, 2018-03-28:
Just checked a datasheet for the SH1106: 4MHz should be ok for SPI interface. Howhever 4.7 will be too much I guess. 

At the moment 8MHz are configured. I do not remember how this was introduced, maybe I need to reduce this to 4MHz:
https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_ssd1306_128x64_noname.c#L299

Another difference to your tests is, that u8g2 will use the more newer SPISettings interface. 

Can you change the clock speed from 8000000 to 400000 in 
https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_ssd1306_128x64_noname.c#L299
You need to locate this file on your local hard disk.

Will it work with this reduced value?

a0908896067, 2018-03-28:
according to SH1106 datasheet (V0.1), the min. serial clock cycle time is 500ns ( or 2MHz). based on the spec. I read, 4MHz should not work. But I will test it tomorrow as it's mid night here in Taiwan. Thanks your help.

[https://www.elecrow.com/download/SH1106%20datasheet.pdf](url) SPI spec. on page 40

olikraus, 2018-03-28:
If voltages is higher than 2.4V, then 250ns (4MHz) are permitted according to your datasheet.

a0908896067, 2018-03-29:
Yes, you are right, if Voltage > 2.4V, then 250ns is OK.

I have changed the function from SPI.setClockDivider to SPISettings as below, the SCK speed is still 4.6MHz.
SPISettings(1000000, MSBFIRST, SPI_MODE0);    //Initialize the SPI_1 port.
  //SPI.setBitOrder(MSBFIRST); // Set the SPI_1 bit order
  //SPI.setDataMode(SPI_MODE0); //Set the  SPI_2 data mode 0
  //SPI.setClockDivider(SPI_CLOCK_DIV64);      // DIV16=4.6MHz, DIV64=1.15MHz

my next step is to change sck_clock_hz settings in the u8x8_d_ssd1306_128x64_noname.c from 8000000UL to 4000000UL, and will update latest status later

olikraus, 2018-03-29:
ok, great. Anyhow, I have to reduce clock speed to 4MHz.

For the first/next page, see also here: https://github.com/olikraus/u8g2/wiki/setup_tutorial#u8g2-full-buffer--page-buffer-and-u8x8-mode

U8g2 offers several drawing methods. With an ATMega2560 better use the full buffer mode.

olikraus, 2018-04-15:
done...

closed 2018-04-15 by olikraus