olikraus/u8g2 #553: ssd1315兼容吗？

13590787364, 2018-03-29:
可以支持ssd1315吗？ssd1315好像是2017年的一款，与1306兼容。

RobotGB, 2018-04-04:
According to Google Translate, @13590787364 seems to be saying:
"Can you support ssd 1315? Ssd 1315 seems to be a 2017, compatible with 1306."
According to [the wiki](https://github.com/olikraus/u8g2/wiki/u8g2setupcpp), ssd1306 is compatible with u8g2, so if ssd1315 is compatible with ssd1306, it should work.
根据[wiki](https://github.com/olikraus/u8g2/wiki/u8g2setupcpp)，ssd1306与u8g2兼容，所以如果ssd1315与ssd1306兼容，它应该可以工作。

olikraus, 2018-04-04:
so.... nothing needs to be done here?

olikraus, 2018-04-15:
so... I assume I can close this...

closed 2018-04-15 by olikraus

SunchinSekian, 2022-07-28:
> 可以支持ssd1315吗？ssd1315好像是2017年的一款，与1306兼容。

请问成功了吗，我失败了

piogrammer, 2023-06-04:
Yes, bro, SSD1315 code are mostly compatible with SSD1306,
but it's strange not to be compatible with SSD1312 or 1316, isn't it? 
All in all, it's a bit confusing, but it works.~_~