olikraus/u8g2 #554: Fonts are not sharp

funkamator, 2018-03-29:
Hi Oliver , First of all , thank you very much for the great lib , U8g2.
So, i tried to convert new convert , ( first ttf to bdf by FontForge , than bdf to c file by bdfconv.exe ) it seems ok and works.But glyphs are not shape.I used bdfconv.exe like this;

"bdfconv.exe -f 1 -b 0  -m "32-355" -n Typo_r_21 -o Typo_r_21.c Typo_r_21.bdf" , is that true? Or is there any way to increase  the resolution?

One more thing : would you create " expansiva font" ( https://www.dafont.com/expansiva.font) as c file (between 32 - 355 , because i need Turkish font )?

Mesut

olikraus, 2018-03-29:
> So, i tried to convert new convert , ( first ttf to bdf by FontForge , than bdf to c file by bdfconv.exe ) it seems ok and works.But glyphs are not shape.I used bdfconv.exe like this;

bdfconv.exe does not modify the bitmap. Usually the ttf to bdf process makes the glyph not sharp. In many cases the conversion from ttf to bdf requires a very specific point size (usuall 8 or 16). Also ensure that you disable hinting and antialiasing during this process (at least this helps when using otf2bdf tool).

> One more thing : would you create " expansiva font" ( https://www.dafont.com/expansiva.font) as c file (between 32 - 355 , because i need Turkish font )?

I assume Turkish glyphs should be included in all fonts, ending in "e"
The following fonts already exist in u8g2 and should include Turkish chars:

```
u8g2_font_10x20_te    u8g2_font_helvB12_te  u8g2_font_helvR24_te  u8g2_font_ncenR12_te  u8g2_font_t0_13_te   u8g2_font_t0_18b_te
u8g2_font_6x12_te     u8g2_font_helvB14_te  u8g2_font_ncenB08_te  u8g2_font_ncenR14_te  u8g2_font_t0_14b_te  u8g2_font_t0_18_te
u8g2_font_6x13_te     u8g2_font_helvB18_te  u8g2_font_ncenB10_te  u8g2_font_ncenR18_te  u8g2_font_t0_14_te   u8g2_font_t0_22b_te
u8g2_font_7x13_te     u8g2_font_helvB24_te  u8g2_font_ncenB12_te  u8g2_font_ncenR24_te  u8g2_font_t0_15b_te  u8g2_font_t0_22_te
u8g2_font_8x13_te     u8g2_font_helvR08_te  u8g2_font_ncenB14_te  u8g2_font_t0_11b_te   u8g2_font_t0_15_te
u8g2_font_9x15_te     u8g2_font_helvR10_te  u8g2_font_ncenB18_te  u8g2_font_t0_11_te    u8g2_font_t0_16b_te
u8g2_font_cu12_te     u8g2_font_helvR12_te  u8g2_font_ncenB24_te  u8g2_font_t0_12b_te   u8g2_font_t0_16_te
u8g2_font_helvB08_te  u8g2_font_helvR14_te  u8g2_font_ncenR08_te  u8g2_font_t0_12_te    u8g2_font_t0_17b_te
u8g2_font_helvB10_te  u8g2_font_helvR18_te  u8g2_font_ncenR10_te  u8g2_font_t0_13b_te   u8g2_font_t0_17_te

```
You can look at these fonts here:
https://github.com/olikraus/u8g2/wiki/fntgrpx11
https://github.com/olikraus/u8g2/wiki/fntgrpttyp0
https://github.com/olikraus/u8g2/wiki/fntgrpadobex11

funkamator, 2018-04-04:
Many thank for your support..

Mesut

olikraus, 2018-04-04:
:-)