olikraus/u8g2 #555: ST7920 8bit stm32f407

ilyadesign, 2018-03-31:
u8x8cb.c
```
uint8_t u8x8_gpio_and_delay_stm32(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
{
    switch (msg)
  {
  case U8X8_MSG_GPIO_AND_DELAY_INIT:  // called once during init phase of u8g2/u8x8
      GPIO_LCD_DB_Init_OUTPUT();
      break;              // can be used to setup pins
  case U8X8_MSG_DELAY_NANO:     // delay arg_int * 1 nano second
      delay_ns(arg_int);
      break;

  case U8X8_MSG_DELAY_100NANO:    // delay arg_int * 100 nano seconds
	  delay_ns(100 * arg_int);
      break;

  case U8X8_MSG_DELAY_10MICRO:    // delay arg_int * 10 micro seconds
	  delay_us(10 * arg_int);
      break;

  case U8X8_MSG_DELAY_MILLI:      // delay arg_int * 1 milli second
	  HAL_Delay(arg_int);
      break;

  case U8X8_MSG_GPIO_D0:        // D0 or SPI clock pin: Output level in arg_int
      if (arg_int) { HAL_GPIO_WritePin(LCD_DB0_GPIO_Port,LCD_DB0_Pin,GPIO_PIN_SET); }
      else { HAL_GPIO_WritePin(LCD_DB0_GPIO_Port,LCD_DB0_Pin,GPIO_PIN_RESET); }
      break;

  case U8X8_MSG_GPIO_D1:        // D1 or SPI data pin: Output level in arg_int
      if (arg_int) { HAL_GPIO_WritePin(LCD_DB1_GPIO_Port,LCD_DB1_Pin,GPIO_PIN_SET); }
      else { HAL_GPIO_WritePin(LCD_DB1_GPIO_Port,LCD_DB1_Pin,GPIO_PIN_RESET); }
      break;

  case U8X8_MSG_GPIO_D2:        // D2 pin: Output level in arg_int
      if (arg_int) { HAL_GPIO_WritePin(LCD_DB2_GPIO_Port,LCD_DB2_Pin,GPIO_PIN_SET); }
      else { HAL_GPIO_WritePin(LCD_DB2_GPIO_Port,LCD_DB2_Pin,GPIO_PIN_RESET); }
      break;

  case U8X8_MSG_GPIO_D3:        // D3 pin: Output level in arg_int
      if (arg_int) { HAL_GPIO_WritePin(LCD_DB3_GPIO_Port,LCD_DB3_Pin,GPIO_PIN_SET); }
      else { HAL_GPIO_WritePin(LCD_DB3_GPIO_Port,LCD_DB3_Pin,GPIO_PIN_RESET); }
      break;

  case U8X8_MSG_GPIO_D4:        // D4 pin: Output level in arg_int
      if (arg_int) { HAL_GPIO_WritePin(LCD_DB4_GPIO_Port,LCD_DB4_Pin,GPIO_PIN_SET); }
      else { HAL_GPIO_WritePin(LCD_DB4_GPIO_Port,LCD_DB4_Pin,GPIO_PIN_RESET); }
      break;

  case U8X8_MSG_GPIO_D5:        // D5 pin: Output level in arg_int
      if (arg_int) { HAL_GPIO_WritePin(LCD_DB5_GPIO_Port,LCD_DB5_Pin,GPIO_PIN_SET); }
      else { HAL_GPIO_WritePin(LCD_DB5_GPIO_Port,LCD_DB5_Pin,GPIO_PIN_RESET); }
      break;

  case U8X8_MSG_GPIO_D6:        // D6 pin: Output level in arg_int
      if (arg_int) { HAL_GPIO_WritePin(LCD_DB6_GPIO_Port,LCD_DB6_Pin,GPIO_PIN_SET); }
      else { HAL_GPIO_WritePin(LCD_DB6_GPIO_Port,LCD_DB6_Pin,GPIO_PIN_RESET); }
      break;

  case U8X8_MSG_GPIO_D7:        // D7 pin: Output level in arg_int
      if (arg_int) { HAL_GPIO_WritePin(LCD_DB7_GPIO_Port,LCD_DB7_Pin,GPIO_PIN_SET); }
      else { HAL_GPIO_WritePin(LCD_DB7_GPIO_Port,LCD_DB7_Pin,GPIO_PIN_RESET); }
      break;

  case U8X8_MSG_GPIO_E:       // E/WR pin: Output level in arg_int
      if (arg_int) { HAL_GPIO_WritePin(LCD_E_GPIO_Port,LCD_E_Pin,GPIO_PIN_SET); }
      else { HAL_GPIO_WritePin(LCD_E_GPIO_Port,LCD_E_Pin,GPIO_PIN_RESET); }
      break;

  case U8X8_MSG_GPIO_CS:        // CS (chip select) pin: Output level in arg_int
	  u8x8_SetGPIOResult(u8x8, 1);      // default return value
	  break;

  case U8X8_MSG_GPIO_DC:        // DC (data/cmd, A0, register select) pin: Output level in arg_int
      if (arg_int) { HAL_GPIO_WritePin(LCD_RS_GPIO_Port,LCD_RS_Pin,GPIO_PIN_SET); }
      else { HAL_GPIO_WritePin(LCD_RS_GPIO_Port,LCD_RS_Pin,GPIO_PIN_RESET); }
      break;

  case U8X8_MSG_GPIO_RESET:     // Reset pin: Output level in arg_int
	  if (arg_int) { HAL_GPIO_WritePin(LCD_RESET_GPIO_Port,LCD_RESET_Pin,GPIO_PIN_SET); }
	  else { HAL_GPIO_WritePin(LCD_RESET_GPIO_Port,LCD_RESET_Pin,GPIO_PIN_RESET); }
	  break;

  case U8X8_MSG_GPIO_MENU_SELECT:
      u8x8_SetGPIOResult(u8x8, /* get menu select pin state */ 0);
      break;
  case U8X8_MSG_GPIO_MENU_NEXT:
      u8x8_SetGPIOResult(u8x8, /* get menu next pin state */ 0);
      break;
  case U8X8_MSG_GPIO_MENU_PREV:
      u8x8_SetGPIOResult(u8x8, /* get menu prev pin state */ 0);
      break;
  case U8X8_MSG_GPIO_MENU_HOME:
      u8x8_SetGPIOResult(u8x8, /* get menu home pin state */ 0);
      break;
  default:
      u8x8_SetGPIOResult(u8x8, 1);      // default return value
      break;
  }
  return 1;
}
```
=========================
main.c
```
#include "u8g2.h"
#include "u8x8cb.h"
...
u8g2_t u8g2;
...
u8g2_Setup_st7920_p_128x64_f(&u8g2, U8G2_R0, u8x8_byte_8bit_6800mode, u8x8_gpio_and_delay_stm32);
u8g2_InitDisplay(&u8g2);
u8g2_SetPowerSave(&u8g2, 0);

u8g2_ClearBuffer(&u8g2);
  u8g2_FirstPage(&u8g2);
  do
  {
    u8g2_SetFontMode(&u8g2, 1);
    u8g2_SetFontDirection(&u8g2, 0);
    u8g2_SetFont(&u8g2, u8g2_font_u8glib_4_tf);
    u8g2_DrawStr(&u8g2, 0, 1, "hello world");
  } while( u8g2_NextPage(&u8g2) );
  u8g2_SendBuffer(&u8g2);    // transfer data from uC memory to display
```
Eclipse, STM32F407, ST7920, 8 bit interface. Why doesn`t it work? I think initialization. Thanks!

ilyadesign, 2018-03-31:
Result photo: [https://yadi.sk/i/mP3AU8tE3Tvw33](url)

olikraus, 2018-03-31:
Wiring problem?

ilyadesign, 2018-03-31:
Wiring is OK, because my code is work fine. I tried to change 8080 and 6800 mode = both don't work. I tried change my own init code to u8x8_d_st7920_init_seq instruction = it is work.

olikraus, 2018-03-31:
So all is ok now?

ilyadesign, 2018-03-31:
My code with u8g2 init instruction is work fine, but u8g2 don't work.

ilyadesign, 2018-03-31:
```
void lcd_Init(void)
{
	GPIO_LCD_DB_Init_OUTPUT();
	HAL_Delay(20); //Wait more than 15ms after VDD rises to 0.9VDD
	RESET_LOW();
	delay_us(3000);
	RESET_HIGH();
	HAL_Delay(50);

	lcd_WriteByte(0x30, COM); //Function set 1, DL = 1; 8 bit interface operation
	HAL_Delay(5);
	lcd_WriteByte(0x30, COM); //Function set 2, RE = 0; basic instructions  ** Cannot set both DL and RE at the same time **
	delay_us(150);
	lcd_WriteByte(0x30, COM); //Function set 3, RE = 0; basic instructions  ** Cannot set both DL and RE at the same time **
	delay_us(150);
	lcd_WriteByte(0x0C, COM); //0b00001, 1-Display on, 1-cursor on, 1-char blink on (0x0C)
	delay_us(150);
	lcd_WriteByte(0x01, COM); //Display clear
	HAL_Delay(2);
	lcd_WriteByte(0x06, COM); //Entry mode set: 0b000001, 1-I/D increment AC (0-decrement AC), 0-S normal mode (1-shift display)
	delay_us(1000);
	lcd_WriteByte(0x02, COM); // Set DDRAM AC to "00H", cursor is put at origin (home)
	delay_us(1000);

	lcd_Clear();
	RW_LOW();
	HAL_Delay(10);
}
```
It's OK

ilyadesign, 2018-03-31:
Initialization from u8g2:
```
u8g2_Setup_st7920_p_128x64_f(&u8g2, U8G2_R0, u8x8_byte_8bit_8080mode, u8x8_gpio_and_delay_stm32);
u8g2_InitDisplay(&u8g2);
u8g2_SetPowerSave(&u8g2, 0);
```

Sending command from my code:
```
void lcd_WriteByte(char data, char rs)
{
	lcd_CheckBusy();
	LCD_OUT &= (~LCD_PIN_MASK);
	if (rs == COM) RS_LOW();    // rs = 0  (COM)
	else 			RS_HIGH();   // rs = 1  (DATA)
	RW_LOW();
	LCD_OUT |= (data << 8) & 0xFF00; //Shift 8 bit, PD8~PD15
	E_HIGH();
	delay_us(2);
	E_LOW();
	delay_us(10);
}
```
I shift LCD_OUT data to PD8~PD15.
It's worked.

ilyadesign, 2018-03-31:
Where I should change the code to shift data to target pins (PD8~PD15) ?

ilyadesign, 2018-03-31:
**finally u8g2 began to work!!!**
This is my delay function:
```
void delay_ns(const uint16_t ns)
{
  uint32_t i = ns;
  while (i-- > 0) {
    __NOP();
  }
}
```
for 168MHz stm32f407 __NOP(); is 6ns

```
case U8X8_MSG_DELAY_NANO:     // delay arg_int * 1 nano second
      delay_ns(50 * arg_int);
      break;
```
I  increased ns delay 6*50=300 times and the display began to work.

olikraus, 2018-03-31:
Great!

closed 2018-04-15 by olikraus