olikraus/u8g2 #557: information missing from u8g2reference wiki page

mikekgr, 2018-04-02:
Dear Sirs, Dear @olikraus 
I am truing to learn and then to use this very very good library ( thanks for your excellent work )  and studying the wiki, more precisely the url: 
https://github.com/olikraus/u8g2/wiki/u8g2reference
There, I can not find commands like getDisplayHeight   getMaxCharHeight etc that there are in the examples like Shennong.ino but not in the wiki
Please try to have documentation without missing items.

Thanks and Best Regards,
Mike Kranidis

olikraus, 2018-04-02:
good point. Will add this.

mikekgr, 2018-04-03:
thanks a lot Sir for your fast reaction to this documentation inconsistency problem. Keep the fine work!

olikraus, 2018-04-03:
I leave this open, until i did the change to the wiki

olikraus, 2018-04-15:
getDisplayHeight/Width is done...

mikekgr, 2018-04-15:
Thanks a lot hard worker @olikraus  !

olikraus, 2018-04-15:
:-)

Documented https://github.com/olikraus/u8g2/wiki/u8g2reference#getmaxcharheight

Also getMaxCharWidth is there, but the code will be delayed until release of 2.22.

closed 2018-04-15 by olikraus

matthijskooijman, 2018-10-08:
@olikraus, seems the code for getMaxCharWidth is still not present, but it is documented. What's the plan here?

olikraus, 2018-10-08:
It is implemented and released with version 2.22.
https://github.com/olikraus/u8g2/blob/master/cppsrc/U8g2lib.h#L302
Current version is 2.23, this function should be available since almost a half year.
Which u8g2 version do you use?

matthijskooijman, 2018-10-08:
Hm, seems I was wrong then. I used the github search which showed no code results, so I was either wrong to trust github search, or perhaps I made a typo. So never mind, sorry for the noise :-)

olikraus, 2018-10-08:
no problem