olikraus/u8g2 #558: Add best BDF font editor (win)

ilyadesign, 2018-04-02:
Please, add to u8g2/tools/font/ this BDF font editor [http://hukka.ncn.fi/?fony](url). Best editor for BDF.

olikraus, 2018-04-03:
Yes, indeed a nice editor. What exactly shell I do?

ilyadesign, 2018-04-03:
Sorry, I didn't understand the question.

olikraus, 2018-04-03:
> Sorry, I didn't understand the question.

Well, this is an issue tracker for u8g2 library. Usually people report bugs in u8g2 library or request new enhancements to u8g2 library. So the question is, what is your exact request here for the u8g2 library?

ilyadesign, 2018-04-03:
I just suggested adding the application to the repository.

olikraus, 2018-04-03:
Difficult: I am not the author, I can not provide any support, it does not belong to u8g2, copyright is not clear.

samsam4, 2018-05-15:
Could you, please, post some working link to a repository with Fony 1.4.7 as the link http://hukka.ncn.fi/?fony seems dead long time ago :( 
I've got version 1.4.6 from softpedia.com and try to import few of the open_iconic_****.bdf, but it either crashes or doesn't  show anything - not sure if the problem is with the Fony 1.4.6 or the fonts?

olikraus, 2018-05-15:
reopend this... I will check whether I have a recent version of fony somewhere on my local drive.

open_iconic bdf files unfortunately are not proper bdf files. the open_iconc files are generated from png files with the help of a tool which I recently wrote. The generated bdf just contains enough information to allow bdfconv to work with it.

Reference:
https://github.com/olikraus/u8g2/tree/master/tools/font/png2bdf
https://github.com/olikraus/u8g2/tree/master/tools/font/bdfconv

closed 2018-05-15 by olikraus

samsam4, 2018-05-15:
That is the version  what I'm looking for.  Else the one that I downloaded have is an older one 1.4.6

olikraus, 2018-05-15:
Ok, then it is here for download:
https://github.com/olikraus/u8g2/tree/master/tools/font/fony

hukkax, 2018-05-15:
Author of Fony here. I have no problem with having it here for now. The server hosting my website had a failure recently but the site should be up in a couple of weeks, hopefully.

olikraus, 2018-05-15:
@hukkax Thanks for your great font editor and allowing us to keep a copy of it.

olikraus, 2018-05-25:
http://hukka.ncn.fi/?fony is online :)

Secret-chest, 2023-08-23:
> Author of Fony here. I have no problem with having it here for now. The server hosting my website had a failure recently but the site should be up in a couple of weeks, hopefully.

Licence? Linux support?

olikraus, 2023-08-23:
If I remember correctly, fony did work within "wine" on linux.

hukkax, 2023-08-23:
> Licence? Linux support?

License is closed-source Freeware. Win32 only, but I'd expect it to work on Linux via Wine. (This is an ancient project mostly made in Delphi 7.)

Secret-chest, 2023-08-24:
> License is closed-source Freeware. Win32 only, but I'd expect it to work on Linux via Wine. (This is an ancient project mostly made in Delphi 7.)

Oh ok, I don't use freeware, have to go with FontForge, sorry.

olikraus, 2023-08-24:
Hmm... fontforge is freeware as well...