olikraus/u8g2 #561: ST75256_256*160

EleksMaker, 2018-04-05:
Hi, JLX256128 have new huge display at 256*160 , could you please make one to fit it? XD

olikraus, 2018-04-05:
Did you test the existing JLX256128? Can you attach a picture how the JLX256160 looks with the JLX256128 driver?

EleksMaker, 2018-04-05:
yes i have test out use JLX240160 it works but can fit all screen ( its a nice 5inch screen
[image]

olikraus, 2018-04-05:
Looks good. Nice Display. I think we can use this driver as a base for your display. Probably only the width needs to be adjusted.

EleksMaker, 2018-04-05:
yes , and i try 256*128 too and modify the height to 0X9F(159) too , is also have the messy code is at the bottom , your u8g2 is a wonderful driving code. I think we can use this to make this screen work. maybe is the spi shift or Initialize memory problems?

olikraus, 2018-04-05:
> I think we can use this to make this screen work. maybe is the spi shift or Initialize memory problems?

Multiple things have to be done, but as you can see from the above commits, I already did most of the things. There is now a new constructor:
U8G2_ST75256_JLX256160_1_4W_SW_SPI and friends.

There is also a beta release 2.22.9 available here: https://github.com/olikraus/U8g2_Arduino

Please download the zip file and install it via the add zip library menu of the Arduino IDE.
The above mentioned constructor is already part of all examples: Can you do some testing with the new constructor? Especially, can you check whether the u8g2 page buffer FlipMode example works?
There should be a closed frame all the time.

EleksMaker, 2018-04-06:
wow ， huge thx , but somethings seems goes wrong

first , it display nothing , until i modify this code to 0x50 , it works 

but the screen room seems not correct , i am trying to work out.

[image]

Best Regards,  Z

EleksMaker, 2018-04-06:
[image]

EleksMaker, 2018-04-06:
i found that 256*128 also have problem , witch cant dispaly all 256 pixel , it just can display 256-16 piexl , so i think 256*160 cant work is in same problem.

olikraus, 2018-04-06:
ah, yes, I always forget something (And it also was late night for me). 
You can review the commit above, it should include the changes you mentioned.
I also created a new beta 2.22.10 for download. Will it work now?

olikraus, 2018-04-06:
I forgot to mention this: For U8g2 examples always enable U8g2 16 Bit mode (uncomment the related line in u8g2.h)