olikraus/u8g2 #566: How does font work?

IWenjunSun, 2018-04-09:
I really like your library but I don't need all other functions, I just want to print big "logisoso92" numbers. I went through the code a little bit and it seems like the fonts are compressed? Any hint will be appreciated.

olikraus, 2018-04-09:
Hint for what? What is the problem?

IWenjunSun, 2018-04-09:
How does font being encoded into the u8g2_font_logisoso92_tn[1612] array.

olikraus, 2018-04-09:
https://github.com/olikraus/u8g2/wiki/u8g2fontformat

olikraus, 2018-04-09:
ttf to bdf: use otf2bdf tool or fontforge
bdf to U8g2 format: see the U8g2 FAQ

closed 2018-04-15 by olikraus

RobotGB, 2018-09-19:
A question on the glyph bitmap:
is the bitmap structured something like this after the unicode and number of bytes:
(_m0_) (_m1_) 1 1 1 0 (_m0_) (_m1_) 0 (_m0_) (_m1_) 0 etc.
or is it another structure?
(_m0_ = the bits to set as 0, it is 'a' in the code)
(_m1_ = the bits to set as 1, it is 'b' in the code)

olikraus, 2018-09-20:
Let me assume your starting point:
`. . X . .`
`. X . X .`
`X . . . X`
`X X X X X`
`X . . . X`

This will be represented like this (line breaks removed):
`. . X . . . X . X . X . . . X X X X X X X . . . X`
Now we have a sequence of dots and x. I count the dots and x:
`2 1 3 1 1 1 1 1 3 7 3 1`
The numbers will be small, so I will not waste space to use full bytes to store the numbers, instead i put the numbers into smaller pieces. Let me assume 3 bits for the dots and 3 bits for the x count, then the bitstream will be this (i just converted the numbers from above to bitpattern):
` 0 1 0 | 0 0 1 | 0 1 0 | 0 0 1 | 0 0 1 | 0 0 1 | 0 0 1 | 0 0 1 | 0 1 0 | 1 1 1 | 0 1 0 | 0 0 1`
Now this bitstream is stored as bytes:
` 0 1 0 0 0 1 0 1 || 0 0 0 1 0 0 1 0 || 0 1 0 0 1 0 0 1 || 0 1 0 1 1 1  0 1 || 0 0 0 1 ? ? ? ?`
The last byte has to be filled with some random values (0 in case of u8g2), so we get:
` 0 1 0 0 0 1 0 1 || 0 0 0 1 0 0 1 0 || 0 1 0 0 1 0 0 1 || 0 1 0 1 1 1  0 1 || 0 0 0 1 0 0 0 0`
Adding line breaks, we get this:
```
0 1 0 0 0 1 0 1 
0 0 0 1 0 0 1 0 
0 1 0 0 1 0 0 1 
0 1 0 1 1 1 0 1 
0 0 0 1 0 0 0 0
```

ok, does not look like much of saving for the inital bitmap, but for larger bitmaps, there is significat saving. Also the selection of 3 bits for the dot and x count might be bad. Other values might lead to better compression. In fact the number of bits per count is flexible and calculated separatly for each font to achieve the best compression.