olikraus/u8g2 #567: Button press not detected (but is with `digitalRead`)

erichiller, 2018-04-09:
I'm having no success in getting menu functions to work. Specifically I can't get basic button presses to work. I'm not sure if this is a bug, or if it is some user error/ignorance on my side. I've reviewed 
[u8x8_debounce](https://github.com/olikraus/U8g2_Arduino/blob/master/src/clib/u8x8_debounce.c) and I'm pretty sure I understand it, it is detecting detecting state changes. So there should not be an issue.

In my testing, I've tried using digitalRead() on the same pins as I've assigned to the menu buttons, and they do detect a state change (from high to low) when the button is pressed. Wiring looks like:

```
[gpio]---[breadboard]--->momentary_switch<---[negative/ground on breadboard]---[ground on esp32]
```

I start with
```
	u8g2.begin( MENU_pin_select, MENU_pin_next, MENU_pin_prev, U8X8_PIN_NONE, U8X8_PIN_NONE, U8X8_PIN_NONE ); 
```

Then utilize later on in my `loop()` with

```
	int8_t event = u8g2.getMenuEvent( );
	switch( event ) {
		case U8X8_MSG_GPIO_MENU_SELECT:
			sprintf( print_action, "%s", "menu_select" );
			Serial.println( print_action );
			break;
		default:
			sprintf( print_action, "%s %i", "pressed:", event );
			Serial.println( print_action );
			break;
	}
```

If I'm doing something off, please let me know, I've included what I thought was the relevant code, but the entirety of my source can be found [here](https://github.com/erichiller/fan-controller-esp32/blob/master/temp-controller-esp32.ino)

olikraus, 2018-04-09:
So what is the output? It will always execute the default path, right?
You should at least do not output something if the return value is 0 for getMenuEvent()

erichiller, 2018-04-09:
I reworked it a bit to output better diagnostics, changed up the switch statement to cover all options(left default just in case) and made one pin have a default LOW. Here is what happens for the default LOW going HIGH. The same happens for the HIGHs going LOW.

```
DEFAULT pressed: 0
MENU_pin_select == HIGH
MENU_pin_next == HIGH
MENU_pin_prev == LOW
MENU_pin_up == HIGH
89055 ms || 714 milliVolt || 20.00 degrees C || 69.80 degrees F
DEFAULT pressed: 0
MENU_pin_select == HIGH
MENU_pin_next == HIGH
MENU_pin_prev == LOW
MENU_pin_up == HIGH
90063 ms || 720 milliVolt || 21.00 degrees C || 71.60 degrees F
DEFAULT pressed: 0
MENU_pin_select == HIGH
MENU_pin_next == HIGH
MENU_pin_prev == HIGH
MENU_pin_up == HIGH
91071 ms || 560 milliVolt || 22.00 degrees C || 42.80 degrees F
DEFAULT pressed: 0
MENU_pin_select == HIGH
MENU_pin_next == HIGH
MENU_pin_prev == HIGH
MENU_pin_up == HIGH
92079 ms || 704 milliVolt || 6.00 degrees C || 68.00 degrees F
DEFAULT pressed: 0
MENU_pin_select == HIGH
MENU_pin_next == HIGH
MENU_pin_prev == HIGH
MENU_pin_up == HIGH
93087 ms || 708 milliVolt || 20.00 degrees C || 68.00 degrees F
DEFAULT pressed: 0
MENU_pin_select == HIGH
MENU_pin_next == HIGH
MENU_pin_prev == HIGH
MENU_pin_up == HIGH
94095 ms || 672 milliVolt || 20.00 degrees C || 62.60 degrees F
DEFAULT pressed: 0
MENU_pin_select == HIGH
MENU_pin_next == HIGH
MENU_pin_prev == LOW
MENU_pin_up == HIGH
95103 ms || 674 milliVolt || 17.00 degrees C || 62.60 degrees F
DEFAULT pressed: 0
MENU_pin_select == HIGH
MENU_pin_next == HIGH
MENU_pin_prev == LOW
MENU_pin_up == HIGH
```

I changed the switch statement too:

```
	int8_t event = u8g2.getMenuEvent( );
	switch( event ) {
		case U8X8_MSG_GPIO_MENU_SELECT:
			sprintf( print_action, "%s", "menu_select" );
			Serial.println( print_action );
			break;
		case U8X8_MSG_GPIO_MENU_NEXT:
			sprintf( print_action, "%s", "MENU_NEXT" );
			Serial.println( print_action );
			break;
		case U8X8_MSG_GPIO_MENU_PREV:
			sprintf( print_action, "%s", "MENU_PREV" );
			Serial.println( print_action );
			break;
		case U8X8_MSG_GPIO_MENU_UP:
			sprintf( print_action, "%s", "MENU_UP" );
			Serial.println( print_action );
			break;
		case U8X8_MSG_GPIO_MENU_DOWN:
			sprintf( print_action, "%s", "MENU_down" );
			Serial.println( print_action );
			break;
		case U8X8_MSG_GPIO_MENU_HOME:
			sprintf( print_action, "%s", "MENU_home" );
			Serial.println( print_action );
			break;
		default:
			sprintf( print_action, "%s %i", "DEFAULT pressed:", event );
			Serial.println( print_action );
			break;
	}
```

Does that help any?

olikraus, 2018-04-09:
I mean, you should have an additional case for value 0:

case 0: break;

which does nothing
All GPIOs should be active low (so the should be HIGH by default).

olikraus, 2018-04-09:
BTW: Did you try one of the examples which require some input? Like the selection list examples? Will these examples work?

erichiller, 2018-04-10:
I have them all setup now to be HIGH default, switch sets them LOW. I tried the demo using this code:

```
	while( true ) {
		u8g2.setFont( u8x8_font_chroma48medium8_r );
		r = u8g2.userInterfaceMessage( "Message", "Box", NULL, " Ok \n Cancel " );
		Serial.println( "userInterfaceMessage ... " );
		if( r == 0 ) {
			u8g2.userInterfaceMessage( "You pressed the", "Home/Quit", "Button", " Ok " );
			Serial.println( "was 0" );

		} else if( r == 1 ) {
			u8g2.userInterfaceMessage( "You selected the", "Ok", "Button", " Ok " );
			Serial.println( "was 1" );

		} else if( r == 2 ) {
			u8g2.userInterfaceMessage( "You selected the", "Cancel", "Button", " Ok " );
			Serial.println( "was 2" );
		}
	}
```

There was never any response, just an infinite loop.

olikraus, 2018-04-10:
Hmm.. not sure what i can do here... For me this works fine.

erichiller, 2018-04-10:
I'm not either. I put some logging into the u8g2 library and maybe this will help? To me it looks like I should be getting a returned menu value?

Same format as before.  But I modified `u8x8_debounce` 
```
#include <Arduino.h>

static uint8_t u8x8_read_pin_state(u8x8_t *u8x8)
{
  uint8_t i;
  uint8_t pin_state;
Serial.println("in u8x8_read_pin_state:: ");
  Serial.print("U8X8_PIN_INPUT_CNT=[");
  Serial.print(U8X8_PIN_INPUT_CNT);
  Serial.print("] ");
  
  pin_state = 255;	/* be compatible with the setup of the default pin setup, which is 255 */
  for( i = 0; i < U8X8_PIN_INPUT_CNT; i++ )
  {
    pin_state <<= 1;
    
  Serial.print("i=[");
  Serial.print(i);
  Serial.print("] ");
    /* the callback function should put the return value into this variable */
    u8x8->gpio_result = 1;
    u8x8_gpio_call(u8x8, U8X8_MSG_GPIO(i+U8X8_PIN_OUTPUT_CNT), 0);
    pin_state |= u8x8->gpio_result & 1;
    Serial.print(u8x8->gpio_result, BIN);
  Serial.print("} ");
    
  }

  Serial.print(" pin_state={");
    
    Serial.print(pin_state, BIN);
  Serial.print("} ");

Serial.println(" ;;end u8x8_read_pin_state");
  
  return pin_state;
}
```

```
==== 0 ====
MENU_pin_select == HIGH
MENU_pin_next == HIGH
MENU_pin_prev == HIGH
MENU_pin_up == HIGH
MENU_pin_down == HIGH
MENU_pin_home == HIGH
944185 ms || 752 milliVolt || 25.00 degrees C || 77.00 degrees F
in u8x8_read_pin_state:: 
U8X8_PIN_INPUT_CNT=[6] i=[0] 1} i=[1] 1} i=[2] 1} i=[3] 1} i=[4] 1} i=[5] 1}  pin_state={11111111}  ;;end u8x8_read_pin_state
==== 0 ====
MENU_pin_select == HIGH
MENU_pin_next == HIGH
MENU_pin_prev == HIGH
MENU_pin_up == HIGH
MENU_pin_down == HIGH
MENU_pin_home == HIGH
945211 ms || 752 milliVolt || 25.00 degrees C || 77.00 degrees F
in u8x8_read_pin_state:: 
U8X8_PIN_INPUT_CNT=[6] i=[0] 1} i=[1] 1} i=[2] 1} i=[3] 1} i=[4] 1} i=[5] 0}  pin_state={11111110}  ;;end u8x8_read_pin_state
==== 0 ====
MENU_pin_select == HIGH
MENU_pin_next == HIGH
MENU_pin_prev == HIGH
MENU_pin_up == HIGH
MENU_pin_down == LOW
MENU_pin_home == HIGH
946237 ms || 752 milliVolt || 25.00 degrees C || 77.00 degrees F
in u8x8_read_pin_state:: 
U8X8_PIN_INPUT_CNT=[6] i=[0] 1} i=[1] 1} i=[2] 1} i=[3] 1} i=[4] 1} i=[5] 0}  pin_state={11111110}  ;;end u8x8_read_pin_state
==== 0 ====
MENU_pin_select == HIGH
MENU_pin_next == HIGH
MENU_pin_prev == HIGH
MENU_pin_up == HIGH
MENU_pin_down == LOW
MENU_pin_home == HIGH
947263 ms || 753 milliVolt || 25.00 degrees C || 77.00 degrees F
in u8x8_read_pin_state:: 
U8X8_PIN_INPUT_CNT=[6] i=[0] 1} i=[1] 1} i=[2] 1} i=[3] 1} i=[4] 1} i=[5] 1}  pin_state={11111111}  ;;end u8x8_read_pin_state
==== 0 ====
MENU_pin_select == HIGH
MENU_pin_next == HIGH
MENU_pin_prev == HIGH
MENU_pin_up == HIGH
MENU_pin_down == HIGH
MENU_pin_home == HIGH
948289 ms || 752 milliVolt || 25.00 degrees C || 77.00 degrees F
in u8x8_read_pin_state:: 
U8X8_PIN_INPUT_CNT=[6] i=[0] 1} i=[1] 1} i=[2] 1} i=[3] 1} i=[4] 1} i=[5] 1}  pin_state={11111111}  ;;end u8x8_read_pin_state
```
you can see that the pin state change is detected.

So I have no idea why a menu event isn't returned.

olikraus, 2018-04-10:
strange...

closed 2018-05-11 by olikraus

tkircher, 2025-03-14:
Also had this issue with a MUI demo program, had to #define U8X8_USE_PINS and use pinMode(pin, INPUT) to get it to work, then noticed the button states were backward (by default all buttons in almost any hardware should be active high, pulled to ground when not pushed). The SpaceTrash demo, on the other hand, worked correctly with no modifications.

olikraus, 2025-03-14:
I personally suggest not to use the u8g2 build-in button detection. There are nice reliable Arduino libs for this purpose...

tkircher, 2025-03-14:
> I personally suggest not to use the u8g2 build-in button detection. There are nice reliable Arduino libs for this purpose...

Thanks for the response. I did notice that in #1524 you mentioned wanting to do this, so is the MUI input debounce example the recommended way to handle this moving forward? Calling mui.sendSelect() to send these events externally, for example.

I saw that in the documentation [here](https://github.com/olikraus/u8g2/wiki/muimanual) and in the code examples with the Arduino library there's still mostly internal button handling by passing button GPIOs to u8g2.begin() with a few examples of external button handling by calling MUI events from the main Arduino loop.

olikraus, 2025-03-14:
Correct, use an external lib and translate the corresponding events to MUI actions: 
https://github.com/olikraus/u8g2/blob/2b75f932b5ef4b4de8edf73e1a690702a35b1976/sys/arduino/u8g2_page_buffer/MUIInput2BtnBounce2/MUIInput2BtnBounce2.ino#L540-L550

There are even more examples on for MUI and bounce2 lib.