olikraus/u8g2 #568: Support for 64x32 SH1106

micooke, 2018-04-10:
Could you please create a profile for a 64x32 OLED with SH1106 controller, HW SPI.

I will try and get a datasheet for you, but it is for the OLED inside many of the cheap NRF bluetooth smart watches i.e. ID107, DO003.

Unfortunately the SSD1306 does not work for this instance, I get noise for the bottom half and the top half of the screen is clipped (I will post a photo).

olikraus, 2018-04-10:
I have created beta 2.22.11, download (as usual) from here: https://github.com/olikraus/U8g2_Arduino

The constructor is not yet part of the examples, but all constructor variants exists, example:
```
U8G2_SH1106_64X32_1_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);
```

I took over code from the sh1106 72x40 device, but I just updated the dimensions.

micooke, 2018-04-11:
Tested, a few minor changes were needed. Please see PR #570

olikraus, 2018-04-11:
> Tested, a few minor changes were needed.

Obviously ;-) 
I did only minimal changes. Thanks for the PR!

You need another beta release?

micooke, 2018-04-11:
Please :+1:

micooke, 2018-04-11:
No rush :blush:

olikraus, 2018-04-11:
beta 2.22.12 is available: https://github.com/olikraus/U8g2_Arduino

olikraus, 2018-04-15:
I think I can close this...

closed 2018-04-15 by olikraus

micooke, 2018-04-15:
Ah yep, sorry i forgot :no_mouth: