olikraus/u8g2 #569: How do I add fonts in u8g2?

skyldh89325, 2018-04-10:
Hello

I would like to add Arial font in u8g2.

I would like to add Arial Font 21px, 20px, 12px. I'd appreciate it if you could tell me how to add this.

olikraus, 2018-04-10:
Sure, you need the font bdf format, then use bdfconv.exe as described in the u8g2 FAQ.

However, there is already Helvetica available. Would this be an option?
There is not so much of a difference: https://www.webdesignerdepot.com/2013/03/arial-vs-helvetica-can-you-spot-the-difference/

Here is the existing font: https://github.com/olikraus/u8g2/wiki/fntgrpadobex11#helvr12

olikraus, 2018-04-10:
Correct, you can just use u8g2_font_helvB18_tf (the bold variant) or u8g2_font_helvR18_tf (regular variant) as argument for the setFont (https://github.com/olikraus/u8g2/wiki/u8g2reference#setfont) command.

skyldh89325, 2018-04-10:
Thank you, have a nice day

closed 2018-04-15 by olikraus