olikraus/u8g2 #571: UTF-8 not working on MKR Zero

RobotGB, 2018-04-11:
Hi,
I've just received my MKR Zero in the mail and followed the installation guide.
However, when I ran the Hello World program, '????' showed up instead of '你好世界'.
At first, I expected this: my Mega also didn't display the characters.

I then uploaded Shennong, the full code is at the bottom.
The only changes I think I made is:
- adding `#pragma execution_character_set("utf-8")` to make the compiler recognise UTF-8
- adding `U8G2_SSD1306_128X64_NONAME_1_SW_I2C u8g2(U8G2_R0, /* clock=*/ PIN_WIRE_SCL, /* data=*/ PIN_WIRE_SDA, /* reset=*/ U8X8_PIN_NONE);   // All Boards without Reset of the Display` above the constructor list (all other constructors work)
- uncommenting `#define FONT u8g2_font_wqy14_t_chinese3` (3 instead of 1)

But now, the display is blank, apart from 'Shen Nong' at the beginning of runtime, and '????????' shows up otherwise **only when the line is scrolling out of view**.
(I've also made my own program that reads text from a file and displays it, but nothing non-ASCII shows up. No question marks though.)
How can I help solve this? I've made sure that the problem isn't Arduino itself (at least, to my knowledge) because in my program, the text in the file is also output on the serial monitor, and the monitor displays the text well.

Many thanks.

```cpp
#pragma execution_character_set("utf-8")
/*

  Shennong.ino
  
  Scroll through Chinese Short Story
  "The Farmer God Shen Nong Tastes All the Plants" 
  
  2KB Ram and >200KB Flash required

  Universal 8bit Graphics Library (https://github.com/olikraus/u8g2/)

  Copyright (c) 2016, olikraus@gmail.com
  All rights reserved.

  Redistribution and use in source and binary forms, with or without modification, 
  are permitted provided that the following conditions are met:

  * Redistributions of source code must retain the above copyright notice, this list 
    of conditions and the following disclaimer.
    
  * Redistributions in binary form must reproduce the above copyright notice, this 
    list of conditions and the following disclaimer in the documentation and/or other 
    materials provided with the distribution.

  THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS AND 
  CONTRIBUTORS "AS IS" AND ANY EXPRESS OR IMPLIED WARRANTIES, 
  INCLUDING, BUT NOT LIMITED TO, THE IMPLIED WARRANTIES OF 
  MERCHANTABILITY AND FITNESS FOR A PARTICULAR PURPOSE ARE 
  DISCLAIMED. IN NO EVENT SHALL THE COPYRIGHT HOLDER OR 
  CONTRIBUTORS BE LIABLE FOR ANY DIRECT, INDIRECT, INCIDENTAL, 
  SPECIAL, EXEMPLARY, OR CONSEQUENTIAL DAMAGES (INCLUDING, BUT 
  NOT LIMITED TO, PROCUREMENT OF SUBSTITUTE GOODS OR SERVICES; 
  LOSS OF USE, DATA, OR PROFITS; OR BUSINESS INTERRUPTION) HOWEVER 
  CAUSED AND ON ANY THEORY OF LIABILITY, WHETHER IN CONTRACT, 
  STRICT LIABILITY, OR TORT (INCLUDING NEGLIGENCE OR OTHERWISE) 
  ARISING IN ANY WAY OUT OF THE USE OF THIS SOFTWARE, EVEN IF 
  ADVISED OF THE POSSIBILITY OF SUCH DAMAGE.  

  English translation:
  The Farmer God Shen Nong Tastes All the Plants 

  Source:
  http://chinesereadingpractice.com/2012/08/22/the-farmer-god-shen-nong-tastes-all-the-plants/

  All his life, Shen Nong had a crystal abdomen, and one could 
  clearly see all of his internal organs. At that time, humans were 
  often getting sick and even dying because they ate things 
  indiscriminately [not knowing if they were good or bad]. Shen 
  Nong determinedly tasted everything everywhere; the 
  good-tasting things he put in a bag on his left side, those were 
  for people to eat; the bad-tasting things he put in a bag on his 
[... 324 lines cut ...]

void loop(void) {

  start_line = top_window_pos / glyph_height;
  top_offset = top_window_pos % glyph_height;
  
  line_cnt = total_lines - start_line;
  if ( line_cnt > lines_per_draw )
    line_cnt = lines_per_draw;

  u8g2.clearBuffer();					// clear the internal memory
  for( i = 0; i < line_cnt; i++ )			// draw visible lines
  {
    /* copy a line of the text to the local buffer */
    u8x8_CopyStringLine(buf, i+start_line, c_str);
    
    /* draw the content of the local buffer */
    u8g2.drawUTF8(0, i*glyph_height-top_offset +glyph_height, buf);
  }
  u8g2.sendBuffer();					// transfer internal memory to the display
  
  delay(SCROLL_DELAY);
  top_window_pos += SCROLL_DELTA;
}
```

olikraus, 2018-04-11:
You use the Arduino IDE? Do you see the Chinese glyphs in the Editor of the Arduino IDE? Are you sure you have configured the Arduino IDE Editor to utf-8?
Also: #pragma execution_character_set("utf-8") ist not required.

RobotGB, 2018-04-11:
1. Yes.
2. Yes.
3. What do you mean by 'configure'?

olikraus, 2018-04-11:
hmmm.. The Arduino IDE creates a temp folder for the build. It includes a c++ version of your ino script. Can you attach this here?

RobotGB, 2018-04-11:
Sure. I found the newest modified folder in my AppData/Local/Temp folder to get this:
[Shennong (2).zip](https://github.com/olikraus/u8g2/files/1899593/Shennong.2.zip)
However, something interesting happened.
I noticed the display now had shown the Chinese characters, but still only when going out of view.

olikraus, 2018-04-11:
ok, zip content looks good. 
I have tested the example with my own mkrzero. Everything works as expected. I assume some other error here.

RobotGB, 2018-04-12:
Update: tried uploading on Linux Mint and the Zero started working (despite some problems with the Arduino GUI), not sure what happened the first time but I guess reinstalling Arduino would help.
Thanks.

closed 2018-04-12 by RobotGB

olikraus, 2018-04-12:
:)