olikraus/u8g2 #572: Only soft-i2c on d-duino (NodeMCU based)

TomTheDragon, 2018-04-12:
Hi,

i played around a little and finally got the 128x64 SSD1306 oled display to run on my d-duino. But only in software mode, which is super slow. I know that it could do better since i tested an other library before.

olikraus, 2018-04-12:
u8g2 includes a hardware SPI device, but I do not know much about NodeMCU. Maybe ask in some NodeMCU forum how to use hardware SPI.

TomTheDragon, 2018-04-13:
I was talking about i2c, not spi and i know that it can do "much" faster then it did on my tests.

olikraus, 2018-04-13:
oh, ok, but same is true for I2C. You have to connect to the I2C pins, which i do not know on your target system.

TomTheDragon, 2018-04-13:
The screen is directly on the board, so its always connected to D1 and D2
http://dong-sen.com/wp-content/uploads/2016/02/D-duino.png

I was using "ESP8266 OLED SSD1306" before, it worked really well and fast. But it cant do things like polygons, so wanted to give u8g2 a try.

olikraus, 2018-04-13:
Did you try HW SPI with HW Pin Remapping?

```
U8G2_SSD1306_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, U8X8_PIN_NONE , D2, D1);
```
or maybe
```
U8G2_SSD1306_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, U8X8_PIN_NONE , 4, 5);
```

olikraus, 2018-04-15:
closing this for now...

closed 2018-04-15 by olikraus

vijju27, 2018-05-03:
I made a pong game using u8g2 library on nodemcu using hardware i2c. My OLED display has sh1106 though.
It works like a charm for me.
Refresh rate is awesome.

olikraus, 2018-05-03:
Sounds good, thanks for the feedback. Even I think HW I2C should work without problem.