olikraus/u8g2 #573: Feature-request: png2bdf executable

NikolaiPrzewalski, 2018-04-13:
Hello,

could you add either:
 a) an Windows executable (hard linked libpng and zlib would be perfect) or
 b) png.h and pnglibconf.h

to the png2bdf tool? I was not able to build the tool with MinGW64. I suspect dependency on a specific version of libpng.

Kind regards
Nicolas

olikraus, 2018-04-13:
I need to check this on my windows machine. My png version is 1.6.34.

olikraus, 2018-04-13:
I have created an exe, but for execution you  need also the two other .dll files in the same folder:
https://github.com/olikraus/u8g2/tree/master/tools/font/png2bdf

olikraus, 2018-04-15:
Did this work for your?

NikolaiPrzewalski, 2018-04-15:
I'm sorry! The weather was too nice the last two days, so I did not test until now.

It works perfectly fine! Thank you!

olikraus, 2018-04-15:
:-)

closed 2018-04-15 by olikraus