olikraus/u8g2 #576: Support for LX-12864P1 LCD (ST7565R)

IMAN4K, 2018-04-15:
Hi @olikraus 
I was thinking to add support for these kind of light-weight LCDs with `ST7565R` driver:

[image]
These are available shots for pinout and connection:

[image]
[image]
[image]

Assuming `D6` as `clock`, `D7` as `data` and `CS` as `chip select` i choose this constructor:
` U8G2_ST7565_64128N_F_3W_SW_SPI(const u8g2_cb_t *rotation, uint8_t clock, uint8_t data, uint8_t cs, uint8_t reset = U8X8_PIN_NONE)`
but no drawing result!

Regards.
IMAN.

olikraus, 2018-04-15:
You must
 * connect A0 (which is named "dc" in u8g2) to a GPIO line
 * connect reset line to a GPIO output
 * use a 4-wire SPI constructor

IMAN4K, 2018-04-15:
Don't know what's wrong with this !?

[image]

I'v tried all `ST7565_*_F_4W_SW_SPI` constructors and the shot is from `U8G2_ST7565_64128N_F_4W_SW_SPI`

olikraus, 2018-04-15:
This already looks quite good. There is nothing wrong with your display, Just the internal memory architecture is different.

So you have tested the other constructors, like this person:
https://github.com/olikraus/u8g2/issues/440

Did you also test the `U8G2_ST7565_LM6059_F_4W_SW_SPI`? If not, can you send me a picture of the result with this display?

IMAN4K, 2018-04-15:
`U8G2_ST7565_LM6059_F_4W_SW_SPI` AND `U8G2_NT7534_TG12864R_F_4W_SW_SPI` both seems working but maybe we've back light problem!

[image]

IMAN4K, 2018-04-15:
The results with different constructors.
`U8G2_ST7565_LM6059_F_4W_SW_SPI`:

[image]

`U8G2_NT7534_TG12864R_F_4W_SW_SPI`:

[image]

`U8G2_ST7565_64128N_F_4W_SW_SPI`:

[image]

`U8G2_ST7565_ZOLEN_128X64_F_4W_SW_SPI`:

[image]

`U8G2_ST7565_ERC12864_F_4W_SW_SPI`:

[image]

`U8G2_ST7565_EA_DOGM128_F_4W_SW_SPI`:

[image]

`U8G2_ST7565_NHD_C12864_F_4W_SW_SPI` same as `U8G2_ST7565_ZOLEN_128X64_F_4W_SW_SPI`

Do you have any idea why changing constructors also change pixels to black like the first picture ?

olikraus, 2018-04-15:
> Do you have any idea why changing constructors also change pixels to black like the first picture ?

This is simple: The contrast needs to be adjusted.
See, it is like this: Your GLCD has two main components: The ST7565 controller and the Liquid Crystal itself. These two things can be connected in a different way, which results in a different memory architecture. Moreover the Liquid Crystal itself has different characteristics. Some need 8V to operate others need 10V others only 6V. And also the timing requirements are different. All in all there are hundreds of different types of ST7565 displays, and we have to figure out what exactly is needed for your liquid crystal. 

My suggestion is to go forward with the `U8G2_ST7565_LM6059_F_4W_SW_SPI`, which looks most promising (because the memory architecture seems to be correct). Now use u8g2.setContrast() to choose a different contrast value. Current default is the value 24, so I would start to try lower values and see whether this improves the picture. Try  `u8g2.setContrast(20)` or `u8g2.setContrast(16)`. Finally you have to test several values, until you find the best one.

olikraus, 2018-04-15:
For the `U8G2_ST7565_LM6059_F_4W_SW_SPI`:
How will u8g2.setContrast(80) look like?

IMAN4K, 2018-04-15:
Using `U8G2_ST7565_LM6059_F_4W_SW_SPI` at contrast `80`:

[image]

olikraus, 2018-04-15:
ok, what about smaller values? like 60 or 40?
Will it change? Will it be better?

IMAN4K, 2018-04-16:
For value 45:

[image]

and 15:

[image]

There is no improvement of changing contrast at all!
Unfortunately i couldn't find any datasheet for the  `LX-12864P1` or `LX-12864N8` module, only for `ST7565R` :(

olikraus, 2018-04-16:
very strange. There should be some improvement.

ok, let us try a different approach: Please locate file u8x8_d_st7565.c on your hard disk

Then goto line 646:

https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_st7565.c#L646

You should see the following:
```
  U8X8_CA(0x0f8, 0x000),		/* set booster ratio to 4x (ST7567 feature) */
  U8X8_C(0x027),		                /* set V0 voltage resistor ratio to max  */
U8X8_CA(0x081, 0x018), /* set contrast, contrast value, EA default: 0x016 */
```
Change this to 
```
  U8X8_CA(0x0f8, 0x000),		/* set booster ratio to 4x (ST7567 feature) */
  U8X8_C(0x027),		                /* set V0 voltage resistor ratio to max  */
U8X8_CA(0x081, 0x01e), /* set contrast, contrast value, EA default: 0x016 */
```

IMAN4K, 2018-04-16:
Yeah it's working.
I updated lib to `v.2.20.13` and changed `u8x8_d_st7565.c(644)`
Half box drawing with contrast `5`:

[image]

and contrast `24`:

[image]

But logo drawing doesn't have good quality as box (?):

[image]

olikraus, 2018-04-16:
ok, looks good. I will change the repository and create a new constructor for you.
Regarding the quality: Try to use lower contrast with the logo (like in your first pic of your last post).

Another reason for the low quality might be your capacitors. All the 2U2 caps need to be high quality ceramic caps. Do not use electrolytic capacitors. Same is true for the 100N caps.

IMAN4K, 2018-04-16:
Thanks for helping on this issue.
The quality is perfect with those cap you said.
Can i add this `LX-12864P1` to Gallery or should be tested on your desk ?

King Regards.
IMAN.

olikraus, 2018-04-16:
Looks great. Good job. I can add the above picture to the gallery.
Can you send me your modifications to the c code? Or is it only the "a2" value?
What contrast value did you use?

IMAN4K, 2018-04-16:
It was just `0x0a3` -> `0x0a2`.
Last picture is with contrast level `2`

olikraus, 2018-04-16:
I have created beta 2.22.15
It contains the constructor U8G2_ST7565_LX12864_F_4W_SW_SPI for your display (see the examples).

Download: https://github.com/olikraus/U8g2_Arduino

olikraus, 2018-04-19:
:-)

closed 2018-04-19 by olikraus