olikraus/u8g2 #578: gpio&delay message problem

Multichiller, 2018-04-16:
Hey Olli, 
I try to use the u8g2 code with a mikrocontroller (c-code) and the included SPI-4 wire (sw), but there is no display on the screen. There is also no data transfer to the display, as I can see on the logic analyser.
The pin should be declarated and the delay messages also handled. Is there any missing part or did I make it wrong?
I would really appreciate your help. Thanks

The complete code.
```
#include <msp430.h>
#include <stdio.h>
#include "u8g2.h"

void initializeIO(void)
{
	//set Direction = Output
	P3DIR |= BIT2;		//SCK 		3.2
	P3DIR |= BIT0;		//MOSI 		3.0
	P4DIR |= BIT1;		//CS 		4.1
	P4DIR |= BIT2;		//CD 		4.2
	P8DIR |= BIT1;		//RST 		8.1
}
void delay100ns(arg_int)
{
	__delay_cycles(1.045);
}
void delay10µ(arg_int)
{
	__delay_cycles(10.45);
}
void delay1ms(arg_int)
{
	__delay_cycles(1045);
}
uint8_t u8x8_gpio_and_delay_template(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
{
  switch(msg)
  {
    case U8X8_MSG_GPIO_AND_DELAY_INIT:	// called once during init phase of u8g2/u8x8
    	initializeIO();
      break;							// can be used to setup pins

    case U8X8_MSG_DELAY_100NANO:		// delay arg_int * 100 nano seconds
    	delay100ns(arg_int);
      break;

    case U8X8_MSG_DELAY_10MICRO:		// delay arg_int * 10 micro seconds
    	delay10µ(arg_int);
      break;

    case U8X8_MSG_DELAY_MILLI:			// delay arg_int * 1 milli second
    	delay1ms(arg_int);
      break;

    case U8X8_MSG_GPIO_D0:				// D0 or SPI clock pin: Output level in arg_int
    //case U8X8_MSG_GPIO_SPI_CLOCK:
    	if(arg_int)
    	{
    		P3OUT |= BIT2; //set Pin high
    	}
    	else
    	{
    		P3OUT &= ~BIT2; //set Pin low
    	}
      break;

    case U8X8_MSG_GPIO_D1:				// D1 or SPI data pin: Output level in arg_int
    //case U8X8_MSG_GPIO_SPI_DATA:
    	if(arg_int)
    	    	{
    	    		P3OUT |= BIT0;
    	    	}
    	    	else
    	    	{
    	    		P3OUT &= ~BIT0;
    	    	}
      break;

    case U8X8_MSG_GPIO_CS:				// CS (chip select) pin: Output level in arg_int
    	if(arg_int)
    	    	{
    	    		P4OUT |= BIT1;
    	    	}
    	    	else
    	    	{
    	    		P4OUT &= ~BIT1;
    	    	}
      break;

    case U8X8_MSG_GPIO_DC:				// DC (data/cmd, A0, register select) pin: Output level in arg_int
    	if(arg_int)
    	    	{
    	    		P4OUT |= BIT2;
    	    	}
    	    	else
    	    	{
    	    		P4OUT &= ~BIT2;
    	    	}
      break;

    case U8X8_MSG_GPIO_RESET:			// Reset pin: Output level in arg_int
    	if(arg_int)
    	    	{
    	    		P8OUT |= BIT1;
    	    	}
    	    	else
    	    	{
    	    		P8OUT &= ~BIT1;
    	    	}
      break;

  }
  return 1;
}

u8g2_t u8g2;
void displayInit(void){

	u8g2_Setup_uc1611_ea_dogxl240_f(&u8g2, U8G2_R0, u8x8_byte_4wire_sw_spi, u8x8_gpio_and_delay_template);
	u8g2_InitDisplay(&u8g2);		//Send initialization code to the display
	u8g2_SetPowerSave(&u8g2, 0);
}

void main(void)
{
	//WDTCTL = WDTPW+WDTHOLD;			// Stop watchdog timer
	u8g2_SetDrawColor(&u8g2, 1);
	u8g2_SetFontMode(&u8g2, 0);
	u8g2_SetFont(&u8g2, u8g2_font_helvB08_tr);
	u8g2_SetContrast(&u8g2, 255);
	u8g2_ClearBuffer(&u8g2);
	u8g2_DrawStr(&u8g2, 55, 55, "Hello, world!");
	u8g2_SendBuffer(&u8g2);
}
```

olikraus, 2018-04-16:
the three delay procedures should look like this:

```
void delay1ms(arg_int)
{
  while( arg_int > 0 )
  {
	__delay_cycles(1045);
      arg_int--;
  }
}
```

Rest looks fine from my perspective.

Multichiller, 2018-04-17:
Thanks.
I have changed the code and there is a data transfer. But the microcontroller only sends the initialization (endless times) to the display and nothing else. (Command/Data line low). 
[image]
[image]
The display works fine with an arduino and your code.
Is there a timing problem?

olikraus, 2018-04-17:
> The display works fine with an arduino and your code.

I am confused. What does this mean?
The orginal u8g2 works fine on an Arduino Board, but your gpio&delay procedure does not work on MSP430?

Multichiller, 2018-04-17:
I only want to note that the display hardware works fine. That is has to be a problem in my code.

Multichiller, 2018-04-17:
I have changed the microcontroller. A Output (C/D) didn't work properly. 
[image]
It seems that the communication is working.
Every Output does his job, but it still gets stuck in the initialization.
The constructor gets activated, also the InitDisplay function and the sleep mode disabled.
I have checked the code multiple times, but I can't find any faults.
Is there any missing part?

olikraus, 2018-04-17:
first bytes look like this: 
```
  U8X8_C(0x02f),            			/* internal pump control */
  U8X8_CA(0x0f1, 0x07f),			/* set COM end */
  U8X8_CA(0x0f2, 0x000),		/* display line start */
  U8X8_CA(0x0f3, 127),		/* display line end */
  U8X8_C(0x0a3),            			/* line rate */
  U8X8_CA(0x081, 0x08f),		/* set contrast */
  
  
  U8X8_C(0x0d1),            			/* display pattern */  
  U8X8_C(0x089),            			/* auto increment */
  U8X8_CA(0x0c0, 0x004),            	/* LCD Mapping */
  U8X8_C(0x000),		                /* column low nibble */
  U8X8_C(0x010),		                /* column high nibble */  
  U8X8_C(0x060),		                /* page adr low */
  U8X8_C(0x070),		                /* page adr high */
```

Only 7 bytes are transfered. It looks like that the transmission stops before   U8X8_C(0x0a3).
In fact even all the values are correct.

Is there some watchdog timeout?

Multichiller, 2018-04-18:
I included one another Watchdog timeout in my displayInit(). It works fine now. Thank you for your help.

If I want a faster transfer, do I have to write my own communication or is there any command to increase the writing speed to the display?
Is there a way to write the value of a variable (like the print function) to the display?

JamesGKent, 2018-04-18:
if you want faster transfer you'll need to use hardware spi instead of software

olikraus, 2018-04-18:
> If I want a faster transfer, do I have to write my own communication or is there any command to increase the writing speed to the display?

You need to replace also the byte level function: u8x8_byte_4wire_sw_spi
On that level you can use your local SPI subsystem for example. There should be some hints on the wiki and also several issues here which deal with this topic.

> Is there a way to write the value of a variable (like the print function) to the display?

well, you could use sprintf, but there are also some undocumented functions: 
https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_u16toa.c
https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_u8toa.c

Multichiller, 2018-04-18:
With sprintf it works perfectly. Thanks
I will try to setup the SPI communication tomorrow.

Multichiller, 2018-04-19:
I've got the SPI-template from here:
https://github.com/olikraus/u8g2/issues/117
uint8_t u8x8_communication_spi(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
```
{
  uint8_t i, b;
  uint8_t *data;
  uint8_t takeover_edge = u8x8_GetSPIClockPhase(u8x8);
  uint8_t not_takeover_edge = 1 - takeover_edge;
  WDTCTL = WDTPW+WDTHOLD;

  switch(msg)
  {
    case U8X8_MSG_BYTE_SEND:
    		data = (uint8_t *)arg_ptr;

         	b = *data;
    		while (!(UCB0IFG&UCTXIFG));				// USCI_A0 TX buffer ready?
    		UCA0TXBUF = b;
    		 for(i = 20; i>0; i--);					// Add time between transmissions to
    		                                        // make sure slave can process information

/*
	 u8x8_gpio_SetSPIClock(u8x8, not_takeover_edge);
	 u8x8_gpio_Delay(u8x8, U8X8_MSG_DELAY_NANO, u8x8->display_info->sda_setup_time_ns);
	 u8x8_gpio_SetSPIClock(u8x8, takeover_edge);
	 u8x8_gpio_Delay(u8x8, U8X8_MSG_DELAY_NANO, u8x8->display_info->sck_pulse_width_ns);
	 data++;*/

    case U8X8_MSG_BYTE_INIT:

    	 P3SEL |= BIT0+BIT1+BIT2;               // P3.0,1,2 option select MOSI MISO CLK
    	 UCB0CTL1 |= UCSWRST;                   // **Put state machine in reset**
    	 UCB0CTL0 |= UCMST+UCSYNC+UCMSB+UCCKPL+UCMODE_2; // 4-pin, 8-bit SPI master
    	                                        // MSB, clock polarity high
    	 UCB0CTL1 |= UCSSEL_2;                  // SMCLK Clock Auswahl
    	 UCB0BR0 = 0x04;                        // Baudrate 13Mhz/4
    	 UCB0BR1 = 0;                           //The clock frequency is scaled by UCB0BR0 + UCB0BR1 * 256
    	 UCA0MCTL = 0;           			    // No modulation
    	 __bis_SR_register(GIE); 				// enable interrupts
    	 UCB0CTL1 &= ~UCSWRST;

      // disable chipselect
      u8x8_gpio_SetCS(u8x8, u8x8->display_info->chip_disable_level);
      // no wait required here

      // for SPI: setup correct level of the clock signal
      u8x8_gpio_SetSPIClock(u8x8, u8x8_GetSPIClockPhase(u8x8));
      	  UCB0CTL0 |= UCCKPH;						// Clock phase
      break;
    case U8X8_MSG_BYTE_SET_DC:
      u8x8_gpio_SetDC(u8x8, arg_int);
      break;
    case U8X8_MSG_BYTE_START_TRANSFER:
      u8x8_gpio_SetCS(u8x8, u8x8->display_info->chip_enable_level);
      u8x8->gpio_and_delay_cb(u8x8, U8X8_MSG_DELAY_NANO, u8x8->display_info->post_chip_enable_wait_ns, NULL);
      break;
    case U8X8_MSG_BYTE_END_TRANSFER:
      u8x8->gpio_and_delay_cb(u8x8, U8X8_MSG_DELAY_NANO, u8x8->display_info->pre_chip_disable_wait_ns, NULL);
      u8x8_gpio_SetCS(u8x8, u8x8->display_info->chip_disable_level);
      break;
    default:
      return 0;
  }
  return 1;
}
```
You wrote that the U8X8_MSG_BYTE_SEND has to be replaced. What statement has to be in this case?  The transfer of one byte? Is it necessary to set CS low to "activate" the slave in this case?

An in this case:
```
// for SPI: setup correct level of the clock signal
      u8x8_gpio_SetSPIClock(u8x8, u8x8_GetSPIClockPhase(u8x8));
      	  UCB0CTL0 |= UCCKPH;						// Clock phase
```
should the Clock level be defined like in the gpio-mapping or is it right to define the clock phase?

olikraus, 2018-04-19:
Remark on your code: I think a `break;` statement is missing for the `U8X8_MSG_BYTE_SEND` message.

Another template is in the code itself:
https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_byte.c#L107

As you can see from that example (https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_byte.c#L118) you need to send `arg_int` bytes, because `arg_ptr` will point to a buffer size `arg_int`. So these `arg_int` bytes have to be transmitted. This is also mentioned along with the description of `U8X8_MSG_BYTE_SEND` , see here: https://github.com/olikraus/u8g2/wiki/Porting-to-new-MCU-platform#communication-callback-eg-u8x8_byte_hw_i2c

> You wrote that the U8X8_MSG_BYTE_SEND has to be replaced. What statement has to be in this case?

Maybe my wording was incorrect. You need to handle `U8X8_MSG_BYTE_SEND`. This means you need to take the values from the `arg_ptr` buffer and send them via your SPI subsystem. 

> The transfer of one byte? 

No, `U8X8_MSG_BYTE_SEND` requests you to handle as many bytes as given in `arg_int`.

> Is it necessary to set CS low to "activate" the slave in this case?

The CS signal has to be changed only for the `U8X8_MSG_BYTE_START_TRANSFER` and `U8X8_MSG_BYTE_END_TRANSFER` messages. See the template for this. There is no need to touch the CS line if you receive the  `U8X8_MSG_BYTE_SEND` message. 

> should the Clock level be defined like in the gpio-mapping or is it right to define the clock phase?

If you plan to use your code only for one specific display, you can hardcode the correct clock phase into your code. 

U8g2 will ALWAYS call your cb function in the following sequence:
1. U8X8_MSG_BYTE_INIT
2. U8X8_MSG_BYTE_START_TRANSFER
3. Any mix of `U8X8_MSG_BYTE_SEND` and `U8X8_MSG_BYTE_SET_DC`
4. U8X8_MSG_BYTE_END_TRANSFER

CS will be set correctly during step 2 and CS will be released during step 4. So, no need to worry about CS during step 3.

Multichiller, 2018-04-21:
> If you plan to use your code only for one specific display, you can hardcode the correct clock phase into your code.
So, it does also work, if I change the value of u8x8_GetSPIClockPhase in u8x8.h from 0x01 to 0x00?

Multichiller, 2018-04-21:
No, I set the clock phase in the U8X8_MSG_BYTE_INIT.
I have a transfer of data to the display, but nothing is displayed. 
The initialization of the display is transmitted normally. (CS Low & CD Low)
The transmittion of the data (for the RAM) doesn't look usual. The CS & CD level does always change.
Should it be like that?
That's the first "block" of data after the initialization.
[image]

That's my current code: Could you please take a look at it, if something is missing or false handeld.

```
uint8_t u8x8_communication_spi(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
{
  uint8_t *data;
  WDTCTL = WDTPW+WDTHOLD;

  switch(msg)
  {
    case U8X8_MSG_BYTE_SEND:
    		data = (uint8_t *)arg_ptr;
    		while( arg_int > 0 )
    		      {
    				while (!(UCB0IFG&UCTXIFG));				// USCI_B0 TX buffer ready?
    				 __delay_cycles(5);
    				UCB0TXBUF = *data;
    				arg_int--;
    				data++;
    		      }
     break;

    case U8X8_MSG_BYTE_INIT:
    	 P3SEL |= BIT0+BIT1+BIT2;							// P3.0,1,2 option select MOSI MISO CLK
    	 UCB0CTL1 |= UCSWRST;								// **Put state machine in reset**
    	 UCB0CTL0 |= UCMST+UCSYNC+UCMSB;					// 4-pin, 8-bit SPI master
    	                                        			// MSB
    	 UCB0CTL1 |= UCSSEL_2;								// SMCLK Clock Auswahl
    	 UCB0BR0 = 0x02;									// Baudrate 13Mhz/4
    	 UCB0BR1 = 0;										// The clock frequency is scaled by UCB0BR0 + UCB0BR1 * 256
    	 UCB0CTL0 &= ~UCCKPL;								// Clock polarity
    	 UCB0CTL0 &= ~UCCKPH;								// Clock phase
    	 UCB0CTL1 &= ~UCSWRST;
         break;

    case U8X8_MSG_BYTE_SET_DC:
      u8x8_gpio_SetDC(u8x8, arg_int);
      break;
    case U8X8_MSG_BYTE_START_TRANSFER:
      u8x8_gpio_SetCS(u8x8, u8x8->display_info->chip_enable_level);
      u8x8->gpio_and_delay_cb(u8x8, U8X8_MSG_DELAY_NANO, u8x8->display_info->post_chip_enable_wait_ns, NULL);
      break;
    case U8X8_MSG_BYTE_END_TRANSFER:
      u8x8->gpio_and_delay_cb(u8x8, U8X8_MSG_DELAY_NANO, u8x8->display_info->pre_chip_disable_wait_ns, NULL);
      u8x8_gpio_SetCS(u8x8, u8x8->display_info->chip_disable_level);
      break;
    default:
      return 0;
  }
  return 1;
}
```

olikraus, 2018-04-21:
> No, I set the clock phase in the U8X8_MSG_BYTE_INIT.

ok, but you do not read the phase information from "u8x8_GetSPIClockPhase". I mean it is the task of your callback function to read u8x8_GetSPIClockPhase() and write the correct values to your SPI subsystem.

Maybe you should also try with lower speed first.

> Should it be like that?

It depends on your display. But basically it looks similar to the software version. To me it looks correct.

olikraus, 2018-04-23:
> In what way does the libary need the information?

The correct clock phase is required by the target display. So this is why it is documented in the data structures of the library. But it is not used or read by the the library itself. Instead it should be read by the byte functions so that the SPI interface can be setup correctly. However, if you implement your own byte function for one specific display, then you can ignore this information and just setup your HAL code for your target display directly.

So: If it works for your specific display, then great, no worry about this. This topic is only important if you want to write a generic byte function.

Multichiller, 2018-04-23:
Ok. I have a proper transmittion, but it doesn't work yet. Any clue what it can be? The SPI protocol does work in a test program. I'll try to find a solution the next days.

olikraus, 2018-05-11:
any progress?

olikraus, 2018-05-15:
Ok, thanks for the feedback. But this somehow still looks like it is a special problem with your hardware or your low-level software, right?

Multichiller, 2018-05-15:
yes. The libary also uses the delayfunction from the u8x8_gpio_and_delay_template for the start and end transmission cases, right? So maybe I should use timers instead of delaycycles...

olikraus, 2018-05-15:
... could be...

Multichiller, 2018-05-15:
I will keep you up to date.

olikraus, 2018-06-17:
hmm... no further comments, can I close this?

Multichiller, 2018-06-30:
Sorry, I totally forgot to comment. I couldn't solve the issue, so I added a Arduino and a second bus-transmission. Thanks anyways for the help!

olikraus, 2018-07-01:
ok, thanks for the feedback.

closed 2018-07-01 by olikraus