olikraus/u8g2 #580: font table format

NikolaiPrzewalski, 2018-04-16:
Hallo Oliver,
jetzt muss ich leider doch noch einmal nachfragen. (Ärgerlich, wenn man bedenkt, wieiviele Tickets Du am Wochenende weggeschafft hast - aber hier hört mein Verstehen des U8G2-Systems auf.)

[Font_Table.zip](https://github.com/olikraus/u8g2/files/1915001/Font_Table.zip)

Ich habe mit dem bdfconv.exe einen Auszug aus dem Font open_iconic_all_4x.bdf erstellt. Wenn ich mir das Ergebnis anschaue (siehe view -> Output.txt), geht der Sprung des letzten Zeichens mit Unicode < 255 in eine mit Nullen gefüllte Lücke und nicht auf das erste Zeichenmit Unicode >= 255, das erst zwei Bytes später kommt.

Beim Durchsuchen der Glyphen mit u8g2_font_get_glyph_data() fällt das nicht auf, weil für Zeichen >= 255 sowieso direkt eingesprungen wird in start_pos_unicode. Mir ist das Ganze aufgefallen, weil ich eine Funktion schreiben wollte, die alle Zeichen des Fonts hintereinander ausgibt.

Ist das eine Designentscheidung oder ein Bug?

Viele Grüße
Nicolas

olikraus, 2018-04-16:
> Ärgerlich, wenn man bedenkt, wieiviele Tickets Du am Wochenende weggeschafft hast

Naja, ich wollte mal wieder ein Release machen... Aber so ist das halt...

Wenn ich mich recht entsinne, war das eine Designentscheidung. Die null-lücke markiert das Ende der nicht-unicode Zeichen. Darüber wir die Such-Schleife verlassen:
https://github.com/olikraus/u8g2/blob/master/csrc/u8g2_font.c#L608

NikolaiPrzewalski, 2018-04-16:
Wenn man's weiß, kann man es ja berücksichtigen. Danke für die Information! Ich pflege das demnächst in die Font-Doku ein.

closed 2018-04-16 by NikolaiPrzewalski

olikraus, 2018-04-16:
wow, ganz schön fleißig

NikolaiPrzewalski, 2018-04-17:
Nur aus Neugier: Wie läßt Du Dir eine Übersicht über den Zeichensatz auf dem Target anzeigen?

olikraus, 2018-04-17:
Auf dem Target? Also für u8g2 gibt es ein bitmap target (TGA), das ist einbebaut in die font tool chain:
https://github.com/olikraus/u8g2/blob/master/tools/font/build/build.c#L1088

Ausserdem besitzt bdfconv eine option, die das TGA bild erzeugt (das sieht ein bischen anders aus). Es wird dann als bdf.tga abgespeichert (man kann den Namen nicht abändern).
Das passiert hier: https://github.com/olikraus/u8g2/blob/master/tools/font/bdfconv/main.c#L430

NikolaiPrzewalski, 2018-05-11:
Meinst Du bei der Adresse ("the above calculated address points to the junp table") die Adresse bei Offset 21+22 ?

NikolaiPrzewalski, 2018-05-11:
Spontan fiele mir ein: 
Wenn die Adresse bei Offset 21 weiterhin auf den ersten 16-Bit-Glyph zeigte, und im Byte davor der Offset zur Glyph-Offset-Tabelle 2 hinterlegt wäre, wäre das neue Format 100% kompatibel zum Vorgänger. Etwa so:
 - letztes 8-Bit-Glyph
 - Null-Byte als Marker für Ende des 8-Bit-Bereichs
 - Neuer Glyph-Table
 - Offset des neuen Glyph-Tables
 - Erstes 16-Bit-Glyph

Das benötigte allerdings exakt einen Sprung mehr.

NikolaiPrzewalski, 2018-05-12:
Wenn die Arbeit mit dem Neuberechnen aller Fonts schon erledigt ist, ist mein Vorschlag natürlich sinnlos.