olikraus/u8g2 #581: More efont fonts.

RobotGB, 2018-04-17:
Hello,
I would like to include another (b14) font into the u8g2 list.
The website to get the efont files is [here](http://openlab.ring.gr.jp/efont/dist/unicode-bdf/efont-unicode-bdf-0.4.2.tar.bz2).
I would like to include the following map (the zip file also includes the b14 font and its bold version): 
[b14.zip](https://github.com/olikraus/u8g2/files/1919704/b14.zip)
The only difference from the usual kanji map is that it includes more obscure characters as well as Halfwidth and Fullwidth Forms.
Please can you also include Unicode range '0-127'?
I know how to use bdfconv, but I would like to be able to easily implement the code without pasting it.
Perhaps you could name it b14_ext?

Many thanks.

olikraus, 2018-04-18:
b14 and b14_b are already in the repo:
https://github.com/olikraus/u8g2/tree/master/tools/font/bdf

I will use the existing bdf.

Is there some better name for kanjimap2? How shell I name the font? u8g2_font_b_12_t_ext?
Or would "u8g2_font_b_12_t_full" fit better?

RobotGB, 2018-04-18:
Another name could be KanjiHyogai (hyogai means characters outside the list).
It's 14 so something like u8g2_font_b14_t_full would be great.

olikraus, 2018-04-18:
ok

olikraus, 2018-04-19:
I leave it open, until I have fully implemented this...

olikraus, 2018-04-19:
I will call the font japanese3, to be more consistent with the existing fonts. Hope this is ok...

olikraus, 2018-04-19:
... also applied the map file to the b16 fonts

olikraus, 2018-04-19:
hmmm... I get a lot of errors with the b14 font:
```
../bdfconv/bdfconv  -f 1 -b 0 -M 'japanese1.map' ../bdf/b14.bdf -n u8g2_font_b14_t_japanese1 -o font.c && cat font.c >>../../../csrc/u8g2_fonts.c
Glyph size problem: encoding=9332/0x2474, width=7, height=14, requested position x=7, y=4 (use BBX 8 ...?)
Glyph size problem: encoding=9333/0x2475, width=7, height=14, requested position x=7, y=4 (use BBX 8 ...?)
Glyph size problem: encoding=9334/0x2476, width=7, height=14, requested position x=7, y=4 (use BBX 8 ...?)
Glyph size problem: encoding=9335/0x2477, width=7, height=14, requested position x=7, y=4 (use BBX 8 ...?)
Glyph size problem: encoding=9336/0x2478, width=7, height=14, requested position x=7, y=4 (use BBX 8 ...?)
```

olikraus, 2018-04-19:
The problem is, that the bitmap is larger than the specified bounding box. For example the 2474 glyph looks like this:
```
STARTCHAR U+2474
ENCODING 9332
SWIDTH 500 0
DWIDTH 7 0
BBX 7 14 0 -2
BITMAP 
00
02
4A
4A
99
...
```
The first argument of the BBX command is the glyph width, however the bitmap line 99 (10011001) has a width of 8 bits. bdfconv would truncate the glyph and change the line to 1001100 (removal of the last 1)

olikraus, 2018-04-19:
I also crosschecked the files, from your zip file: Same error is included there.
My suggestion is this: I will ignore b14 and apply your map file to b12 and b16 only.

RobotGB, 2018-04-20:
That's fine.

olikraus, 2018-04-21:
Added the following fonts:
u8g2_font_b12_b_t_japanese3
u8g2_font_b16_b_t_japanese3
u8g2_font_b12_t_japanese3   
u8g2_font_b16_t_japanese3
u8g2_font_unifont_t_japanese3

RobotGB, 2018-04-22:
Fantastic, thanks!

closed 2018-04-30 by olikraus