olikraus/u8g2 #582: Support for ST7529 240x160 8080 type display.

RazvanDi, 2018-04-18:


RazvanDi, 2018-04-18:
Hello,

I just purchased a display from Farnell, **MCCOG240160C6W-FPTLW**, 
[](url)
240x160 pixels. ST7529 chip, and it is configured for 8080 type communication.
pinout for the display module is: 
1=VDD
2=VSS
3=/CS
4=RS
5=/WR
6=/RD
7=DB0
8=DB1
9=DB2
10=DB3
11=DB4
12=DB5
13=DB6
14=DB7
15=/RS
16=BLA
How can I use this LCD with u8glib2 library, is there support for it ?
I have problems understanding which pins to use on which inputs declarations in the construct line
[image]

olikraus, 2018-04-18:
> How can I use this LCD with u8glib2 library, is there support for it ?

Unfortunately, this controller is not supported.

closed 2018-04-21 by olikraus